tulip-control/tulip-control #120: continuous.py "fails" when the spec is unrealizable

necozay, 2015-02-19:
I have changed a few lines in  `continuous.py`:
```
input_bound = 1.0
uncertainty = 10.01
```
so that it is unrealizable. When I run it with `jtlv`, `ctrl` is not defined in the end but I don't get any error messages.

When I run it with `gr1c`, it fails with an error message about mealy machines (logging output is below). 

I don't know which behavior is the expected one. Though telling the user that the spec is unrealizable without failing within `synthesize` seems like a better option than some error regarding labelled graphs or no output at all.

```
 gr1c input:
 ENV: park;
SYS: home lot X0reach loc [0, 4];
ENVINIT:;
ENVTRANS:;
ENVGOAL: []<>(( ! park ));

SYSINIT: (X0reach)
& (( ( ! ( loc = 0 ) ) | ( lot & ( ! home ) ) ))
& (( ( ! ( loc = 1 ) ) | ( home & ( ! lot ) ) ))
& (( ( ! ( loc = 2 ) ) | ( ( ! home ) & ( ! lot ) ) ))
& (( ( ! ( loc = 3 ) ) | ( ( ! home ) & ( ! lot ) ) ))
& (( ( ! ( loc = 4 ) ) | ( ( ! home ) & ( ! lot ) ) ));
SYSTRANS: [](( ( (  X0reach' ) <-> lot ) | ( X0reach & ( ! park ) ) ))
& [](( ( loc = 0 ) -> (  False ) ))
& [](( ( loc = 1 ) -> (  False ) ))
& [](( ( loc = 2 ) -> (  False ) ))
& [](( ( loc = 3 ) -> (  False ) ))
& [](( ( loc = 4 ) -> (  False ) ))
& []((  ( ( loc' = 0 ) -> ( lot' & ( ! home' ) ) ) ))
& []((  ( ( loc' = 1 ) -> ( home' & ( ! lot' ) ) ) ))
& []((  ( ( loc' = 2 ) -> ( ( ! home' ) & ( ! lot' ) ) ) ))
& []((  ( ( loc' = 3 ) -> ( ( ! home' ) & ( ! lot' ) ) ) ))
& []((  ( ( loc' = 4 ) -> ( ( ! home' ) & ( ! lot' ) ) ) ));
SYSGOAL: []<>(home)
& []<>(X0reach);

------------------------------------------------------------
                               gr1c return code: 255

                               gr1c stdout, stderr:
 

INFO:tulip.synth:converting strategy (compact) to Mealy machine
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "continuous3.py", line 106, in <module>
    sys=disc_dynamics.ts, ignore_sys_init=True)
  File "/home/collab/partial_obs/tulip/synth.py", line 1095, in synthesize
    ctrl = strategy2mealy(strategy, specs)
  File "/home/collab/partial_obs/tulip/synth.py", line 1207, in strategy2mealy
    mach.states.add_from(A)
  File "/home/collab/partial_obs/tulip/transys/labeled_graphs.py", line 175, in add_from
    self.graph.add_nodes_from(new_states, check, **attr)
  File "/home/collab/partial_obs/tulip/transys/labeled_graphs.py", line 900, in add_nodes_from
    for n in nodes:
TypeError: 'NoneType' object is not iterable
```

vraman, 2015-02-20:
With JTLV, the default behaviour of synthesize() when the specification is
unrealizable is to return a list of counter-examples (environment-winning
initial states). For gr1c, it is to return None.

Neither is completely defensible, but in either case, there should be some
indication that synthesis has failed.

Vasu

necozay, 2015-02-20:
Then this is probably a bug because I don't see any counter-examples when using jtlv and I don't see none when using gr1c. Maybe one needs to check the strategy content before line 1095 in `synth.py`:
```
ctrl = strategy2mealy(strategy, specs)
```

johnyf, 2015-02-20:
With gr1c v0.9.0 the above returns:
```
gr1c -r temp.gr1c
Not realizable.
```
I think that for unrealizable designs one is supposed to see `None`. The above reminds me of something, as if I had fixed it at some point, by adding a missing check and early `return` at the top of `mealy2strategy`, but I can't find it. I see that it originates from labeled graphs, which don't exist any more in my version.

necozay, 2015-05-22:
I have a small fix for this "bug", which involves the following changes:

```
@@ -38,6 +38,7 @@ import warnings
 from tulip import transys
 from tulip.spec import GRSpec
 from tulip.interfaces import jtlv, gr1c
+import networkx as nx
 
 # slugs is an optional dependency, so fail cleanly if it is missing.
 try:
@@ -1108,6 +1109,13 @@ def synthesize(
     else:
         raise Exception('Undefined synthesis option. ' +
                         'Current options are "jtlv", "gr1c", and "slugs"')
+    # check if a strategy is found before converting to Mealy
+    if not isinstance(strategy,nx.DiGraph):
+        msg = (
+            'The spec is unrealizable.\n'
+            'The controller is set to None.')
+        warnings.warn(msg)
+        return None
     ctrl = strategy2mealy(strategy, specs)
     try:
         logger.debug('Mealy machine has: n = ' +
```

Should I push it directly to the master branch or should I open a pull request? It seems a bit of an overkill to import `networkx` but given the different objects `jtlv` and `gr1c` interfaces return in case of unrealizability, this was a relatively uniform solution.

slivingston, 2015-09-08:
@johnyf: in [your comment above](https://github.com/tulip-control/tulip-control/issues/120#issuecomment-75172308), you seem to refer to a separate implementation ("I see that it originates from labeled graphs, which don't exist any more in my version."). Is it worth considering for this issue? If yes, could you provide a specific reference? E.g., is it the "other" branch in your fork at [johnyf/tulip-control](https://github.com/johnyf/tulip-control)?

slivingston, 2015-09-10:
I reproduced the bug using the example described in the OP when using `'gr1c'` via `synth.synthesize()` but not when using `'jtlv'`. The latter produces some output, but I expect it to be wrong. Anyway, I was able to reproduce the bug for both solvers using a smaller example that I am going to add as a regression test.

closed 2015-09-10 by slivingston

slivingston, 2015-09-10:
@necozay I just committed (2c2377d5b2057ce1350c022ba372409f7625ca27) a patch that relies on `strategy == None` to indicate that the request to `synth.synthesize()` is not realizable. My motivation is that we should handle the details of each solver within a conditional block, e.g., `if option == 'gr1c'`, and then once we are outside the per-tool treatment, we should be able rely on `strategy` being one of several types independent of the particular external tools.

Otherwise, I think that the patch you proposed could be used, and we can switch to it if my argument about using `None` is not convincing. In any case, note that importing `networkx` as in your proposed patch incurs almost no cost because it will have already been imported no later than the call, e.g., to `interfaces.gr1c.synthesize()`.

johnyf, 2015-09-11:
Re [above](https://github.com/tulip-control/tulip-control/issues/120#issuecomment-138654973): I had addressed a similar issue about `slugs` on branch `johnyf/other`, in [`d99fc27976ae44771ca2fc817711491c17cc9046`](https://github.com/johnyf/tulip-control/commit/d99fc27976ae44771ca2fc817711491c17cc9046). It is similar to 2c2377d5b2057ce1350c022ba372409f7625ca27. The difference is checking `not A` as a precondition of the function `synth.strategy2mealy`, instead of before calling it from the body of `synth.synthesize`.

Regarding the [comment about `None`](https://github.com/tulip-control/tulip-control/issues/120#issuecomment-139157857), I think that checking for `None` is safer, because it remains unchanged if one returned another class as result from synthesis.

Besides, I have concluded in other work that it may be preferable to return `True` or `False`, and nothing else. The synthesis result would be stored in a file, to be loaded and enumerated or not as a separate step (post-processing). This avoids difficult-to-maintain conditionals that try to handle both enumerated and symbolic strategies.

Currently, this conclusion does not apply to `tulip`, because strategies have one representation (enumerated). But, in general, they may be either, so returning a different object in each case would be inconsistent. In more detail, one can argue that a symbolic strategy should be returned, and the enumeration step is the further (optional) step. This depends on the level of interface. For research on synthesis itself, this level of detail is convenient. For introductory purposes, examples with limited universal quantification are involved, so enumerated strategies do not blow up.

commit 2c2377d5 (2015-09-10): BUGFIX: Return None when request to synthesize() unrealizable

and a regression test for the same. The bug arose from two causes.
First, different solver interfaces return values of different types if
the specification is not realizable. Second, the function that
constructs a MealyMachine object from the returned value was placed
before any check about the type of that value.

Closes #120