tulip-control/tulip-control #123: Where to offer the JAR file that contains the JTLV-based solver?

slivingston, 2015-03-25:
As of [10033772549067960fbd021008393b4d4d90dd1b](https://github.com/tulip-control/tulip-control/commit/10033772549067960fbd021008393b4d4d90dd1b), [gr1c](http://scottman.net/2012/gr1c) is the default GR(1) solver. However, the JAR file that contains the JTLV-based solver (`tulip/interfaces/jtlv_grgame.jar` in the source tree) is still included among release distribution files, via [MANIFEST.in](https://github.com/tulip-control/tulip-control/blob/10033772549067960fbd021008393b4d4d90dd1b/MANIFEST.in). Because it is

* an optional dependency and
* a large file (2.3 MB) relative to other parts of a TuLiP release,

we should consider providing some other means to obtain the solver. However, I am not aware of any alternative point of distribution for it. Also note the current location of the JAR file: [package data](https://github.com/tulip-control/tulip-control/blob/10033772549067960fbd021008393b4d4d90dd1b/setup.py#L307), as declared in `setup.py`. While we may continue to include it in the TuLiP repository, as an optional dependency it may be more appropriate under some other directory, like `extern/`.

The issue to be resolved here is how users should be able to obtain the JTLV-based solver if they want it. Possibilities:

1. keep it as-is, included in TuLiP releases and under `tulip/interfaces/` (i.e., part of the `tulip` package itself).
2. include it as a supplemental file in the TuLiP release, but move it outside `tulip`, so that is not a part of the `tulip` Python package.
3. remove it entirely from the repository and place it on some public file server. In this case, we could try coordinating with other projects, e.g., [LTLMoP](https://github.com/LTLMoP/LTLMoP).

johnyf, 2015-03-26:
The response that I originally favored was to move it under `extern`, and not include it in `MANIFEST.in`, because it blows up the distribution on PyPI, unnecessarily increasing the server load.

I support deleting it completely from the `tulip` repository, for the following reasons:

- moving it to `extern` may create a new object in the `git` history, unnecessarily increasing the repository size (though I am not sure about this)

- source code is not available for this particular binary, so its maintenance is impossible. Over time, its use will decrease, until it is deprecated completely, and useful only for archiving reasons to compare with past studies.

- since optional, it is not by definition necessary to include it in the Python distribution archive.

- if a Python solver is included or imported in the future, then that can serve as a default, inefficient pure-Python solver.

Instead, I suggest that a (very simple) script be added under `extern`, which:

1. fetches the file directly from either:
  - the GitHub repository (an older commit, where it exists undeleted, in a similar way as we can navigate to a specific "raw" file at a specific commit hash), or
  - from an archive placed at GitHub.
2. checks for `java`, and raises a warning or error if it is not found in the `PATH`.

This script can be included in `MANIFEST.in`, which conforms also to the purpose of `extern`.
Note that this treats (this specific binary of) `jtlv` as a separate entity for distribution.

In the future, command-line or "extras" options could be added to `setup.py` that invoke scripts from `extern` to fetch dependencies selected by the user.

slivingston, 2015-08-09:
I strongly support the recommendation by @johnyf above to delete it from the repository. This issue has been open for a while, and no one has provided arguments to keep the JAR file, so we shall wash ourselves. The action item is as follows.

Create a script under `extern` that fetches the JAR file. It will be easiest and most robust to obtain the file using an old commit, e.g., the last one in which the file was touched, f2c2d7203e795cbabcf9c75516f437475f9faa9e. The corresponding GitHub URI is <https://github.com/tulip-control/tulip-control/raw/f2c2d7203e795cbabcf9c75516f437475f9faa9e/tulip/interfaces/jtlv_grgame.jar>

johnyf, 2015-08-09:
I think that a fetching script is a good idea. Related projects (`gr1c` and others) use similar approaches for fetching CUDD, `slugs` or other dependencies, especially for automated testing with Travis.

Regarding the status of `jtlv`, I can confirm that it is currently unmaintained. There do exist current users, but I do not think that there is any reason for not using a pure Python solution, instead of one in Java.

murrayrm, 2015-08-09:
I agree we should delete it from the distribution.  Leaving a path for people who want to install it to fetch it from our old distribution seems fine.  I wouldn't install it by default (for the many reasons already listed above).

slivingston, 2015-09-10:
Could someone using Mac OSX try extern/get-jtlv.sh ?

closed 2015-09-10 by slivingston

slivingston, 2015-09-10:
Regarding the solution [proposed above](https://github.com/tulip-control/tulip-control/issues/123#issuecomment-86259426) by @johnyf , I do not check for `java` in the script that fetches the JAR file. The presence of `java` is already checked in `setup.py`, and I think that the script is generally more useful without requiring `java`.

johnyf, 2015-09-11:
Tested under OS X v.10.8.5 as of 1fa52c06dc14d7e7ba2c930a489c72225cdf7dc6.

I agree that the check for presence of `java` should better happen at a single place.

Regarding the comment [above](https://github.com/tulip-control/tulip-control/issues/123#issuecomment-86259426) about a Python solver, noting that such a solver will become available for `import` in the near future.

commit f717086b (2015-09-10): Delete JTLV-based solver; create infrastructure to fetch it

The fetching script is extern/get-jtlv.sh. The comments therein
descibe usage and what it does. I have only tested it on Linux x86_64.
Note that it uses cURL, whereas extern/ecos/get.sh uses wget.

This changeset is nontrivial because we now manipulate package data
depending on the presence of an optional item. Messages emitted when
attempting to install or check dependencies of TuLiP should be
sufficienly descriptive to aid users who want the JTLV-based solver.
The code introduced here should be useful for similar cases of
predicate [... cut]