tulip-control/tulip-control #125: finding the post states in a FTS

necozay, 2015-05-04:
I have a FTS `env_fts`. I want to find from state `s1`, with `sys_action = 'a1'`, what the possible post states are. Is there a simple command for doing this? My current solution requires couple of lines of code.

slivingston, 2015-05-04:
Try something like
```python
set([e[1] for e in env_fts.transitions.find('s1',with_attr_dict={'sys_action':'a1'})])
```

slivingston, 2015-05-12:
@johnyf Do you have any recommendations?

@necozay Does my reply qualify as "simple" from your perspective, so as to be sufficient to close this issue? Or was your original intent in opening this issue to collect a variety of expressions for computing possible post-states?

necozay, 2015-05-12:
Thanks @slivingston, your response is close to what we ended up doing. When done in a less pythonic way (as I usually do things), it takes a few lines of code. I was expecting something simpler and thought that we could not find it due to lack of documentation. Therefore, I opened the issue.  

It might be convenient to have a method for this as it is a very natural operation when working with transition systems. Might be OK to close the issue if this can be addressed together with #18, and #89. Or, it can be labeled as "feature request" or "enhancement" if it makes sense to simplify this operation in the future.

johnyf, 2015-05-13:
An approach using the `networkx` method [`edges_iter`](https://networkx.github.io/documentation/latest/reference/generated/networkx.MultiDiGraph.edges_iter.html?highlight=edges_iter#networkx.MultiDiGraph.edges_iter)  is

```python
[u for u, d in env_fts.edges_iter(s1, data=True) if d['sys_action'] == 'a1']
```

This is also what [`Transitions.transitions.find`](https://github.com/tulip-control/tulip-control/blob/972ba362efcc5658197d56d44f7b19bd1a553694/tulip/transys/labeled_graphs.py#L586) is doing, albeit in a much more convoluted way.

commit d435e598 (2015-10-23): DOC: finding the post states in a FTS

Closes #125. [ci skip]
closed 2015-10-23 by slivingston