tulip-control/tulip-control #126: docstring of synthesize requires revision

necozay, 2015-05-22:
As far as I see, both `gr1c.synthesize` and `jtlv.synthesize` return `Digraphs` when a strategy exists, but the doc strings say they return an instance of a `MealyMachine`.

slivingston, 2015-05-23:
Confirmed. For `gr1c.synthesize()`, the two relevant changesets leading to this error are
* 7db7e9fdd40a4661ac4c03651fa263b024f36cb9, where the return value of `aut_xml2mealy()` changed to be of type `networkx.DiGraph` for the second return path; note that `aut_xml2mealy()` has since been renamed to `load_aut_xml()`
* 74807640a1a1fe6da2e579189dc5ce4a0b0ea118, where `gr1c.synthesize()` was modified to accept the new form of return value from `load_aut_xml()`.

I write "second return path" above because there is an alternative `return` that may be reached in `load_aut_xml()` and which has return type `(GRSpec, None)`. Thus the docstrings need to be updated for both of these functions.

slivingston, 2015-05-23:
For `jtlv.synthesize`, the two relevant changesets are
* afd427365ec8e0a580a3bab09a4ecad1171f595c, where the return type of `load_file()` changed to `networkx.DiGraph`; note that `load_file()` has since been renamed to `jtlv_output_to_networkx()`.
* 772eeaf6bcc8d4ed60c494c973f2924491ee9331, where the docstring of `jtlv_output_to_networkx()` was updated to declare the correct return type, but not `jtlv.synthesize()`.

commit ad86bca1 (2015-06-01): DOC: Correct return types in docstrings

Closes #126
closed 2015-06-01 by slivingston