tulip-control/tulip-control #129: Reaction function in Machines.py does not work as intended

iamkaushik5, 2015-07-31:
Hi, my name is Kaushik and I am a graduate student working with Necmiye Ozay on Tulip as part of a summer project and I wanted to bring this issue to everyone's attention regarding the reaction function in machines.py

Currently, the reaction function allows for stepping through a created mealy machine by providing the next input. However, the current function requires the input specification to match the entirety of the input spec in the mealy machine. However, when running most simulations, only a few inputs are required to identify a single path from the current state. Please see the code below:

  enabled_trans = [
            (i, j, d)
            for i, j, d in self.edges_iter([from_state], data=True)
            for k in inputs.iteritems() if k in project_dict(d, self.inputs).items()] #added
            # if project_dict(d, self.inputs) == inputs]

The commented line at the end is what is currently in the function. Notice how it requires that the input be directly equal to the input spec in the machine. Instead, the for statement above was added as a temporary fix. This for statement basically enables a transition if a single element of the inputs matches a single element of the output. This allowed me to simulate my example that I am working on because I only needed to provide the next state as input. However, this doesn't help for cases where multiple inputs are actually required to identify a single path from the current state. In that case, this fix would actually enable the same transiton twice, which will cause a determinism error because the program thinks there are two possible paths. Does anybody have any suggestions as to a permanent fix?

Second,
The error messages are not proper. The function seems to default to the 'this is not deterministic' error regardless of whether that is the issue. For example, if I sent inputs into the function that are not valid options from the current state, I get a determinism error, when really the error should say that 'these are not valid inputs'

Thanks.

-Kaushik Kannan

slivingston, 2015-07-31:
I may not understand your first item, but I think that requiring all inputs is correct and as intended. Your observation that knowing all inputs is not necessary in some cases is a particular case and, I can argue, not appropriate for this function. Perhaps you want to propose a new function, or some new way to request lazy evaluation of inputs when calling `reaction()`?

Regarding the second item, could you elaborate all of the errors and messages that you want?

Note that the error message is not as you quote it, but rather "must be input-deterministic". Input-determinism in different from determinism.

My comments above are based on the tip of master in the repo. Let me know if you are using a particular release of TuLiP.

necozay, 2015-07-31:
A related issue to the first point is #29. I think the issue comes up when simulating synthesis problems with an environment transition system as in abstractions of mode-only switched systems. For instance, to simulate the controller synthesized in ``only_mode_controlled.py``, in addition to ``eloc``, one needs to specify the values of ``home`` and ``lot`` propositions. Maybe an alternate reaction function that allows lazy evaluation of inputs but still checks input-determinism would be useful.

iamkaushik5, 2015-08-04:
To follow up on the error message issue, here is an example from only_mode_controlled.py. In the synthesized controller, there is only 1 path from 'Sinit' and that path is defined by the input: {'eloc': 's0', 'park':1, 'home':1, 'lot': 0}. However, if I input the following instead: {'eloc': 's1', 'park':1, 'home':0, 'lot': 0}, I get the error: "Exception: must be input-deterministic, found enabled transitions: []"

In this case, I think a more useful error would be something like "exception: not a valid input from current state, possible inputs are: [{'eloc': 's1', 'park':1, 'home':0, 'lot': 0}]" I think this would be a bit more user friendly also because it would list out the possible options from the current state. In the current error, the enabled transitions are empty because a transition is only enabled if your given input matches a possible input path from the current state, so in the case where your input doesn't match any, the error doesn't give you much useful information. 

Thanks!

slivingston, 2015-08-04:
I just changed the title of this issue to focus it on the first item of the OP.

@iamkaushik5: Regarding the second item, could you make a proposal about the form of error messages by opening a pull request or sending a patch via email or as a `diff` code block in a comment on this issue?

@necozay: Let me know if this does not fall under the U. Michigan copyright.

slivingston, 2015-08-04:
@iamkaushik5: We can wait to discuss this once you propose a patch, but a word of caution: because there may be a very large number of possible inputs from a particular state, an exception message that lists possibilities may involve a very large string that crowds the output. Alternatives include:

* listing at most 3 (or some fixed number of) possible inputs;
* not listing possible inputs, and instead describing the mismatch between the given input and the set of feasible inputs;
* not listing possible inputs, and instead providing the current state, or example code for how to obtain all possible inputs from that state.

iamkaushik5, 2015-08-04:
That is a good point @slivingston . I will keep these alternatives in mind when I try to code the patch. Thanks!

slivingston, 2015-09-08:
For posterity, here is an example that demonstrates my proposed realization of the desired behavior.

```python
from __future__ import print_function
from tulip.spec import GRSpec
from tulip import synth

spc = GRSpec(env_vars={'e0', 'e1'},
             env_init = ['e0','!e1'],
             env_safety=['e0 <-> X !e0', 'e1 <-> X !e1'],
             sys_vars={'y'},
             sys_prog=['y'])
M = synth.synthesize('gr1c', spc)

print(M.reaction(1, {}, lazy=True))

```
which yields the output `(0, {'y': 1})` as of 5cb9d24f0c507f742a73c2d6b0c3e35e60aaf5d2 after applying my forthcoming patch of `reaction()`. N.B., the assumption of a state named `1` is fragile, but the intended behavior should be clear.

slivingston, 2015-09-08:
Realized in e6ddd91d7500837200e5d550ff4ec6e208f7172c and bb693b45e4cd7e4effcf6466350ad673617812bb

slivingston, 2015-09-08:
Continuing the example [above](https://github.com/tulip-control/tulip-control/issues/129#issuecomment-138724086), the error message that suggests possible inputs is demonstrated with
```python
print(M.reaction(1, {}))
```
which results in "Exception: not a valid input, some possible inputs include: [{'e1': 0, 'e0': 1}]". Also,
```python
print(M.reaction(1, {'e0':0}, lazy=True))
```
which results in "Exception: not a valid input, some possible inputs include: [{'e0': 1}]". Please tell me if this is not the intended behavior.

closed 2015-09-08 by slivingston

necozay, 2015-09-09:
The only condition that comes to my mind that might not be covered is the following. If someone does not use the default "trimming" function for the synthesized mealy machines and ends up in a state where there is no valid input (e.g., environment spec is violated), then the message will say something like ""Exception: not a valid input, some possible inputs include:[]" (empty set), right? Though if the person is experienced enough to change the default settings, they can probably figure out what is going on.

slivingston, 2015-09-09:
Your assessment is correct about a case in which the empty set is given. Do you recommend that, in the case you describe, the exception message should be like "state 1 is a dead-end. There are no possible inputs from it."?

necozay, 2015-09-09:
I think the error message you proposed would be the most appropriate in such a case where there is a dead-end. This is a very minor issue so I am fine if we do not address it before the upcoming release.

slivingston, 2015-09-10:
Done in ef67131ee362be0d984f4cb84a7d56d31f7447b4. A demonstration is
```python
from tulip.transys.machines import MealyMachine
M = MealyMachine()
M.states.add(2)
print(M.reaction(2, {}, lazy=True))
```
which yields the output "Exception: state 2 is a dead-end. There are no possible inputs from it.". N.B., this example is distinct from the use-case [described above](https://github.com/tulip-control/tulip-control/issues/129#issuecomment-138737284) but captures that the `from_state` is a dead-end.