tulip-control/tulip-control #131: Expand Travis CI configuration to perform all tests

slivingston, 2015-10-19:
https://travis-ci.org/tulip-control/tulip-control

Currently (as of 54d122fbd76a0188f138521628c53132b2aea8ce), only the "base" family of tests is performed. We should expand the CI configuration to perform all tests, i.e.,
```sh
./run_tests.py full
```

slivingston, 2015-10-23:
While addressing this issue, also improve the manner of getting scipy. Currently it sometimes requires so much time that the build fails due to timeout.

johnyf, 2016-06-16:
`conda` has been used in 7e9c4b42f29100b85dda7433454116058e25db0b to install `scipy` in less time.

johnyf, 2016-08-06:
As of b644de60e0a24e48f65fba925dccbb8443c46a6a, all tests for supported features pass, with pure Python support for GR(1) (and Rabin(1)) synthesis. The omitted tests are for:

- `tulip.interfaces.jtlv` (to be completely removed in the future)
- `tulip.interfaces.lily` (experimental, not a production feature)
- `tulip.interfaces.slugs` (outdated, unmaintained, depending on a module available via a fork, more functionality available via other interfaces, and in pure Python)

The `full` family of tests includes these, so a new test family `ci` has been created in `run_tests.py`.

slivingston, 2016-08-06:
I did not know that there was a distinction with some features being "production". I think that if it is tested in the `full` family, then it should be tested as part of CI jobs. This issue is not addressed until that is the case, so there is no need for a special `ci` family.

slivingston, 2016-08-06:
The relevant issue about completely removing `tulip.interfaces.jtlv` in the future (as described [above](https://github.com/tulip-control/tulip-control/issues/131#issuecomment-238011328)) is #158.

johnyf, 2016-08-06:
fa52b12d3bfc4bd5e54b7249f03526377fc6a6b4 tests also `jtlv` and `lily` on Travis CI.
The `slugs` test remains (discussed below).

In https://github.com/tulip-control/tulip-control/issues/154#issuecomment-236415425 the "all tests" was intended to refer to the optional but maintained features. Therefore I misunderstood the OP of this issue.

I will not be able to maintain `tulip.interfaces.slugs`. I do not know whether it runs with the current version of `slugs` and the Python interface available with `slugs`. If the interface can be maintained, then the corresponding test could also be updated, if it needs changes. If being unmaintained would imply that this interface be deprecated / removed, then I do not object to its removal.

My attempt to build `slugs` be6b5148f4f56391d6928649cf6f7c2372e007fb on my computer following the instructions in the README results in:
```
error: ‘nextafter’ is not a member of ‘std’
```
If it can be of any use:
- a script to install an earlier version of `slugs` on Travis CI can be found [here](https://github.com/johnyf/openpromela/blob/8eb71e733b723761fbacc4c2c22eea095cd2ab11/install_slugs.sh).
- a more recently used interface can be found [here](https://github.com/johnyf/openpromela/blob/7cea73fbbba40fd1f8a8886027f71928956bbfde/openpromela/slugs.py).

`slugs_test.py` is the only one among the files `tests/*_test.py` in the above commit (which are those included in `full`) that is not run on Travis CI on the branch `enumeration`.
The other tests included in `full` [pass](https://travis-ci.org/tulip-control/tulip-control/builds/150258881).

The `ci` family will be removed.

The tests that I was planning to successfully run with Travis CI for the purpose of testing work on the branch `enumeration` and to ensure that whatever does not use `cvxopt` does not fail due only to failure to import `cvxopt` have been updated and have passed. Given this, I will now remove myself from the list of assignees.

johnyf, 2016-08-20:
The `ci` family has been removed, and most of the above tests (and necessary corrections) applied either to branches `master` or `enumeration`. The only tests currently separate are for `jtlv` and `lily`, placed in e3ef20dd9645fc4ef5c8bb67b64bc6112e771ee3. The `jtlv` test can remain there until before #158 is effected, at which point, for leaving a consistent state in history, the test can be included in `master`, and then `jtlv` removed.

The `slugs` interface may follow the `jtlv` interface (depending on future discussion), which would resolve this issue.

slivingston, 2016-08-21:
Following discussion in #163, additional tasks for this issue are the following:
* move tests that depend on gr1c to the `full` family;
* separate tests that involve `dd` and require CUDD from those tests that do not require CUDD.

slivingston, 2016-08-23:
I was able to build `slugs` on Debian GNU/Linux 8.5 using GCC (g++) 4.9.2. However, I failed to find any tests, so I asked in the `slugs` issue tracker https://github.com/VerifiableRobotics/slugs/issues/15

I am willing to eventually drop `slugs`, but they just published a "tool paper" at CAV 2016, so it is worth some attention again, at least enough to compare it with others using benchmarks, or to decide if it offers some substantial features.

johnyf, 2016-08-23:
It is fine with me to not support `slugs` here. In any case, the intermediate layer of `omega` is in a superset of the Slugsin format, so it can be used for input to `slugs` if needed (I have used it in this way).

murrayrm, 2016-08-23:
I would advocate keeping in at least some support for various synthesizers that others have written, as a way of allowing comparisons between solvers but also increasing the use of TuLiP as a toolbox that provides access to a wide variety of tools.

If it is a pain to maintain the interfaces (and unit tests, CI), we could move to the contrib/ directory or something along those lines.

slivingston, 2016-08-28:
I found a solution for fetching and using `slugs`. It can be simplified if they begin to use versioning, which I have requested at https://github.com/VerifiableRobotics/slugs/issues/16

johnyf, 2016-09-01:
Implemented by #165.

closed 2016-09-01 by johnyf

slivingston, 2016-12-01:
@johnyf notified me via email about the existence of pre-compiled binaries for `numpy` and `scipy`, which can be obtained using

    pip install --only-binary=numpy,scipy numpy scipy

If we use this as part of the CI configuration, then we can remove `conda` from [.travis.yml](https://github.com/tulip-control/tulip-control/blob/02a407765f2f4ca995a6691587e8170fdbd385e6/.travis.yml#L22-L33).

slivingston, 2016-12-06:
With relevance to #44, I was able to successfully and quickly complete the above command with Python 3.5.2 and `pip` 8.1.2