tulip-control/tulip-control #132: PLY caching fails during installation

slivingston, 2015-10-28:
Multiple users have reported that for release version 1.1a attempting installation multiple times does not cause the warning message
```
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
     Failed to build PLY table.  Please run setup.py again.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! 
```
to go away.

johnyf, 2015-10-28:
What are the circumstances under which it has been observed? (`pip` or `python setup.py install`).
Also, `ply` has evolved since then (not incompatibly, but my recent experience has been with code written using newer `ply` versions).

slivingston, 2015-10-28:
For one of the reports,

    python setup.py install

* RHEL (Red Hat Enteprise Linux)
* installed as root
* system-wide, under the directory /usr/local/python/lib/python2.7/site-packages
* Python 2.7.9
* PLY 3.6
* setuptools 18.1

slivingston, 2015-10-28:
[The other report](https://sourceforge.net/p/tulip-control/mailman/message/34528765/) was copied to the tulip-control discuss mailing list around 8 Oct 2015. The sender describes the platform as "Ubuntu 14.04.3 LTS Trusty" and sent copies of "the detailed error log" and the output obtained using run_tests.py.

necozay, 2015-10-29:
Are they doing it over ssh? I was having a similar problem on "Ubuntu 14.04.3 LTS Trusty" when I run the tests over ssh (running them directly on the machine with the same user works without any problem). I resolved the issue in the end but I could not locate my solution now.

johnyf, 2015-10-29:
That may suggest write permissions. I have not had a similar experience with `ssh`, but the "installed as root" differs in my case, i.e., I install as the user I am. Perhaps trying with normal user access rights would be a candidate experiment.

johnyf, 2015-10-30:
With another package, I observed a different behavior that results in an outdated table being installed. This appears to apply to all packages that use `ply`. `pip >= 7.0` creates and caches wheels on its own, even for packages that don't distribute a wheel. As a result, it does *not* re-run `setup.py` when re-installing.

As a result, when `ply` is updated, re-installing a package dependent on `ply` does not rebuild the tables. It is re-installed from the existing wheel, that was created with an older `ply` version. Yet, I am not sure whether that table file is the one distributed with `sdist`, or the one rebuilt when `pip` locally made the wheel.

This caching behavior of `pip` can be disabled with [`--no-cache-dir`](http://pip.readthedocs.org/en/stable/reference/pip_install/#wheel-cache), thus forcing an update of the `ply` tables upon re-installation of a package.

These are the consequences of double premature optimization: from both `pip`'s and `ply`'s side. We will see how worse it can get.

johnyf, 2016-06-16:
In absence of further evidence, and not being able to reproduce, closing.

closed 2016-06-16 by johnyf