tulip-control/tulip-control #137: encoding of Mealy machine states

necozay, 2015-12-08:
The states of the synthesized Mealy machines look a bit odd in the examples as of the head of master branch:
```
>>> print ctrl.states
States:
[u'0x9980E0',
 u'0x997D30',
 u'0x997F30',
 u'0x997F70',
 u'0x997C60',
 u'0x9978D0',
 'Sinit',
 u'0x997DC0',
 u'0x997E00',
 u'0x998210',
 u'0x9980A0',
 u'0x997B60']
```
Is this an intentional change or is this something to do with my setup?

johnyf, 2015-12-08:
If you are using `gr1c`, then they appear normal, because it gives hexadecimal values to states, as in the example shown in [its documentation](http://slivingston.github.io/gr1c/md_formats.html).

necozay, 2015-12-08:
I was just confused seeing different results in TuLiP in two different machines. Didn't notice this was due to different versions of `gr1c` installed on these machines. This seems to happen with `gr1c 0.9.0` but not with `gr1c 0.10.2`.

johnyf, 2016-06-12:
Hexadecimal names strategy nodes result also with `gr1c == 0.11.0`.

closed 2016-06-12 by johnyf