tulip-control/tulip-control #138: Bug in discretize, conservative=False option

necozay, 2015-12-09:
I think `orig_ppp` should be updated to convexified partition within the part of the `discretize.py` code handling the `conservative=False` function (lines 530-548). If this is not done, it is not clear how this option can be used to construct a controller.

necozay, 2015-12-09:
The following addition to `get_input` can be used as a test:
```
@@ -240,6 +240,14 @@ def get_input(
     else:
         # Take original proposition preserving cell as constraint
         P1 = original_regions[orig[start]]
+        # must be convex (therefore single polytope?)
+        if len(P1) > 0:
+            if len(P1) == 1:
+                P1 = P1[0]
+            else:
+                print P1
+                raise Exception("conservative = False flag requires "
+                                "original regions to be convex")
```

If this exception is not caught, `get_input` fails because it sends a region to `get_input_helper` instead of a polytope.

johnyf, 2016-11-18:
The patch [above](https://github.com/tulip-control/tulip-control/issues/138#issuecomment-163139511) has been included in https://github.com/tulip-control/tulip-control/commit/4d1cac37fc39846e9d7986536e6a496c7dbc37f4#diff-db73a44c4631fa5fc7c04bff0f7d729aR251.