tulip-control/tulip-control #141: LICENSE at a single place

johnyf, 2016-01-26:
Since now `tulip` does have a license file at the top level, as added in 36f3d4135023cf633871e0ff32f896e4a4198ad9, would it still remain necessary to replicate the BSD text in every file?
This suggests a candidate revision to the policy for declaring the license within `tulip` sources.
In particular, we could consider removing the license copies from the Python source files.

cf also https://github.com/tulip-control/tulip-control/pull/135#issuecomment-167295876

slivingston, 2016-01-29:
It is not necessary legally. An important motivation for the practice is to prevent loss of information about the origin when subsets of the code are used or re-distributed. The license precludes that from happening, but carelessness can cause otherwise.

However, there may be institutional rules at Caltech or U. Michigan that require it. @murrayrm or @necozay, is that the case?

Elaborating the suggestion in the OP, we could replace the license text with a pointer to the LICENSE in the tulip-control repository and the URL <http://opensource.org/licenses/BSD-3-Clause>. The section "How to Apply These Terms to Your New Programs" of the [GNU GPL](http://www.gnu.org/licenses/gpl.html) gives a similar recommendation.

johnyf, 2016-01-29:
Thanks for refining the suggestion to including a notice of license terms. I do think that this is useful, and have done so myself too in some cases, though trying it led to some observations:

- Unless split over multiple lines, the copyright notice widens an otherwise narrow source file.
- It seems more appropriate to place the notice outside the module docstring, i.e., as:
  
  ```python
  """Magic module."""
  # Copyright (BSD) 2015-2016 by California Institute of Technology
  import this
  ```
  
  However, there are some undesirable effects:
  - The copyright holder name can be quite wide (as above),
    so the name needs to be split over multiple lines.
  - There may be multiple copyright holders, requiring multiple lines (e.g., [`networkx`](https://github.com/networkx/networkx/blob/08338519a06227537a7778fe92322ef5445cc989/networkx/classes/digraph.py#L2)).
  - The year may need update commits (e.g., [`networkx`](https://github.com/networkx/networkx/commit/08338519a06227537a7778fe92322ef5445cc989)).
  - Upon [copyright expiry](http://copyright.cornell.edu/resources/publicdomain.cfm) (in the distant future), update commits are needed for all files.
  - Proving more information (BSD, link, etc.) can lead to additional lines.
- Multi-line comments require moving lower in the file to read code, whenever it is opened.
  (searching and relative moves within a text editing program can mitigate this)
- The comment below the docstring can be somewhat visually distracting.
- From a coding style perspective, the intermediate comment between docstring and imports is extraneous.
- [A sample of `numpy`](https://github.com/numpy/numpy/blob/75d5b59bca181ee7e5ba872999014006c4b6c3f3/numpy/linalg/linalg.py) does not contain any copyright notice.
- [Another sample of `numpy`](https://github.com/numpy/numpy/blob/75d5b59bca181ee7e5ba872999014006c4b6c3f3/numpy/distutils/cpuinfo.py#L5) does contain a copyright notice.
- Looking [at `scipy`](https://github.com/scipy/scipy/blob/72aa6c4d7531c5455eb49032d29e23e711c12d07/scipy/linalg/decomp.py#L2), files contain some form of local changelog.

I think that the biggest concern is carelessness when files excluding the `LICENSE` are redistributed.
Strictly speaking, a redistribution without the license violates its terms, and the same could be done by erasing the license text from the files, or copying selected lines from them.

Perhaps the `README.rst` could emphasize that the `LICENSE` should accompany the code or any subset of it, however small, and that one may copy the `LICENSE` within the files, if so desired.

slivingston, 2016-01-30:
Thanks for the thorough discussion. Your list of effects raises another concern: acknowledging authors and contributors. For the BSD and similar licenses, acknowledging the creators is the essential requirement. However, append-only author lists like that of the SciPy example that you give are difficult to maintain and not well motivated when tools like `git blame` can be used.

Two comments:

1. When included, I prefer that copyright and license text is placed before the docstring because it should be something that is first encountered by a reader of the source file, before documentation or the code itself.

2. Update commits for copyright year or expiry (I wish the latter was in the near future...) are relatively rare. They usually involve dedicated lines, and thus, history noise occurs at the level of commits but not at the level of lines of code. Nonetheless, simply omitting the year is OK.

Pending comments from @murrayrm or @necozay, I think that the action proposed at the end of your comment is appropriate.

slivingston, 2016-01-30:
In "the end of your comment", I am referring to "the README.rst could emphasize that..." [above](https://github.com/tulip-control/tulip-control/issues/141#issuecomment-176721876).

slivingston, 2016-01-30:
Regarding whether license text should be removed from existing source files, if agreement is not easily reached (or policies not easily determined), we could revisit the question shortly before the next release. Until then, new source files could use one of the new styles proposed above, e.g., copyright text with year and pointer to the license.

necozay, 2016-01-30:
For Michigan, I was told that it is sufficient to include the owner (the regents of the U of Michigan in my case) at the top of the BSD license FULL TEXT license file. Either option is fine for me as long as we keep the license file.

johnyf, 2016-01-30:
I think that the first (or second, in case of a shebang) line is a better location for placing the license notice. This separates visually the code from accompanying information and hints to the shell. Also, this is [the current](https://github.com/tulip-control/tulip-control/blob/70ffe5f3b8c60cf6e3145393e263354d78331ab7/tulip/synth.py#L1) practice within `tulip`, so the change would result in trimming the license text in each file, possibly adding a URL or note about `LICENSE` in the second line.

To be clear, I do not suggest that we edit all files to conform with this decision, rather let new files follow the new convention. Changes to reach a uniform style could be applied in sequence with other style-related maintenance at some unspecified future time.

murrayrm, 2016-01-30:
To my knowledge, there is no institutional rule at Caltech regarding this issue.  The changes suggested above sound fine.

slivingston, 2016-01-30:
As far as I can tell, this issue is resolved. @johnyf will you make the proposed change to the root README, or shall I? After that, we can close this issue.

johnyf, 2016-01-30:
Documented with 6a57e73866d88bf842177ce3780052ffdb634573 and cca964f97c5a7759d903751b1c52490bbfcd067d. An example of application is 96f9c289e16e077e0f9be69c7fa8ec3f4bee08ee.

Comments:

- did not explicitly require in `doc/dev_guide.rst` that the license should be linked to by a URL, so that a relative path remain a choice
- README considers modification and re-distribution of source code, not "use", because I noticed that under `site-packages` (where `python setup.py install` or `pip install` places `tulip`) the `LICENSE` is not present, so only the copyright notices within each file accompany the installed code.

closed 2016-01-30 by johnyf