tulip-control/tulip-control #142: branch management policy

johnyf, 2016-01-26:
Following private communication with @slivingston, we may want to select a default fate for branches that have been merged:

I think that deleting a branch associated with a pull request by default is a good practice, unless the branch has been explicitly protected by:

1. an email announcing so in the `tulip` mailing list
2. a comment in the relevant pull request
3. private communication
4. branch naming convention

While searching about No.4, I came across the use of slashes in branch names.
I think that the branch name need not be constrained, because it will just become longer.
Of course, if the branch author prefers a particular naming convention, e.g., `pr_*`, then they can choose to do so.

No.1 is probably the best practice, considering this as logistics extraneous to the PR comments.

slivingston, 2016-01-29:
I support the proposed policy. Because it concerns merged branches, restoring the branch is easy in cases of mistaken deletion.

johnyf, 2016-02-15:
Having recorded the discussion here, let us use this approach, until otherwise decided.

closed 2016-02-15 by johnyf