tulip-control/tulip-control #143: nx.to_pydot not working with networkx 1.11

miccol, 2016-02-19:
With the current version of networkx (1.11) I get the following error:

`AttributeError: 'module' object has no attribute 'to_pydot'.`

Apparently the developers of networkx have made some changes to the API when moving from 1.10 to 1.11. These changes, involve also the function `to_pydot` used by tulip. see: https://networkx.readthedocs.org/en/stable/reference/api_1.11.html

I have solved the issue downgrading networkx:

`pip install networkx==1.10`

johnyf, 2016-02-19:
Thanks for noting this change, a change compatible with both `networkx < 1.11` and `networkx >= 1.11` is here: [dd/780445c2e2fbc5f9cc43a55fe2dd34e2eb9e4a33](https://github.com/johnyf/dd/commit/780445c2e2fbc5f9cc43a55fe2dd34e2eb9e4a33).

johnyf, 2016-03-29:
Fixed in 503c507cd68465100f7f1971bdd589eff27774c7.

closed 2016-03-29 by johnyf