tulip-control/tulip-control #146: simulation part of continuous.py

necozay, 2016-06-04:
It would be nice to demonstrate how the synthesized controllers can be simulated in TuLiP in one of the examples. I have the following piece of code for the ``continuous.py`` but there are some hacky parts. Maybe we can clean it up and include in the next release. Any suggestions on how to make the initialization better?

```
# Simulation

print '\n Simulation starts \n'
T = 100;

# let us pick an environment signal
# from: http://code.activestate.com/recipes/577944-random-binary-list/

from random import *
randParkSignal = [randint(0,1) for b in range(1,T+1)]

# initialization is a bit hacky (need to be consistent with ctrl init
# states)

s0_part = ctrl[ctrl.edges('Sinit')[1][0]][ctrl.edges('Sinit')[1][1]][0]['loc']

init_poly_v = pc.extreme(disc_dynamics.ppp[s0_part][0])
x_init = sum(init_poly_v)/init_poly_v.shape[0]
x = [x_init[0]]
y = [x_init[1]]

N = disc_dynamics.disc_params['N']
s0_part = find_discrete_state([x[0],y[0]],disc_dynamics.ppp)
ctrl = synth.determinize_machine_init(ctrl, {'loc':s0_part})
(s, dum) = ctrl.reaction('Sinit', {'park':randParkSignal[0]})
print dum,'\n'
for i in range(0,T):
    (s, dum) = ctrl.reaction(s, {'park':randParkSignal[i]})
    u = get_input(
            np.array([x[i*N],y[i*N]]),
            sys_dyn,
            disc_dynamics,
            s0_part,
            disc_dynamics.ppp2ts.index(dum['loc']),
            mid_weight=100.0,
            test_result=False)
    for ind in range(N):
        s_now = np.dot(
                    sys_dyn.A, [x[-1],y[-1]]
                    ) + np.dot(sys_dyn.B,u[ind])
        x.append(s_now[0])
        y.append(s_now[1])
    
    s0_part = find_discrete_state([x[-1],y[-1]],disc_dynamics.ppp)
    s0_loc = disc_dynamics.ppp2ts[s0_part]
    print s0_loc, dum['loc']
    print dum,'\n'

show_traj = True
if show_traj:
    import matplotlib.pyplot as plt
    plt.plot(x)
    plt.plot(y)
    plt.show()
```

johnyf, 2016-06-04:
A comment that [`tulip.abstract.find_controller.find_discrete_state`](https://github.com/tulip-control/tulip-control/blob/f3a0f44e44b52aba1a2448492a556b3f98f58744/tulip/abstract/find_controller.py#L453) could also be named the mapping from concrete (continuous) to abstract (discrete) states, i.e., a [refinement mapping](http://research.microsoft.com/pubs/64046/abadi-existence.pdf).

johnyf, 2016-06-04:
This initialization touches the interplay between:

1. quantification of initial conditions, and
2. concretization for implementation

Depending on where we apply these methods, I can imagine that both can be possible real world scenarios:

1. we pick `Init` (initial predicate) and find the game realizable `\A x, y: Init(x, y)` (`\A` means `\forall`).

2. we find some abstract initial states that work (either with `\A x: \E y: Init(x, y)`, or `\E y: \A x: Init(x, y)`. `gr1c` and `omega` implement `\A \E`, but I think that `\E \A` physically makes more sense, so that we should change it). Then, we should place our physical system in one of those winning abstract states, and start execution from there. To do so, we need to concretize the the chosen abstract state, which is what is happening in the initialization given above, starting with the statement `init_poly_v = pc.extreme(disc_dynamics.ppp[s0_part][0])`.

The order of initialization would be, respectively:

1. Pick any continuous state in `Init` (since `Init => WinningSet`) and map it with the refinement mapping `find_discrete_state` to the corresponding abstract state `s0_part`. This is done anyway inside the loop, so it would amount to shifting the call to `find_discrete_state` to before `get_input` within the loop, and not calling `find_discrete_state` at all outside the loop.

2. Pick an abstract state in the discrete winning states, then concretize it, then place the plant there, then start executing. This is done above.

I would suggest linearizing the thought flow, by turning case 2 into 1. To do so, we need a function that will take the abstract winning states and the partition, and return a (nonconvex) polytope of winning concrete states (the concrete initial condition from where we can win). This is the union of polytopes that are winning abstract states.

We can then pick (as in "give me one satisfying assignment", which is done with the lines
```python
x_init = sum(init_poly_v)/init_poly_v.shape[0]
x = [x_init[0]]
y = [x_init[1]]
```
above) one point in this nonconvex polytope, by calling a function called `sat_iter` or something similar, in analogy to BDD APIs) to get a satisfying assignment for the coordinates.
We can then run the loop, with the refinement mapping called first, as mentioned earlier.
It is essentially the method of *picking* a satisfying assignment that models whether we or someone else picked the initial condition.

The other thing that would make the code simpler is to replace `ctrl` (the machine of an input/output formalism) with a state-space approach of using directly the next-state action of the strategy (transition relation).

This would require that we initialize all (discrete) variables ourselves, instead of hiding them inside `ctrl`. It can be done by simply calling a generator version of the first-order enumeration [`omega.symbolic.enumeration._enumerate_bdd`](https://github.com/johnyf/omega/blob/master/omega/symbolic/enumeration.py#L166). I am about to release a more comprehensive first-order wrapper for BDDs, so I should add a `sat_iter` that does this.

What confuses me a little bit in the code above is why `s0_part` is re-initialized. Shouldn't it be the same? (i.e., shouldn't `assert s0_part == find_discrete_state([x[0],y[0]],disc_dynamics.ppp)` pass?)

necozay, 2016-06-04:
As a side note, this simulation only works if in abstraction ``closed_loop=False`` and ``conservative=True`` flags are used.