tulip-control/tulip-control #152: Specs tagged as unrealizable when gr1c exits on segmentation fault

iincer, 2016-07-03:
Tulip checks for the return code of gr1c in order to determine whether an implementation is realizable. gr1c returns code 0 if the spec is realizable; unix adds 128 to the return code of a program that exits on segmentation fault, so tulip gets something different from zero when gr1c crashes; this means that tulip will always tag a specification as unrealizable if gr1c exits after segmentation fault. Tulip should say that the realizability test failed when gr1c crashes.

johnyf, 2016-07-03:
Thanks for reporting this behavior. Can you please provide some details about the following:

- Does this arise when calling the function [`tulip.interfaces.gr1c.check_realizable`](https://github.com/tulip-control/tulip-control/blob/c72abfe57f5b72a97bd0b0d023d7a7c0f5caffa3/tulip/interfaces/gr1c.py#L424)?
  I expect that if `gr1c` crashed in the function `synthesize`, then an exception would be raised upon attempting to [parse output from `gr1c`](https://github.com/tulip-control/tulip-control/blob/c72abfe57f5b72a97bd0b0d023d7a7c0f5caffa3/tulip/interfaces/gr1c.py#L513).

- What causes `gr1c` to crash? Is it malformed input, so an error in `tulip` output to `gr1c`, an error in the specification written by the user (so justified for `gr1c` to raise an error), or an error in `gr1c` that needs to be reported in [this issue tracker](https://github.com/slivingston/gr1c)? A segmentation fault likely is the last case, since otherwise `gr1c` would exit with an assertion error.

johnyf, 2016-07-03:
Also, is the addition of 128 by unix documented somewhere? [From this discussion](http://stackoverflow.com/a/18731911/1959808) it appears to be system-dependent.

iincer, 2016-07-04:
My pleasure. Regarding the questions...
- Correct. This is the function under which the crash occurs. A subprocess is created to run gr1c, and tulip waits for this subprocess to end execution. Note, however, that even if gr1c crashes, you will not get any exception under tulip; tulip will always see a return value from gr1c, even if it crashes: you will simply get the return code of the crashed gr1c.
- The crash is caused by testing realizability of a specification that uses all my memory. I found it because I created a parametrizable specification which I expected to be always realizable, but I was puzzled that for some values of my parameters, tulip was telling me that my specification was not; looking further, I noticed that when asking tulip for realizability, gr1c was spawned and attempted to use a lot of memory; eventually, gr1c would exit, and tulip would say non realizable. This happens because when the gr1c subprocess exits on segmentation fault, the OS will provide a nonzero return value, and tulip's test for realizability is whether or not gr1c returned a zero (l 424-425 of the function you linked).
- The comment about the 128 adder was based on [this discussion](http://stackoverflow.com/questions/14599670/what-error-code-does-a-process-that-segfaults-return).

This suggests a comment: I was surprised gr1c returns nonzero values to report the status of the specification. I see this status as the 'data' of the program, what the program does. In my experience, the return value of a program is used to indicate whether there were problems with the program itself (whether it entered into an erroneous condition, for instance). If the mechanism gr1c used to report anything regarding the specification (e.g. whether or not it is realizable) did not used main() return values, then any nonzero return code would mean a problem happened with the program itself. I am aware, however, that this comment may mean a fairly decent amount of refactoring, but perhaps it may be considered in another version of gr1c.

johnyf, 2016-07-04:
For reference, the exit codes of `gr1c` are documented [here](http://slivingston.github.io/gr1c/index.html). I am not the author of `gr1c`, so I cannot comment on this design decision. @slivingston may have comments on it.

Regarding segmentation faults due to insufficient memory, this is likely due to CUDD behavior, as discussed here: https://github.com/slivingston/gr1c/issues/23.

About memory limits, the synthesizer in the package [`omega`](https://github.com/johnyf/omega), interfaced to via [`tulip.interfaces.omega`](https://github.com/tulip-control/tulip-control/blob/c72abfe57f5b72a97bd0b0d023d7a7c0f5caffa3/tulip/interfaces/omega.py) and available in `synth.synthesize` as option `'omega'`, can be modified to call `dd.cudd.BDD.configure` to [set max memory for CUDD](https://github.com/johnyf/dd/blob/94bc402664fec5ab2f2c76459db598d385b996a7/dd/cudd.pyx#L468). This may again lead to a segmentation fault, though it should protect against the machine becoming unresponsive.

iincer, 2016-07-04:
Many thanks for suggesting omega and for the references. I have only used the gr1c support so far.
I looked at the implementation of support for other engines, like ltl2a, and it seems these will throw an exception when a nonzero code is received from the executable. From what I have seen, it is only for gr1c that tulip will sometimes report that some specs are not realizable when they happen to be realizable (and it will do so because gr1c will exit with a nonzero code).
Please let me know if additional information would be helpful in addressing this issue.

slivingston, 2016-07-10:
@iincer Thanks for your comments about interpretation of exit codes. I created a feature request (https://github.com/slivingston/gr1c/issues/26) in the issue tracker of gr1c about it, where we can continue discussion.

Regarding the behavior of TuLiP, I will create a workaround that we can use until exit codes from gr1c are changed.