tulip-control/tulip-control #154: run using scipy instead of cvxopt

johnyf, 2016-07-28:
Likewise to `polytope`, which can now run w/o `cvxopt`, `tulip.abstract.find_controller` should [not import `cvxopt` only](https://github.com/tulip-control/tulip-control/blob/9aa793a473aeb5f4a656e2fe45112cbea83becda/tulip/abstract/find_controller.py#L51). This is the only place where `cvxopt` is used. Due to `tulip.abstract.__init__`, it causes failure of discretization too, whenever `cvxopt` is absent.

necozay, 2016-07-29:
Just to make sure we do not replicate work: This is on my agenda for UM TuLiP working meeting tomorrow. @johnyf if you have already started working on it, let me know. Otherwise, if we cannot get to this, I can leave it to you.

johnyf, 2016-07-29:
Thanks, I haven't yet touched this, please go ahead. The only place that `grep -ilr "cvxopt" ./*` shows modifications are needed is [`tulip.abstract.find_controller`](https://github.com/tulip-control/tulip-control/blob/fc5d04fdcac9dab2104802a77f5a2c718c7e6bf3/tulip/abstract/find_controller.py#L51).

#131 is relevant to thoroughly test this, both with:

- `scipy` present but `cvxopt` absent, and
- `scipy` present and `cvxopt` present *and* linked to `glpk`

For an example, this has already been done in [`polytope`'s `.travis.yml` on branch `dev`](https://github.com/tulip-control/polytope/blob/065f7699ebb9c4b10a520647d138e850b12b80f2/.travis.yml#L39).

johnyf, 2016-07-29:
For performance reasons, we probably want to run twice (one with `scipy` and once with `cvxopt.glpk`) only a selected subset of tests that is representative of abstraction and continuous feedback control functionality.

necozay, 2016-07-31:
@sahiny is looking into this.

johnyf, 2016-07-31:
Thanks for the note. @sahiny, please let us know in case there is anything we can help with / answer (both here and over email work for me).

slivingston, 2016-07-31:
If I understand this issue correctly, it does not necessarily involve changing the Travis CI script.  I assigned #131 to myself, so I can ensure that tests touch `scipy` and `cvxopt`.

johnyf, 2016-07-31:
Running all the tests has been done on branch `enumeration` (though they will need `omega >= 0.0.9` to pass on Travis CI. Running with both `scipy` and `cvxopt` has not been done.

Please see:

https://github.com/tulip-control/tulip-control/commit/ce91fa489dbc2a3726c03abaeb2de17a4fc6816b

slivingston, 2016-07-31:
If you are working on an issue, please assign it to yourself or announce it on the mailing list. In this case, perhaps you should assign #131 to yourself.

johnyf, 2016-08-06:
I have assigned that issue to myself.

johnyf, 2016-08-06:
Any progress on this issue? `find_controller` uses quadratic programming, so is [`scipy.optimize.minimize(method='SLSQP', ...)`](http://docs.scipy.org/doc/scipy/reference/optimize.minimize-slsqp.html#optimize-minimize-slsqp) the replacement?

johnyf, 2016-08-06:
Also, running `abstract_test.py` without `cvxopt.glpk` on Travis CI [times out](https://travis-ci.org/tulip-control/tulip-control/builds/150232588). So, we may want to create a test configuration that runs less demanding tests using `scipy`, run that with `scipy`, and then install `cvxopt.glpk` and run the entire test suite for supported features (omits `jtlv`, `lily`, `slugs`).

sahiny, 2016-08-06:
Hi Ioannis,

I was traveling to Turkey, so I couldn’t have time to replace the QP used in find_controller to LP. I’m working on it right now. 
—
Yunus Emre

johnyf, 2016-08-07:
@sahiny I was asking mainly to learn whether another QP solver will be used, and if that will be from `scipy`. Please take your time, I am not suggesting a time constraint.

sahiny, 2016-08-07:
As far as I have been told, I am going to change QP in ‘get_input’ with LP from spicy. But I am running into weird issues with my Python, which prevents me from actually trying my code. I will keep you posted with any updates.

Thanks,
—
Yunus Emre

necozay, 2016-08-07:
As far as I know, there is no QP solver in `scipy`. What we planned to do was to replace "weighted 2-norms" (leading to a quadratic objective function) with "weighted infinity or 1-norms" to obtain a linear objective function. That way, we can mimic the "get close to center" type objective as well - just with a different norm.

slivingston, 2016-10-28:
Is there a proposed signature for `get_input()` that would be introduced while addressing this issue?

Currently (as of commit 8955f63d1d1bbd3a96f469718dde49454b275fd5), `get_input()` has parameters that are weight matrices for use in the quadratic objective function. [The comment above](https://github.com/tulip-control/tulip-control/issues/154#issuecomment-238056016) proposes using weighted \infty-norms or 1-norms, which would imply a different interpretation of the parameters that provide weights to `get_input()`.

One idea is to add the parameter `ord` that can take values from a subset of the possible values of the `ord` parameter of [numpy.linalg.norm](https://docs.scipy.org/doc/numpy/reference/generated/numpy.linalg.norm.html#numpy.linalg.norm). The default would be `ord=2`, which indicates the quadratic norm as already implemented.

necozay, 2016-10-28:
@sahiny is fixing some of the bugs in the branch `find_controller_LP`. We are thinking about making minimal changes to the function signature, with only an additional parameter for the type of norm used. Using `ord` is a good idea. BTW, this is a good time to let us know if you have any suggestions.

The proposed signature will slightly change the meaning for the weight matrices `R`and `Q` when `ord=2`. In the current version, we minimize `x^TRx`; in the new version, we will minimize `||Rx||_2`. So, to get the same result when `R = K^TK` in the old version, one needs to set `R=K` in the new version. 

We plan to have `ord=1` as default, and `ord=inf` will also be implemented as these both lead to LPs. We will leave `ord=2` as unimplemented with a proper error message as this leads to a QP and `scipy` (as far as I know) does not have a solver for constrained convex QPs. It is also possible to check availability of a `cvxopt` QP solver and have this option available though it will complicate the dependencies.

slivingston, 2016-10-29:
Is there any way that I can support the effort? Besides code review, I can debug and contribute missing pieces.

necozay, 2016-11-05:
@slivingston `find_controller_LP` branch currently implements the base functionality without `cvxopt`. Feel free to play/review/debug. As far as I can tell what is missing is a correct implementation of `get_input` for `closed_loop=True`. Note that there is a bug in the earlier implementations (see my comments [here](https://github.com/tulip-control/tulip-control/issues/162#issuecomment-239633913) ). Do you want to contribute to that part? @sahiny said he could look into that as well. Maybe if you can look through the code, you might be able to locate other missing pieces and we can coordinate accordingly. I can create a pull request for this version or wait until we fixed the closed loop control input extraction.

slivingston, 2016-11-17:
@necozay I will read through your recent pull request and provide additional review, in case it is still useful. I noticed that @johnyf [already indicated that it is ready to merge](https://github.com/tulip-control/tulip-control/pull/168#issuecomment-261222293), so I do not want to cause delays.

johnyf, 2016-11-18:
The above changes have been implemented and merged in #168, with an `ord` parameter, and the case of quadratic cost as an available option (given `ord = 2`), provided `cvxopt` is present. The implementation of `closed_loop = True` remains, but is beyond the scope of the OP.

closed 2016-11-18 by johnyf