tulip-control/tulip-control #157: swap arguments `option, specs` to `synth.synthesize`

johnyf, 2016-07-29:
The function `synthesize` operates on specifications, not on tools. The tool is just an option, that, albeit important, should appear later. Moreover, since we will have a pure Python solver, no external dependencies are needed (compiler, CUDD, `java`, etc.), which means that this option is not required any more. It should be a keyword argument. This will allow code to run generically, without updating the examples every time we decide on a different default solver (last time this happened was in 1d465f76ea1907462d51444000adfa6a8d60b91c).

Currently, there is no default solver in `synth.synthesizer`, because `option` is a positional argument.
Same arguments apply to the function `is_realizable`.

johnyf, 2016-07-29:
Moreover, we can go one step further, and consider adding support for a user config file in the future, e.g., under `~/.tulip_config` that stores preferences like default solver. An alternative is to cycle through all solvers, until one is found (I don't support this alternative: explicit is better than implicit).

slivingston, 2016-07-30:
One of the original motivations to include an option string was to allow cases that were beyond particular tools, e.g., including `'strict'` as a suffix or prefix to indicate the notion of realizability to use.

johnyf, 2016-07-30:
This role has now been assigned to attributes of `GRSpec`, to allow the information to easily propagate down the call graph that leads to synthesis, without having to modify the arguments of intermediate functions (this also allows for differing attributes recognized by each backend, with the same middle layer.

In principle, one could incorporate the selected solver as one more attribute of `GRSpec`. However, it is related only to implementation details, and not a property of the entity represented by `GRSpec` (whereas most or all of the other attributes represent some part of the mathematical definition).

slivingston, 2016-07-30:
While I agree with the ambitions of the proposed change, I want to discuss it further without risking the delay of releasing 1.3.0. Can we change the milestone?

johnyf, 2016-07-30:
I have not applied this change yet. We could change the milestone, though my thinking was that since 1.3.0 breaks the API in this area sufficiently that user code will have to change, it might as well include this change too.

johnyf, 2016-11-26:
As an action item, I propose changing the signature of function `synth.synthesize` from:

```python
synthesize(
    option, specs, env=None, sys=None,
    ignore_env_init=False, ignore_sys_init=False,
    rm_deadends=True)
```

(in 02a407765f2f4ca995a6691587e8170fdbd385e6) to:

```python
synthesize(
    specs, option='omega', env=None, sys=None,
    ignore_env_init=False, ignore_sys_init=False,
    rm_deadends=True)
```

(If only Python 3 was supported, then I would suggest keyword-only arguments.)
Perhaps the keyword could also be changed, to `solver`.
This will lead to a number of changes, mainly to examples, as commented in the OP.

Are there any objections?

slivingston, 2016-11-27:
I support the proposed change.

Regarding whether to change the keyword `option`, I originally imagined that it would be an argument that allows sophisticated configurations to be expressed, e.g., `option='omega-sylvan'` to use `omega` and `dd.sylvan`. By defining configuration abbreviations, we can avoid adding more function parameters and thus make call signatures easier to read. If we eventually do this, then `option` is more appropriate than `solver` as the keyword. To be explicit, I sketch this feature only to motivate keeping `option`, not to add more tasks for this issue.

necozay, 2016-11-27:
Not a very strong objection but just thought I would write it anyway. I think `synthesize` operates on `spec`, `sys` and `env` (not only on the specification). It can operate only on the `spec` but the main usage in tulip examples is specification plus system. Having `option` in between these looks a bit odd to me. For comparison:

```
ctrl = synth.synthesize('omega', specs, sys=sys)
ctrl = synth.synthesize(specs, 'omega', sys=sys)

ctrl = synth.synthesize('omega', specs, sys=sys_swe, ignore_sys_init=True)
ctrl = synth.synthesize(specs, 'omega',  sys=sys_swe, ignore_sys_init=True)
```
Would it make sense to move it to a later spot?

necozay, 2016-11-27:
I just realized `omega` will be default in what is suggested above so in the examples it would be ignored anyways. Then, it will be fine.

johnyf, 2016-11-27:
I totally agree with the [suggestion](https://github.com/tulip-control/tulip-control/issues/157#issuecomment-263140115) to move `option='omega'` to the end, or at least after `env` and `sys`.

This is the case also in the signature of the function [`synth.synthesize_many`](https://github.com/tulip-control/tulip-control/blob/02a407765f2f4ca995a6691587e8170fdbd385e6/tulip/synth.py#L927):

```python
def synthesize_many(specs, ts=None, ignore_init=None,
                    bool_actions=None, solver='gr1c')
```

I support placing it at the end. The reason behind this thinking is similar to why I prefer to pass symbolic automata and BDDs as the last argument (not uniformly so everywhere, because at the time of first writing of the code I had not fixed this decision, but will gradually impose uniformity). They are the "context" (contain symbol tables etc.), so the same ("least active") argument that propagates down the call graph. Placing them earlier only creates distraction.

Having said this, one reason I placed it earlier was to make the change appear smaller, but that doesn't help, because user code will still break. So, if a user updates their code, it shouldn't make much of a difference whether it is the first or last keyword argument.

In more detail, the difference it makes is to force the user to use a keyword argument (because, most likely, they aren't passing all the keyword arguments). Thus, the added benefit of placing `option` at the end is that it enforces keyword-only arguments in practice (unless one passes `env` and `sys` as positional arguments, but that conceptually that should be fine).

In Python 3 (only), if [keyword-only arguments](https://www.python.org/dev/peps/pep-3102/) were used, there wouldn't be any difference other than in reading convenience. The syntax to define a function signature with keyword-only arguments is:

```python
synthesize(
    specs, *, option='omega', env=None, sys=None,
    ignore_env_init=False, ignore_sys_init=False,
    rm_deadends=True)
```

johnyf, 2017-04-03:
As of cb6aa6f223c205e3fd50a2560d980a6bf7c1fa6b I have applied the changes discussed above. The API change is in 43d4ceba808398034782fea920b41955429234c1. I removed the (first positional) argument `option` from the functions `tulip.synth.synthesize` and `tulip.synth.is_realizable` and added the keyword argument `solver` last in the signature. I used "solver" because I think it is a little more self-explanatory for unfamiliar users than "option". We could change this before merging or in the future.

I would like to merge these changes directly into branch `master` after 5 days, unless any objections are expressed.

slivingston, 2017-04-04:
@johnyf I read through the changes, i.e., commits 51cd8cd11a2709d2702df667171f638850544ba2 - cb6aa6f223c205e3fd50a2560d980a6bf7c1fa6b (branch `swap_synth_args`). They are good. I think that waiting 1 more day for comments is sufficient.

necozay, 2017-04-04:
I like the change in API.

johnyf, 2017-04-06:
Merged as ce72e782ae38bfb2f5119e649341ebb4255038b5.

closed 2017-04-06 by johnyf