tulip-control/tulip-control #158: deprecate and eventually delete `interfaces.jtlv`

slivingston, 2016-07-29:
As far as I know, there are no current users of the JTLV-based solver interface, and no one wants to maintain it because the performance, features, etc. are available and better elsewhere. As such, we should deprecate and, later, delete `interfaces.jtlv` and any other uses of the JTLV-based solver in TuLiP.

Anyone interested in keeping it should comment here. Besides technical arguments about using the solver itself, other motivations can include the existence of legacy code and interesting use-cases that indirectly involve JTLV.

johnyf, 2016-08-29:
More specifically:

1. in v1.3.0 we can [show a `DeprecationWarning`](https://docs.python.org/3/library/warnings.html#warnings.warn) if a "public" function from `tulip.interfaces.jtlv` is called, and
2. in v1.4.0:
    - remove the code that copies `jtlv` from `setup.py`
    - move `tulip.interfaces.jtlv` under `contrib/`
    - move `extern/get-jtlv.sh` user `contrib/`
    - remove mention of `jtlv` from user documentation and docstrings

johnyf, 2016-11-26:
As of 02a407765f2f4ca995a6691587e8170fdbd385e6, JTLV is already declared as deprecated in the docstring of `synth.synthesize`. I think that the simplest next action is to move it under contrib, or remove it. I've made these changes on branch `rm_jtlv` (moving it under `contrib/` for now).

The syntax of `jtlv` is supported also within `tulip.spec`. There isn't any significant reason for removing that too, but it should be considered unmaintained (that it may be removed without notice in the future). This raises the question of whether it would be better to delete the `jtlv` interface module, instead of moving it under `contrib/`.

The motivation to delete is that code under `contrib/` won't work without syntax support from within `tulip`, so the code under `contrib` will become an anchor for whether to touch `jtlv` syntax in `spec` in the future or not, which is inconvenient (not that all the other code under `contrib/` works, so perhaps this isn't implicitly implied).

slivingston, 2016-11-27:
I imagine two use-cases where the interface with the JTLV solver is required:

1. running legacy code (i.e., old, mostly unmaintained source code that is worth keeping) that might or might not depend on an old version of TuLiP;

2. using the JTLV solver because it has some unique and valuable features that we failed to notice when we decided to remove it.

For case 1, it can be useful to have some indication in current releases that there was support for the JTLV solver in the past. This could be within the documentation, e.g., a new page that offers a brief history of TuLiP. Or, this could be as unmaintained code under `contrib/`. In any case, it should suffice for the user in case 1 to try an old release of TuLiP before deprecation of `interfaces.jtlv`. For this use-case, I think that there is little or no motivation to keep the code under `contrib/`.

For case 2, having the code under `contrib/` so that someone can immediately begin to hack on it can provide some motivation for resurrecting `interfaces.jtlv`. Even though the code as under `contrib/` might be broken, it serves as a substantial hint about where to begin, and it can be found without requiring the user of case 2 to dig through commit history.

With the above cases in mind, I think that the main question is how likely we would want to reinvigorate `interfaces.jtlv`.

johnyf, 2016-11-27:
I agree with the conclusion for case 1 [above](https://github.com/tulip-control/tulip-control/issues/158#issuecomment-263104092), namely that using a consistent snapshot from the past (a version of `tulip` that corresponds to the legacy code and data) should be the preferred approach.

This will always work, whereas trying to run something from `contrib/` can easily prove to be a trap for losing time (i.e., it may look simpler, or even the presence of `contrib/jtlv` may be understood to suggest that the interface remains compatible with `tulip`). Moreover, using an older version shouldn't be disruptive, because the user can create a virtual environment for that purpose.

I think that it is unlikely that using some valuable feature of `jtlv` will be sufficient reason to reintroduce it. Instead, it will be easier, more dependable, instructive, and maintainable (and probably require about the same amount of time) to reimplement those features in Python.

Taking these observations into account, it seems that the conclusion is to delete the `jtlv` interface.

johnyf, 2016-11-27:
Also, I agree that, for documentation purposes, a note should point to where the interface was, for anyone that needs to find it. I think that a history page in the documentation may become an extra thing to maintain. The CHANGELOG is, in a way, the history page of the documentation, so we could clearly note in there the last commit hash and `tulip` version that contained the interface.

murrayrm, 2016-11-27:
Mainly out of curiosity: if we delete the JTLV interface and someone were to want to use JTLV as a "custom" solver, could they do it?  Ie, how hard would it be to create a "user interface" to a solver?

I don't see any particular reason to keep the JTLV interface.  The tool is not being maintained (right?) and nobody is currently using it.

slivingston, 2016-11-28:
re https://github.com/tulip-control/tulip-control/issues/158#issuecomment-263105684 I agree that the CHANGELOG suffices as the (ongoing) history page.

slivingston, 2016-11-28:
Unless I missed something above, the conclusion in terms of actions is the following:

* delete `tulip.interfaces.jtlv`;
* make a note of the deletion and the corresponding commit hash in the CHANGELOG.

The translation code for `jtlv` will be kept as it currently is.

johnyf, 2016-11-28:
Indeed. I will complete these changes on branch `rm_jtlv` and open a pull request.

PR #177 (2017-03-06): Rm jtlv
Closes #158.
closed 2017-03-06 by slivingston