tulip-control/tulip-control #159: distinguish between formulae and synthesis problems

johnyf, 2016-07-29:
Notionally or in code too. The issue arises when a graph-based state machine description (transition system) is translated to an assume-guarantee GR(1) formula. This is a just a collection of predicates `EnvInit, EnvNext, EnvRecur_j` and `SysInit, SysNext, SysRecur_i`, to be used for creating a temporal formula. It doesn't say any of:

1. what capabilities the strategy has (`GRSpec.moore`)
2. how strict the implication is (`GRSpec.plus_one`)
3. how initial conditions are quantified (`GRSpec.qinit`)

On the other hand, `GRSpec.__or__` raises an exception if its operands disagree on any of the above attributes (otherwise we'll mix mismatched definitions of synthesis(.

Although explicit, it is somewhat impractical to require that an extra arg to `synth.synthesize` define the above for the spec resulting from the graph. It would also *necessarily* duplicate the existing choices in `spec` (otherwise an `Exception` would be raised). So, there is little point in requiring more than one definition of these attributes. There's only one per synthesis problem.

In addition, it would be notionally wrong to add such attributes to a graph-based description of a state machine. That's just a pair of predicates (`Init, Next`), to be used as deemed appropriate (for a state machine in verification, to assemble A/G properties, etc.).

So, my choice is to use the attributes of the `spec` given to `synth.synthesize` to define (more precisely, overwrite the defaults from instantiation of `GRSpec`) these attributes for the `GRSpec` returned by functions `synth.sys_to_spec` and `synth.env_to_spec`.
Please let me know if there are any objections to the above thinking.

Sidenote: in the future, when `omega.automata` and `omega.symbolic.logicizer` will start serving, and eventually replacing, most of `tulip.transys` and the top (long) part of `tulip.synth`, the above two concerns will be more clearly separated, because synthesis interfaces and flattening of graphs to logic will live in different modules.

slivingston, 2016-08-21:
I understand that this is closed via PR #163. However, this issue is discussion-oriented. @johnyf did you want to keep this open?

johnyf, 2016-08-21:
I think that this can be closed, because it was implemented in 3208fc1c69787ec45d1e17afaf89a3f0d726ba06 and 596599edb07fe4c7b5f3fbd70e046aa69be4b900.

closed 2016-08-21 by johnyf