tulip-control/tulip-control #166: test computation of controller input

johnyf, 2016-09-02:
This refers to `tulip.abstract.find_controller`. If needed, rewrite. It is labeled as a bug because it has been reported that it does not match what abstraction says.

necozay, 2016-09-02:
`find_controller` needs some adjustments to make sure it is consistent with the latest changes in the flags of abstraction (`use_all_horizon`, `closed_loop`, etc.).

But other than that, do we have an example demonstrating a bug? I heard people complaining but nobody shared a concrete example and I could not find a buggy example myself.

johnyf, 2016-09-02:
There is an example that I do not yet have myself, but when I obtain it and investigate, I will try to create a minimal example. The problem reported is that abstraction and synthesis succeed, and a controller is found for horizon length `N = 1`, but increasing to `N > 1` again abstraction and synthesis succeed, but no controller can be found. Unless this is not due to parameter mismatch (which I have not yet confirmed), it must be something else (either missing implementation, or bug in existing).

slivingston, 2016-09-03:
Should this be targeted after release of version 1.3.0 ?

johnyf, 2016-09-04:
Yes, I did not intend to suggest otherwise.