tulip-control/tulip-control #167: Option for DFA minimization post enumeration

dathath, 2016-09-23:
The enumerated strategy is generated sometimes with multiple nodes that accept the same string of inputs. Implementing a standard DFA minimization algorithm (eg. Hopcraft's algorithm) would make the enumerated possibly more human parse-able and definitely smaller in size. 

Reference: Hopcroft, Motwani & Ullman (2001) - See Section 4.4.3, "Minimization of DFA's".

johnyf, 2016-09-23:
[`FAdo`](https://pypi.python.org/pypi/FAdo/1.3.2) is a package that claims to increment Hopcroft's minimization algorithm. This package violates PEP8 and its repository appears to not be where advertised, so I do not recommend using it as a dependency. Nonetheless, it can still be used to evaluate the OP in practice.

If the result is as desired, then it would be reasonable to first ask the `networkx` developers whether they would consider this algorithm as fitting in `networkx`. If yes, then contributing it to `networkx` would make it accessible in a broader context, and it can be called by a suitable function here, in order to further manipulate a strategy after it is enumerated. Otherwise, I would categorize this algorithm under `transys`.

Flat is better than nested, so it is probably better for the user to explicitly make a call to apply an optimization.

johnyf, 2016-09-23:
Also, `omega.games.enumeration` tries to minimize the number of states while enumerating: it prefers edges that lead to states that have already been enumerated, and computes those symbolically. I believe that `gr1c` applies a similar optimization, cf. #150.

Thematically relevant is Sec.4.1: Minimizing the Strategy in [Piterman, Pnueli, Sa'ar, "Synthesis of reactive designs"](http://ysaar.net/data/synth.pdf).

necozay, 2016-09-29:
The question of whether it is possible to "minimize" the enumerated strategy also came up at the tutorial session (from someone who used TuLiP and obtained very large controllers).

slivingston, 2016-11-26:
In case it is useful to reference, I found <http://fado.dcc.fc.up.pt/>, which appears to be the homepage of FAdo, rather than the invalid URL, http://fado.fc.up.pt/, that is listed on [the current FAdo PyPI page](https://pypi.python.org/pypi/FAdo/1.3.2).

slivingston, 2016-11-26:
In addition to violations of PEP8 [as mentioned above](https://github.com/tulip-control/tulip-control/issues/167#issuecomment-249082649), I think that a critical issue in using FAdo as a dependency is that it lacks support for Python 3. Futhermore, there are no tests included in the current release tarball. However, the API documentation is fairly extensive.

I sent an email to the authors asking about the repository of FAdo.

slivingston, 2016-12-01:
I received a response from Nelma and Rogério. (I am happy to forward it to anyone privately.) In summary, the Bitbucket repo URL that is listed on PyPI is correct but is private to their research group. They welcome contributions, suggestions and bug reports, which we can email them directly.