tulip-control/tulip-control #169: consider using `nose2` instead of `nose`

johnyf, 2016-11-13:
`nose` is unmaintained, and its earlier maintainer recommends moving on to use `nose2` (see https://github.com/nose-devs/nose/pull/1004#issuecomment-260155457).

slivingston, 2016-11-18:
The homepage of `nose` lists several alternatives:
> New projects should consider using Nose2, py.test, or just plain unittest/unittest2.

quoting from http://nose.readthedocs.io/en/latest/
where links are:
* nose2: https://github.com/nose-devs/nose2
* py.test: http://pytest.org/

johnyf, 2016-11-27:
`pytest` seems worth considering, it appears to be much more actively developed, and more widely used.

slivingston, 2019-02-15:
concerning differences between `nose` and `nose2`: https://nose2.readthedocs.io/en/latest/differences.html

slivingston, 2019-04-22:
In the tulip-control project, this issue also applies to

- [ ] [polytope](https://github.com/tulip-control/polytope)
- [ ] [dd](https://github.com/tulip-control/dd)

slivingston, 2019-05-22:
I recommend pytest and will move forward with it. This might be useful for legacy code using `nose`: https://docs.pytest.org/en/latest/nose.html

johnyf, 2021-06-03:
`nose` is not compatible with Python 3.10 (due to imports from the module [`collections`](https://docs.python.org/3.9/library/collections.html#module-collections) of classes [that have moved](https://docs.python.org/3.10/whatsnew/3.10.html#removed) to the module [`collections.abc`](https://docs.python.org/3.10/library/collections.abc.html#collections-abstract-base-classes)). I recently (in a local repository) transitioned `dd` from `nose` to `pytest`, which informed me about its usage. I would be happy to transition `tulip` to `pytest`. I also transitioned to `pytest` the packages `polytope` (on branch `py310`) and `omega` (in a local repository).

murrayrm, 2021-06-06:
Using pytest seems like a good path forward.

johnyf, 2021-06-06:
I will start the transition to using `pytest`.

johnyf, 2021-06-11:
The transition to `pytest` has been completed on branch `use_pytest`, not yet merged into the mainline branch. I also transitioned CI testing to GitHub Actions, as described in #247, because Travis CI is no more free.

So the branch `use_pytest` is I think nearly ready to be rebased and merged. I intend to wait a week or more, before merging this branch, in order to ensure that I have not forgotten anything.

## Some remarks

There are various ways in which `pytest` is inequivalent to `nose`, including a number of items that `pytest`, as far as I know, does completely differently.

### Where generated files are written

For example, with `nose` it is possible to run `nosetools` programmatically from a Python script above the directory `tests/` and have the files generated by tests be within the directory `tests/`. I could not find any way of configuring `pytest` to do that, so I programmatically change directory to `tests/` (depending on command-line options given to the script `run_tests.py`). I arranged this particular change as a separate commit, in case `pytest` changes in the future, and there is interest in the previous implementation.

Switching to the directory `tests/` also ensures that only the installed copy of `tulip` is importable, i.e., it makes it impossible to import `tulip` from its source directory within the repository (well, assuming that path is not added to `sys.path` by Python code). Though this is possible to configure in `pytest`, it is an extra layer of safety. (In other packages, I have been `cd tests/` before running tests). This way, even if something in `pytest` changes, importing will only be possible from the installed copy of `tulip` under `site-packages`.

### Interference between test modules

Another observation (and there are several others), which actually does not apply to `tulip`, but was observed for `dd`, is that `pytest` runs tests in a way that separate Python test modules can interact.

For example, what happened in `dd` was that a number of test modules imported common test modules, and set class variables. With `nose`, this worked without interference between the two modules. With `pytest`, the imported modules are shared between the importing modules, which results in the class variables having the same value in the importing test modules.

In more detail, the test modules were:

```
tests
├── autoref_test.py  # imported modules `common`, `common_bdd`, and
    # set the class variables `common.Tests.DD` and `common_bdd.Tests.DD`
    # to the class `dd.autoref.BDD`
├── common.py
├── common_bdd.py
├── common_cudd.py
├── cudd_test.py  # imported modules `common`, `common_bdd`, `common_cudd`, and
    # set the class variables `common.Tests.DD`, `common_bdd.Tests.DD`, `common_cudd.Tests.DD`
    # to the class `dd.cudd.BDD`
├── cudd_zdd_test.py  # imported modules `common`, `common_cudd`, and
    # set the class variables `common.Tests.DD`, `common_cudd.Tests.DD`
    # to the class `dd.cudd_zdd.ZDD`
```

This was an issue in `pytest`, because that class variable determined which class was being tested. Thus, some of the test modules were actually testing the class of another `dd` module than what they were supposed to be testing. I noticed this by coincidence, when trying to fix something else while transitioning from `nose` to `pytest`.

To correct this issue, I turned the class attributes to [data attributes](https://docs.python.org/3/tutorial/classes.html#instance-objects) (i.e., instance attributes), and instead of modifying the classes of tests imported from the modules of common tests (modules `common`, `common_bdd`, and `common_cudd` above), I now subclass them, and set the data attributes within the subclass method `setup_method`. This is the correct approach for ensuring separation between tests.

Even though in this particular case, my initial approach with class attributes was flawed, and using subclasses is a better approach, it is conceivable that there may be use cases where one would like to run tests in a separated way. The approach of `pytest` has the benefit that it will detect such interference, which in principle could be caused also by the package's code (in which case it would have been a bug in the package, and not just an issue with how the tests work). Nevertheless, there may be cases where the opposite behavior (i.e., separated tests) is needed.

(Also, the two plugin packages [`pytest-xdist`](https://github.com/pytest-dev/pytest-xdist) and [`pytest-forked`](https://github.com/pytest-dev/pytest-forked) that are for running tests with parallelism and using subprocesses, exhibit the same behavior as `pytest` in terms of interference between different test modules.)

johnyf, 2021-08-21:
The transition to using `pytest` has been completed. Commit a198b5116887f241171540cae789e524b5f085ab is the main change, and together with followup changes is tested using GitHub Actions as of commit 7ececa4bcae51c85759e7678f706c160e0fc4fa5.

closed 2021-08-21 by johnyf