tulip-control/tulip-control #170: examples/continuous_with_simulation.py gives error

jmcmahon443, 2016-12-14:
We have two machines receiving this error when we run `python continuous_with_simulation.py` with Python 2.7. This issue is just to track it.

```
joe@joe:~/Downloads/tulip-control-master/examples$ python continuous_with_simulation.py 
No handlers could be found for logger "polytope.polytope"
WARNING:tulip.abstract.discretization:	 too small: si \ Pre(sj), so not reached it
WARNING:tulip.abstract.discretization:	 too small: si \ Pre(sj), so not reached it
WARNING:tulip.abstract.discretization:	 too small: si \ Pre(sj), so not reached it
WARNING:tulip.abstract.discretization:	 too small: si \ Pre(sj), so not reached it
WARNING:tulip.abstract.discretization:	 too small: si \ Pre(sj), so not reached it
WARNING:tulip.abstract.discretization:	 too small: si \ Pre(sj), so not reached it
WARNING:tulip.abstract.discretization:	 too small: si \ Pre(sj), so not reached it
WARNING:tulip.abstract.discretization:	 too small: si \ Pre(sj), so not reached it
WARNING:tulip.abstract.discretization:	 too small: si \ Pre(sj), so not reached it
WARNING:tulip.abstract.discretization:	 too small: si \ Pre(sj), so not reached it
WARNING:tulip.abstract.discretization:	 too small: si \ Pre(sj), so not reached it
WARNING:tulip.abstract.discretization:	 too small: si \ Pre(sj), so not reached it
WARNING:tulip.abstract.discretization:	 too small: si \ Pre(sj), so not reached it
WARNING:tulip.abstract.discretization:	 too small: si \ Pre(sj), so not reached it
WARNING:tulip.abstract.discretization:	 too small: si \ Pre(sj), so not reached it
Total abstraction time: 9.2[sec]
removed 0 nodes from 13 total

 Simulation starts 

{'loc': 0, 'lot': True, 'X0reach': True, 'home': False}
Traceback (most recent call last):
  File "continuous_with_simulation.py", line 114, in <module>
    mid_weight=5)
  File "/usr/local/lib/python2.7/dist-packages/tulip/abstract/find_controller.py", line 279, in get_input
    r[idx, :] += -mid_weight * xc
ValueError: non-broadcastable output operand with shape (2,1) doesn't match the broadcast shape (2,2)
```

johnyf, 2016-12-14:
Thank you for the feedback. A user reported this issue last week, and it appears to originate from `polytope`, in particular here: https://github.com/tulip-control/polytope/commit/770a408f74c31992dc7f9f3e2731c4d38794962b.

The example was intended to run in absence of `cvxopt.glpk`. A new release of `polytope` will happen in the near future that is expected to address this issue. Until then, the example runs with `cvxopt.glpk`, but raises the above error with `scipy`.

johnyf, 2017-05-12:
`polytope` addressed this issue, but in a different way than the commit mentioned [above](https://github.com/tulip-control/tulip-control/issues/170#issuecomment-267192059). This issue will be addressed in `tulip` by 87d6da85e69e908f5efe71b35e66d7ba0f4c69ca, part of #180.

johnyf, 2017-07-08:
Addressed as of f177062d241e17518bb17272f59187274d20a9da. The example `continuous_with_simulation.py` now runs using `scipy`, in absence of `cvxopt`. So this example now runs in pure Python, without any errors, in about 45 sec.

closed 2017-07-08 by johnyf