tulip-control/tulip-control #171: New on TuLiP

CosimoSpagnolo, 2017-01-20:
Hi to All,
I'm an Electrical Engineer and i'm doing a PhD. Usually, i m proficent in the design and optimization of electrical machine but my new task is to study the hybridization of electrical distribution system for aircraft.
For this reason i m interested in learning the use of tulip toolbox. Also i m learning to use Python too, could someone explain to me in short terms how learn the tulip toolbox and send to me some examples please ??
I'd like to know also how to load the LTL code in the electronic board to control the switches.
Many thanks

slivingston, 2017-01-20:
Have you tried to read the User's Guide, <https://tulip-control.sourceforge.io/doc/>? I recommend that you begin by reading at least the first 3 sections of it. Next, study examples that are included with the TuLiP source code, which you can find in the `examples` directory. You can browse the examples via GitHub at <https://github.com/tulip-control/tulip-control/tree/98bd53af718c2899418a89bc715fc5d40a0a500e/examples>. 

I do not understand your request about loading code into electronic boards. Can you elaborate?

CosimoSpagnolo, 2017-01-20:
Thanks for the fast answer.
The last question was about if i want to do a conversion from python to matlab and then use drag&drop into FPGA. I'm not expertise in this and could be a wrong question.

Many thanks again

johnyf, 2017-01-20:
About learning Python, I would strongly recommend the following resources:

1. [“The Hitchhiker’s Guide to Python!”](http://docs.python-guide.org/en/latest/) by Kenneth Reitz, Tanya Schlusser, and the Python community
2. [“PEP 20: The Zen of Python”](https://www.python.org/dev/peps/pep-0020/) by Tim Peters
3. [“Effective Python”](http://www.effectivepython.com/) by Brett Slatkin
4. [“Writing Idiomatic Python”](https://www.jeffknupp.com/writing-idiomatic-python-ebook/) by Jeff Knupp

You may also want to try the [Atom](atom.io) editor.

I'm not sure I understand your FPGA-related question, but there are many Python packages related to electronic design automation. Some of them can be found by searching in [PyPI](https://pypi.org/). So, you probably can avoid Matlab entirely.

necozay, 2017-01-20:
For TuLiP, you might want to take a look at `aut2simulink.py` under `contrib` directory for conversion to Simulink. Then you can look for options to go from Simulink to FPGA. `AES` directory includes some old code for specifying electric power system (EPS) problems.

For static EPS problems, we have some python to matlab conversion [here](https://github.com/EPS-Con/eps-reconfig/blob/master/cont_export.py). Note that this uses a different synthesizer than TuLiP for a simpler class of problems.

CosimoSpagnolo, 2017-01-20:
Thank you so much for your help
I'm starting to give a look to the documentation

slivingston, 2017-01-25:
@CosimoSpagnolo if you have found the recommendations given above to be enough, can you close this issue?

CosimoSpagnolo, 2017-01-26:
Hi,
thank you for your support
i'm closing the issue

closed 2017-01-26 by CosimoSpagnolo