tulip-control/tulip-control #173: In examples, inconsistent naming of X0 and X5 as "lot" and "home"

slivingston, 2017-01-27:
As of commit 98bd53af718c2899418a89bc715fc5d40a0a500e (`master` branch), several of the examples are based on a task of repeatedly visiting "home" cell and, when requested, eventually reaching a (parking) "lot" in a 2\times 3 grid. In some of these, the descriptions in the comments and the actual implementation are not consistent with respect to the locations of "home" and "lot". This comment applies to [hybrid.py](https://github.com/tulip-control/tulip-control/blob/98bd53af718c2899418a89bc715fc5d40a0a500e/examples/hybrid.py), [discrete.py](https://github.com/tulip-control/tulip-control/blob/98bd53af718c2899418a89bc715fc5d40a0a500e/examples/discrete.py), [controlled_switching.py](https://github.com/tulip-control/tulip-control/blob/98bd53af718c2899418a89bc715fc5d40a0a500e/examples/controlled_switching.py), [only_mode_controlled.py](https://github.com/tulip-control/tulip-control/blob/98bd53af718c2899418a89bc715fc5d40a0a500e/examples/only_mode_controlled.py), [environment_switching.py](https://github.com/tulip-control/tulip-control/blob/98bd53af718c2899418a89bc715fc5d40a0a500e/examples/environment_switching.py). Compare these with [examples/gr1.py](https://github.com/tulip-control/tulip-control/blob/98bd53af718c2899418a89bc715fc5d40a0a500e/examples/gr1.py)

There are at least two solutions:

1. Swap the labels of 'home' and 'lot'. The exact implementation of this depends on the file. In hybrid.py, use `state_labels = [{'lot'}, set(), set(), set(), set(), {'home'}]`. In discrete.py, use `sys.states.add('X0', ap={'lot'})` and `sys.states.add('X5', ap={'home'})`.

2. In the docstring and comments, change the diagram of the partition to have label `0` in the top-right cell, label `5` in the lower-left, and change the verbal descriptors accordingly, e.g., "...moving to the lower left corner of the grid" becomes "...moving to the upper right corner of the grid."

One advantage of the first solution is that it makes all manifestations of this "home" and "lot" scenario consistent, including in the User's Guide.

johnyf, 2017-01-27:
I am fine with either. Thanks for spotting this inconsistency.

Also, at some point it would be interesting to vary the examples a little, possibly by also changing the "story". Two action items in this area were assigned to me in the past:

1. Tighten the specifications to make them nearly unrealizable. 
2. Use the automated translation of non-GR(1) formulae to both demonstrate that facility, and make some examples more readable.

These still exist in my todo list. Noting them here for shared reference.

slivingston, 2017-02-01:
In the example environment_switching.py, this issue was first noticed by Rui Zhang, who notified me earlier this month via email and to whom thanks is due. That report motivated me to check the other example files.