tulip-control/tulip-control #174: introduce simulation-equivalence based abstraction

necozay, 2017-02-05:
Simulation equivalence based abstractions are in general much less conservative than bisimulation-based abstractions for LTL synthesis as discussed [here](http://web.eecs.umich.edu/~necmiye/pubs/WagenmakerO_allerton16.pdf). Would be nice to change the default abstraction method in TuLiP (which is based on a bisimulation algorithm) with a simulation-equivalence based one.

johnyf, 2017-02-05:
Implementing new abstraction algorithms is an opportunity to try a different design, using `networkx` to represent the abstraction itself, as mentioned some time ago https://github.com/tulip-control/tulip-control/issues/62#issuecomment-36207297. The graph is more natural a data structure, especially given than nodes are being created and removed dynamically. I am not in favor of using a sparse matrix anymore, because it involves index-based reasoning.

The runtime is mainly confused in performing concrete operations (linear programming for polytopic reasoning). So, the sparsity shouldn't play an important role, but rather the readability and practicality of the data structure for the task at hand. Only profiling could justify a complicated choice, and still I support dual implementation: a simple, readable, short one used to teach and also to test (possibly) another that is optimized (the latter probably won't be needed).

Also, I recommend *not* using `transys` anywhere. If the algorithm is independent of `transys`, and uses `networkx`, then it's easy to create a small interface at the boundary, and update that interface in the future, if and when `transys` is replaced by a simpler version.

murrayrm, 2017-02-05:
I agree that it would be very interesting to have more than one abstraction algorithm and to start generalizing TuLiP (if needed) to allow for that.

necozay, 2020-03-02:
done in 19da152d868914da475428839481d64bb6acaadc (namely  dual-simulation)

closed 2020-03-02 by necozay