tulip-control/tulip-control #175: absent key `'message'` used in error message in get_input_helper()

slivingston, 2017-02-22:
To witness the error, try to run examples/continuous_with_simulation.py after applying the patch
```diff
diff --git a/examples/continuous_with_simulation.py b/examples/continuous_with_simulation.py
index 2ddd685..1ca86b1 100644
--- a/examples/continuous_with_simulation.py
+++ b/examples/continuous_with_simulation.py
@@ -104,6 +104,7 @@ ctrl = synth.determinize_machine_init(ctrl, {'loc': s0_part})
 print(dum)
 for i in range(0, T):
     (s, dum) = ctrl.reaction(s, {'park': randParkSignal[i]})
+    sys_dyn.A = sys_dyn.A + 1e3*np.eye(sys_dyn.A.shape[0])
     u = find_controller.get_input(
         x0=np.array([x[i * N], y[i * N]]),
         ssys=sys_dyn,
```
which renders the LP in `get_input` infeasible.

In [`get_input_helper`, line 452 of abstract/find_controller.py](https://github.com/tulip-control/tulip-control/blob/5d161c0ee74020b80c320f119c15a325d6d34a39/tulip/abstract/find_controller.py#L452), which is called by [`get_input`](https://github.com/tulip-control/tulip-control/blob/5d161c0ee74020b80c320f119c15a325d6d34a39/tulip/abstract/find_controller.py#L78), the exception message uses a key `'message'` which is not returned in the dictionary from [`lpsolve` from the `polytope` package](https://github.com/tulip-control/polytope/blob/a037b555758ed9ee736fa7cb324d300b8d622fb4/polytope/polytope.py#L2186).

I did not investigate commit history regarding the origin of this, but notice that the LP solver from SciPy returns a dictionary that has the key `'message'`, which is described in the documentation of `scipy.optimize.linprog` as "A string descriptor of the exit status of the optimization."

This error was reported to me by Rui Zhang (@RuiZhangCSU).

slivingston, 2017-02-22:
I propose as solution that the exception message contains the status integer from the LP solver. E.g.,
```diff
diff --git a/tulip/abstract/find_controller.py b/tulip/abstract/find_controller.py
index 5ace43a..0ac476a 100644
--- a/tulip/abstract/find_controller.py
+++ b/tulip/abstract/find_controller.py
@@ -448,8 +448,8 @@ def get_input_helper(
     if sol['status'] != 0:
         raise Exception(
             "getInputHelper: "
-            "LP solver finished with message " +
-            str(sol['message']))
+            "LP solver finished with status " +
+            str(sol['status']))
     var = np.array(sol['x']).flatten()
     u = var[-N * m:]
     cost = sol['fun']
```

slivingston, 2017-02-22:
The interpretation of a status integer requires checking documentation for the solver. So, we might want to place a description string under the key `'message'`. However, this would involve more maintenance effort of `lpsolve` in the package `polytope` if the interpretation of nonzero status integers changes upstream in a future version of `scipy.optimize.linprog`.

necozay, 2017-02-22:
I think having `lpsolve` in `polytope` populate the key `message` with a uniform description is a good idea. I would expect the `scipy.optimize.linprog` and other solver status messages to be stable (at least at a rough "feasible, infeasible, unknown" level) and `lpsolve` should be maintained to keep up with it.

johnyf, 2017-02-23:
Looking back at ca5a80d8b91e79345c27de0629d683f95cfeb455 and the commits in #168 before rebase, I think that the `'message'` is an error introduced by me here: https://github.com/tulip-control/tulip-control/commit/4d1cac37fc39846e9d7986536e6a496c7dbc37f4#diff-db73a44c4631fa5fc7c04bff0f7d729aR449 during the rebase.

I am fine with either of the solutions proposed. The [first one](https://github.com/tulip-control/tulip-control/issues/175#issuecomment-281783827) is similar to [a few lines above](https://github.com/tulip-control/tulip-control/blob/5d161c0ee74020b80c320f119c15a325d6d34a39/tulip/abstract/find_controller.py#L427). I recall reading the documentation of [`linprog`](https://docs.scipy.org/doc/scipy-0.16.1/reference/generated/scipy.optimize.linprog.html), so that is most likely why I changed `'status'` to '`message`'.

Perhaps we can change to `'status'` for now, in order to address this error. After we release a `polytope` version that propagates the message returned from `scipy`, then we can implement `'message'`.