tulip-control/tulip-control #178: Until operator usage

txlin, 2017-03-14:
I'd like to modify the existing example gr1.py line 112 so that the robot would wait at X0 until the park signal is released. My instinct would be to modify the line to become: 

sys_safe |= {'(X (X0reach) <-> X0) || (X0reach U !park)'}

However, doing this results in a long error message list (attached below). Is this incorrect syntax for what I'm trying to achieve?

Traceback (most recent call last):
  File "single.py", line 134, in <module>
    strategy = synth.synthesize('omega', specs)
  File "/usr/local/lib/python2.7/dist-packages/tulip-1.3.0-py2.7.egg/tulip/synth.py", line 1135, in synthesize
    strategy = omega_int.synthesize_enumerated_streett(specs)
  File "/usr/local/lib/python2.7/dist-packages/tulip-1.3.0-py2.7.egg/tulip/interfaces/omega.py", line 66, in synthesize_enumerated_streett
    a = aut.build()
  File "/usr/local/lib/python2.7/dist-packages/omega/symbolic/symbolic.py", line 236, in build
    aut = _bitblast(self)
  File "/usr/local/lib/python2.7/dist-packages/omega/symbolic/symbolic.py", line 360, in _bitblast
    _bitblast_owner(aut, a, 'sys', t)
  File "/usr/local/lib/python2.7/dist-packages/omega/symbolic/symbolic.py", line 398, in _bitblast_owner
    a.action[owner] = list(map(f, aut.action[owner]))
  File "/usr/local/lib/python2.7/dist-packages/omega/symbolic/symbolic.py", line 395, in f
    return bv.bitblast(x, t)
  File "/usr/local/lib/python2.7/dist-packages/omega/logic/bitvector.py", line 40, in bitblast
    return tree.flatten(t=t)
  File "/usr/local/lib/python2.7/dist-packages/omega/logic/bitvector.py", line 455, in flatten
    x = self.operands[0].flatten(*arg, **kw)
  File "/usr/local/lib/python2.7/dist-packages/omega/logic/bitvector.py", line 456, in flatten
    y = self.operands[1].flatten(*arg, **kw)
  File "/usr/local/lib/python2.7/dist-packages/omega/logic/bitvector.py", line 455, in flatten
    x = self.operands[0].flatten(*arg, **kw)
  File "/usr/local/lib/python2.7/dist-packages/omega/logic/bitvector.py", line 456, in flatten
    y = self.operands[1].flatten(*arg, **kw)
  File "/usr/local/lib/python2.7/dist-packages/omega/logic/bitvector.py", line 460, in flatten
    op=Nodes.opmap[self.operator], x=x, y=y)
KeyError: 'U'

johnyf, 2017-03-14:
The "until" operator `U` is not in the initial predicate + action + GR(1) fragment (state machine + GR(1) for short) of linear temporal logic (action = formula that contains primed and unprimed variables, no other temporal operators), hence the error. Some LTL operators are expressible in this fragment. Not all LTL operators are expressible.

The until operator is expressible in LTL. The translation is an observer that keeps track of what is going on. The caveat is that to *remember* what is going on, the observer must save somewhere its observations for reading them later, so it needs "memory". The memory comes in the form of fresh variables that are added to the problem. That is the cost of expressing some temporal operators in the fragment for which we have polynomial-time synthesis algorithms.

In [TLA+](http://lamport.azurewebsites.net/tla/book.html) those fresh memory variables are called [history variables](https://doi.org/10.1016/0304-3975(91)90224-P) and are existentially quantified with the temporal operator `\EE`, so they are "internal" variables--they don't constrain the behavior. In the [GR(1) literature](https://doi.org/10.1007/11609773_24) these variables represent the current node of a deterministic Buchi automaton. They are history variables because the automaton is deterministic.

The observers are called [temporal testers](http://dx.doi.org/10.1007/978-3-540-69850-0_11). The [variable `X0reach`](https://github.com/tulip-control/tulip-control/blob/238866b45a86117d89ddb32343e42f5739ab64ae/examples/gr1.py#L94) is the memory of a temporal tester that keeps track of satisfaction of the request-response formula `[](park -> X0)`.

Q: What options do you have if you want to express "until" in GR(1)?
A:

1. first call an automated translation from LTL to the state machine + GR(1) fragment that constructs temporal testers automatically. The function [`omega.logic.past.translate`](https://github.com/johnyf/omega/blob/a808dc1a5b02d451ddce6e33640562f00a2a4771/omega/logic/past.py#L247) does this translation. You will need to call it, for example as done in the function `openpromela.logic.map_to_future`, which [calls](https://github.com/johnyf/openpromela/blob/8a74b37b2b481cb9b1914fd9b2b4b38bb8162a73/openpromela/logic.py#L2285) [`map_translate`](https://github.com/johnyf/omega/blob/a808dc1a5b02d451ddce6e33640562f00a2a4771/omega/logic/past.py#L298).

    Beware that the translation is sound and complete for closed systems (no environment), but in the presence of an environment is only sound, not complete. For some operators it is complete (an "until" on its own--not within a negation--is one of them).

2. Call the function [`tulip.spec.gr1_fragment.until_to_gr1`](https://github.com/tulip-control/tulip-control/blob/238866b45a86117d89ddb32343e42f5739ab64ae/tulip/spec/gr1_fragment.py#L421), which translates the until operator to a temporal tester.

3. Write the translation by hand (a good exercise), then compare to the output of the previous options to check your result.

## Error explanation

The function `omega.logic.bitvector.bitblast` takes a formula in [first-order logic](https://en.wikipedia.org/wiki/First-order_logic) and returns another first-order formula, expressed in terms of fresh variables (and currently in another syntax than the input formula). Those fresh variables are usually called "bits". The main point about them is that it suffices to reason only about Boolean assignments to those fresh variables in order to reach conclusions about the generated formula, and so about the input formula too.

Q: How does the "until" operator get to the `omega` package from `tulip`?
A: The formulae are passed to `omega` mostly unchanged, [except for replacing](https://github.com/tulip-control/tulip-control/blob/master/tulip/interfaces/omega.py#L193) "string" variable values with integers.

Q: Why is "until" parsed as a binary operator?
A: The parser in `omega.logic.lexyacc` [knows about it](https://github.com/johnyf/omega/blob/a808dc1a5b02d451ddce6e33640562f00a2a4771/omega/logic/lexyacc.py#L191).

Q: Why does the parser know about "until" if it isn't going to be bitblasted?
A: Usually parsers are written once to handle more cases than the backend. The reasons are to allow extensibility later, to use the same parser for multiple backends, and because there are context-sensitivity and other properties of languages that are easier to handle in the backend, rather than the frontend (the parser).

Q: Does `omega` handle the "until" operator?
A: Yes, see comments above.

Regarding first-order logic: In the `gr1.py` example, the values of the [variables](https://github.com/tulip-control/tulip-control/blob/238866b45a86117d89ddb32343e42f5739ab64ae/examples/gr1.py#L100) that matter for deciding whether the spec is realizable and constructing a controller [are Boolean](https://github.com/tulip-control/tulip-control/blob/238866b45a86117d89ddb32343e42f5739ab64ae/tulip/spec/form.py#L341), but at this layer they could have been integer values too.

johnyf, 2017-03-23:
@txlin did the above answer you question? Can we close this issue?

txlin, 2017-03-23:
Yes, thanks for the help

closed 2017-03-23 by txlin