tulip-control/tulip-control #179: Can't run example discrete.py

andreasjacob, 2017-04-10:
So, with the example file discrete.py as it is, gives an exception _"synthesize() takes at least 2 arguments (2 given)"_

Providing 'omega' as additional argument and trying again results in _AssertionError: renaming target var "loc_2'" at level 9 is essential, rename: {0: 1, 2: 3, 4: 5, 6: 7, 8: 9, 10: 11, 12: 13}, support: set([9, 11])_

Can you kindly point me what am I missing out? I just want to run this example and generate the graph for controller.

Sincerely,
Andreas

slivingston, 2017-04-11:
Which version of `tulip` are you using? E.g., try

    python -c 'import tulip; print(tulip.__version__)'

slivingston, 2017-04-11:
In a new Python 2.7.13 environment with `pip` version 9.0.1, the following quick run using the current tip of `master` branch (commit 5d69ce50b29b4baddfa537feec2acf3c61a3e92b) succeeds for me:
```sh
curl -L -O https://github.com/tulip-control/tulip-control/archive/master.zip
unzip master.zip
cd tulip-control-master
pip install .
cd examples
python discrete.py
```
on Fedora 25 (GNU/Linux).

slivingston, 2017-04-11:
@andreasjacob One hypothesis is that you have the most recent release of TuLiP installed, but you are trying the examples as on the current tip of `master` branch, which includes changes that are not backwards-compatible.

If my hypothesis is true, then you should try to use the examples as of the release v1.3.0. You can fetch a tarball from <https://github.com/tulip-control/tulip-control/releases/tag/v1.3.0>, or if you cloned the repo, then update to the appropriate tag using

    git checkout v1.3.0

johnyf, 2017-04-11:
I could reproduce both the errors that you observed as follows:

```
(dev) examples:pip show tulip
Name: tulip
Version: 1.3.0
Summary: Temporal Logic Planning (TuLiP) Toolbox
Home-page: http://tulip-control.org
Author: Caltech Control and Dynamical Systems
Author-email: tulip@tulip-control.org
License: BSD
Location: /home/ioannis/.virtualenvs/dev/lib/python2.7/site-packages
Requires: scipy, pydot, omega, polytope, ply, numpy, networkx

(dev) examples:git log -1
commit 5d69ce50b29b4baddfa537feec2acf3c61a3e92b
Author: Ioannis Filippidis <jfilippidis@gmail.com>
Date:   Wed Apr 5 19:39:11 2017 -0700

    DOC: add commit prefix "CI" for "continuous integration"

(dev) examples:git status discrete.py 
On branch master
Your branch is up-to-date with 'upstream/master'.
nothing to commit, working tree clean

(dev) examples:python discrete.py 
No handlers could be found for logger "polytope.polytope"
Traceback (most recent call last):
  File "discrete.py", line 137, in <module>
    ctrl = synth.synthesize(specs, sys=sys)
TypeError: synthesize() takes at least 2 arguments (2 given)
```

So the first error was likely caused by running a version of `tulip` older than branch `master`, but attempting to run the example `discrete.py` from the repository checked out at branch `master`. The issue can be avoided by either:

1. running a version of the example that matches the version of `tulip`, e.g., `pip download tulip` and then `tar -xzf`, or
2. `git checkout 1.3.0` and `python setup.py install` or `pip install .`

I could reproduce the second error by upgrading to the latest `omega` and `dd`, which are not compatible with the released `tulip == 1.3.0` (neither with branch `master` yet):

```
(dev) examples:pip show tulip omega dd
Name: tulip
Version: 1.3.0
Summary: Temporal Logic Planning (TuLiP) Toolbox
Home-page: http://tulip-control.org
Author: Caltech Control and Dynamical Systems
Author-email: tulip@tulip-control.org
License: BSD
Location: /home/ioannis/.virtualenvs/dev/lib/python2.7/site-packages
Requires: scipy, pydot, omega, polytope, ply, numpy, networkx
---
Name: omega
Version: 0.1.0
Summary: Symbolic algorithms for solving games of infinite duration.
Home-page: https://github.com/johnyf/omega
Author: Ioannis Filippidis
Author-email: jfilippidis@gmail.com
License: BSD
Location: /home/ioannis/.virtualenvs/dev/lib/python2.7/site-packages
Requires: dd, natsort, pydot, ply, networkx
---
Name: dd
Version: 0.5.1
Summary: Library of decision diagrams and algorithms on them, in pure Python, as well as Cython bindings to CUDD, Sylvan, and BuDDy.
Home-page: https://github.com/johnyf/dd
Author: Ioannis Filippidis
Author-email: jfilippidis@gmail.com
License: BSD
Location: /home/ioannis/.virtualenvs/dev/lib/python2.7/site-packages
Requires: astutils, psutil, pydot, ply, setuptools, networkx

(dev) examples:git status discrete.py 
On branch master
Your branch is up-to-date with 'upstream/master'.
nothing to commit, working tree clean
(dev) examples:git log -1
commit 5d69ce50b29b4baddfa537feec2acf3c61a3e92b
Author: Ioannis Filippidis <jfilippidis@gmail.com>
Date:   Wed Apr 5 19:39:11 2017 -0700

    DOC: add commit prefix "CI" for "continuous integration"

(dev) examples:vi discrete.py

(dev) examples:python discrete.py 
No handlers could be found for logger "polytope.polytope"
WARNING: yacc table file version is out of date
Traceback (most recent call last):
  File "discrete.py", line 137, in <module>
    ctrl = synth.synthesize('omega', specs, sys=sys)
  File "/home/ioannis/.virtualenvs/dev/lib/python2.7/site-packages/tulip/synth.py", line 1135, in synthesize
    strategy = omega_int.synthesize_enumerated_streett(specs)
  File "/home/ioannis/.virtualenvs/dev/lib/python2.7/site-packages/tulip/interfaces/omega.py", line 69, in synthesize_enumerated_streett
    z, yij, xijk = gr1.solve_streett_game(a)
  File "/home/ioannis/.virtualenvs/dev/lib/python2.7/site-packages/omega/games/gr1.py", line 54, in solve_streett_game
    y, yj, xjk = _attractor_under_assumptions(z, goal, aut)
  File "/home/ioannis/.virtualenvs/dev/lib/python2.7/site-packages/omega/games/gr1.py", line 80, in _attractor_under_assumptions
    safe, aut, unless=unless)
  File "/home/ioannis/.virtualenvs/dev/lib/python2.7/site-packages/omega/symbolic/fixpoint.py", line 62, in trap
    pre = ue_preimage(env_action, sys_action, q, aut)
  File "/home/ioannis/.virtualenvs/dev/lib/python2.7/site-packages/omega/symbolic/fixpoint.py", line 91, in ue_preimage
    u = bdd.rename(target, aut.prime)
  File "/home/ioannis/.virtualenvs/dev/lib/python2.7/site-packages/dd/bdd.py", line 81, in _wrapper
    return func(bdd, *args, **kwargs)
  File "/home/ioannis/.virtualenvs/dev/lib/python2.7/site-packages/dd/bdd.py", line 583, in rename
    return rename(u, self, dvars)
  File "/home/ioannis/.virtualenvs/dev/lib/python2.7/site-packages/dd/bdd.py", line 1546, in rename
    _assert_valid_rename(u, bdd, dvars)
  File "/home/ioannis/.virtualenvs/dev/lib/python2.7/site-packages/dd/bdd.py", line 1601, in _assert_valid_rename
    r=dvars, s=s)
AssertionError: renaming target var "loc_2'" at level 9 is essential, rename: {0: 1, 2: 3, 4: 5, 6: 7, 8: 9, 10: 11, 12: 13}, support: set([9, 11])
```

The argument `'omega'` was passed first to `synthesize`.

andreasjacob, 2017-04-11:
Indeed, I was running the example from master with tulip installation of version 1.3.0. (dd v.0.5.1 and omega 0.1.0).

I removed tulip and omega. Enter _python2 setup.py_ install for tulip. And now the example discrete.py (provided with Tulip v.1.3.0 tar) successfully completes.

johnyf, 2017-04-11:
Thanks for noting what caused this issue. A few suggestions for future work with Python:

- It is recommended to use `pip install .` or `pip install -e .` when possible instead of `python setup.py install`. `pip` first collects all dependencies, analyzes the resulting (acyclic) graph, and installs dependencies before dependents. Sometimes this can make a difference, e.g., for generating parser tables at the time of install, so that they be copied to `site-packages` (this refers to parser tables for dependencies, not the package being installed, e.g., `tulip`, because all its dependencies will have been installed before its installation, even if using `setuptools`).

- The reason that `pip` and `python setup.py` fetch suitable versions is that constraints are declared in the file `setup.py`, in particular the `install_requires` argument passed to `setuptools`.

- [virtual environments](http://python-guide-pt-br.readthedocs.io/en/latest/dev/virtualenvs/) are a good way to organize different sets of dependencies, with the possibility of different versions of python (motivated by the `python2` invocation).

- The error `renaming target var "loc_2'" at level 9 is essential` means that it was attempted to substitute the variable `loc_2'` for some other variable (most likely the unprimed `loc_2`) in an operator that depends on `loc_2'` (the operator is represented as a binary decision diagram). This is a constraint in `dd <= 0.5.1` that is unnecessarily restrictive and was recently removed on branch `dev` of `dd` (the implementation actually supports this operation).

- The order of arguments to the function `tulip.synth.synthesize` changed in 43d4ceba808398034782fea920b41955429234c1, so the examples on branch `master` correspond to the newer API.

closed 2017-04-11 by johnyf

slivingston, 2017-04-11:
@andreasjacob we now have a mailing list where user questions can be posed: <https://lists.sourceforge.net/lists/listinfo/tulip-control-users>. In the future, I encourage you to ask there first before opening an issue in this issue tracker.