tulip-control/tulip-control #184: deprecate "states" based API of `transys`

johnyf, 2017-09-01:
The new API of `networkx` is simpler and Python 3 friendly. After #183, the time has come to replace the module `tulip.transys.labeled_graphs` and delete most of `transys`.

### Preview

- some of the API should be removed (unnecessary checks for already existing nodes and edges, and similar restrictions),
- some of the API isn't needed ("typed" attributes and similar complex notions)
- some of the API is obsolete (`networkx` now supports using a user-defined dictionary attribute, so no need to intercept method calls)
- the exporting functionality should be a separate module or subpackage that *takes* graphs as input and *dumps* them in the desired format. Not nested methods for formatting and other operations that unrelated to the concept of graph (relevant to #78).

The `states` and `transitions` attributes should be removed. These names are inaccurate; they give the impression that these are states (mapping variables to values), but they are nodes in a graph that is used as a piece in a larger design. Instead, the `networkx` will be retained. There should be one (preferably only one) way for doing things [PEP 20].

### Recommended roadmap

1. Initially, the new API can be introduced as a reimplementation of `transys` in the form of new modules within it. This "new API" will be very thin, and rely as much as possible on `networkx`.

2. The old API will be deprecated. Usage of `transys` within `tulip` will be replaced by `networkx`, and only if necessary by `transys`. A good design approach is to try to turn `transys` into a collection of functions that operate on `networkx` data structures, instead of data structures ("It is better to have 100 functions operate on one data structure than 10 functions on 10 data structures." [Alan Perlis]).

3. `tulip < 2.0` will keep the deprecated API.

4. `tulip >= 2.0` will drop the deprecated API.

slivingston, 2017-09-02:
I agree with your proposed roadmap.

Given the large scope of this issue, I recommend to create single-task issues about specific parts, which may better allow assigning these tasks to different humans. Of course, having a single issue is OK if you want to take this all on yourself.

johnyf, 2017-09-05:
There are at least two kinds of tasks that I expect to be able to partition the work into:

1. ensuring that no useful functionality is forgotten during reimplementation (a task that benefits from overlapping reviews)
2. implementing proposed API that has already been specified in docstrings.

I think the approach of item 2 will enable specifying an API design and then allocating implementation of different functions and methods to different persons. The specifications will not be intended as rigid, but will render design changes during implementation explicit.

johnyf, 2017-09-28:
Relevant to #18.

johnyf, 2017-09-28:
Relevant to [PEP 4](https://www.python.org/dev/peps/pep-0004/).