tulip-control/tulip-control #185: Error with matlabsim contribution

Tkabb, 2017-09-15:
I am trying to run the examples in the matlabsim contribution but I am getting the following error, does someone has an idea about this issue or how to solve it (maybe the contribution developer @stsuei ):

Traceback (most recent call last):
  File "discrete.py", line 61, in <module>
    tomatlab.export('robot_discrete.mat', ctrl)
  File "../tomatlab.py", line 111, in export
    output['TS'] = export_mealy(mealy_machine, is_continuous)
  File "../tomatlab.py", line 314, in export_mealy
    trans_dict['start_loc'] =  mealy_machine.states[orig_state]['loc']
KeyError: 'loc'

PS: Tulip works will and generate the controller but with small thing that I don't now if it is an issue, that is the states are named with what seems to be a memory location (ex. 0x1603CB0), it might be because I am using gr1c not omega.
 
Best to all,
Tarek

johnyf, 2017-09-15:
The failing line is [`tomatlab.py#L312`](https://github.com/tulip-control/tulip-control/blob/4a6837bbcb42f92a59ac67793b8a4dda7f061889/contrib/matlabsim/tomatlab.py#L312) (this line is 314 above, which means that either it is from another commit, or the file has been edited). What it means is that the variable with name `'loc'` is not assigned any value by the state `orig_state` of the synthesized Mealy machine.

Some possibilities:

1. The specification did not involve translating any transition system to temporal logic formulas. This translation happens in `synth` and the choice of `statevar` (the name of the variable that represents the "current" node of the transition system) is either [`loc`](https://github.com/tulip-control/tulip-control/blob/4a6837bbcb42f92a59ac67793b8a4dda7f061889/tulip/synth.py#L1202) or [`'eloc'`](https://github.com/tulip-control/tulip-control/blob/4a6837bbcb42f92a59ac67793b8a4dda7f061889/tulip/synth.py#L1226). If discretization is involved, then there is a transition system that gets translated. Otherwise, maybe there isn't one (unless your specification includes one).

2. The node variable is named `'eloc'` (see previous item).

3. The node variable has been given some custom name by the user (see [here](https://github.com/tulip-control/tulip-control/blob/master/tulip/synth.py#L1198)).

One way to see what variables are declared, and assigned values at that state, is to

```python
print(mealy_machine.state[orig_state])
```

Comparing this output to `GRSpec.env_vars` and `GRSpec.sys_vars` would reveal whether there is some variable missing, though `'loc'` is a variable added within the call to the function [`tulip.synth.synthesize`](https://github.com/tulip-control/tulip-control/blob/4a6837bbcb42f92a59ac67793b8a4dda7f061889/tulip/synth.py#L1019), if a graph-like transition is passed to `synthesize` (i.e., in addition to the `specs` argument), so it will be absent from the `GRSpec`.

Regarding the second question, the hexadecimal numbers as names are indeed what `gr1c` generates in its latest versions (see also #137).

johnyf, 2017-09-15:
Side note: if your spec does not involve any transition system (e.g., it is written with a few variables and a couple of formulas, as in [`examples/gr1.py`](https://github.com/tulip-control/tulip-control/blob/master/examples/gr1.py)), and you prefer to keep the file `tomatlab.py` unchanged, then you could simply declare a dummy variable named `'loc'` that is an integer (say `\in 0..3`) and doesn't occur in the spec. This may circumvent the error without any other changes.

Though other errors may occur later, e.g., if simulation within Matlab expects to find cells of the abstraction that correspond to `'loc'`. If so, then you may want to use only part of the Matlab interface code on the Matlab side, depending on your application.

Tkabb, 2017-09-16:
Thanks for the reply. I have actually only run (with latest release of Tulip) the example given with the contribution folder, namely "discrete.py", with the following slit change (that I was not able to run with out them and which might be in fact the problem):
specs.qinit = '\A \E' 	
specs.moore = False	
specs.plus_one = False

By printing the following (which is why the error line is 314)
print(mealy_machine.state[orig_state]) 
it gives:
TypedDict({})

'eloc' gives the same error with:
KeyError: 'eloc'

I also am having the same error running the continuous example in the contribution folder, namely "continuous.py" and with about the same changes above (without the last).

In my opinion "tomatlab.py" looks for the default variable 'loc' (assigned by "tulip.synth.synthesize") which represent the current activated system variable for the current state (In the examples "Xn", n={0,1,...} ). Giving those values (using"
trans_dict['outputs']['loc']
Instead of
mealy_machine.states[orig_state]['loc']
), I was apple to generate the matlab file and the stateflow  (but I could not run it in stateflow because of the folowing error:
Chart 'TulipController' has unresolved symbols.

B.R.

Tkabb, 2017-09-16:
BTW, I am having an error in "gr1.py" example in Tulip:
No handlers could be found for logger "polytope.polytope"
Traceback (most recent call last):
  File "gr1.py", line 124, in <module>
    strategy = synth.synthesize(specs)
TypeError: synthesize() takes at least 2 arguments (1 given)

I should add the solver before the specifications to make it work:
strategy = synth.synthesize('omega',specs)
Might be because I installed tulip first with pip, thin by cloning the master branch (where the solver is not needed) and installing it.

johnyf, 2017-09-17:
Thanks for noting these issues. Please find corrections on the branch `matlabsim_fixes` (https://github.com/tulip-control/tulip-control/tree/matlabsim_fixes).

About using a Mealy controller with `\A env0: \E sys0` initial condition, currently `gr1c` supports this option, and other options are supported by `omega`. When `matlabsim` was introduced, this was the only available option, and later it wasn't updated.

The node label is an empty dictionary because only edges are labeled in Mealy machines. To get the value of `'loc'` at `orig_state`, an edge that leads to that node has to be used. Such an edge is contained in the edges that start from "Sinit". So this needs to be done differently. Introduced in https://github.com/tulip-control/tulip-control/commit/68cd8102e9b67e3743421f83b9c4d611ba7da8e8#diff-d7552caf44bbf33ec2be614f8105c543R312. Addressed on branch `matlabsim_fixes`.

About the Stateflow error, the name `TulipController` seems to be set [here](https://github.com/tulip-control/tulip-control/blob/4a6837bbcb42f92a59ac67793b8a4dda7f061889/contrib/matlabsim/discrete_example/load_discrete.m#L10), and in similar files of other examples. I do not know why this error arises, though [this part of Matlab's documentation](https://www.mathworks.com/help/stateflow/ug/resolving-event-data-and-function-symbols-in-stateflow-action-language.html) may be useful for identifying which symbols are undefined.

Branch `master` has a different signature for `synth.synthesize` than `tulip == 1.3.0` on PyPI. Using `pip uninstall -y tulip`, and then installing from the repository can ensure the development version is installed. The signature in the latest version takes the solver as keyword argument `synthesize(..., solver='omega')`.

Tkabb, 2017-09-18:
Ok very good, I will install this branch and work on it.  

For the unresolved symbols in Stateflow, I think that the problem comes from the fact that the env. input signal (e.x. park in discrete example) which is build using "load_discrete.m" [file](https://github.com/tulip-control/tulip-control/blob/matlabsim_fixes/contrib/matlabsim/discrete_example/load_discrete.m) to be selected randomly between 0 or 1. whereas for the generated TulipController, it takes as input and output, the values of "True" and "False" (not the boolean ones in matlab with minuscule first letters). Therefore, a change to the input and output values of TulipController should be done to either be 0s and 1s (worked) or with minuscule letters ( change of size problem) or a change of the env. signal to be a string (more complicated for same size problem).

I would propose some other fixes to [load_discrete.m](https://github.com/tulip-control/tulip-control/blob/matlabsim_fixes/contrib/matlabsim/discrete_example/load_discrete.m) which has missing definition in order to be able to run [load_tulip.m](https://github.com/tulip-control/tulip-control/blob/matlabsim_fixes/contrib/matlabsim/load_tulip.m). Contrary to the continuous case, systype has to defined. So the following line should be added to "load_discrete.m" before calling "load_tulip.m":
systype = 'is_discrete';

In addition, in [load_tulip.m](https://github.com/tulip-control/tulip-control/blob/matlabsim_fixes/contrib/matlabsim/load_tulip.m) the following change should be done so that the discrete case is taken in consideration. The lines 109..112 and 163..166 belong to the switched case, so they should be surrounded by "if" as follow:

if is_switched
    num_modes = length(MPTsys);
        for j = 1:num_modes
            MPTsys(j).env_act = input_value_map(MPTsys(j).env_act);
        end
end

if is_switched
   num_modes = length(MPTsys);
    for j = 1:num_modes
        MPTsys(j).sys_act = output_value_map(MPTsys(j).sys_act);
    end
end

After ruining the "discrete.py" example in the matlabsim_fixes branch and then running [load_discrete.m](https://github.com/tulip-control/tulip-control/blob/matlabsim_fixes/contrib/matlabsim/discrete_example/load_discrete.m) in Matlab, I am getting the following error:

Index exceeds matrix dimensions.

Error in load_tulip (line 253)
    transition_handles{ind}.Source = state_handles{start_state_index};

Error in load_discrete (line 7)
load_tulip;

B.R.

johnyf, 2017-09-19:
Thanks for the suggestions. To coordinate merging of the proposed changes, would it work to open a pull request? If any of the errors are not exposed by any of the current `matlabsim` examples, it may be useful to add small examples or unit tests that exercise those areas of code.

Tkabb, 2017-09-20:
My pleasure, yes it would be good as I might also work on the continuous part. So I can add all proposed corrections directly there. However, as I am short on time I would concentrate first on making it work by fixing errors and I will add supporting code as possible along the way or after I finish.
Thanks for the help.

Tkabb, 2017-09-20:
When running "continuous.py" example in `matlabsim_fixes` , the following error is raised:

```
Traceback (most recent call last):
  File "continuous.py", line 106, in <module>
    disc_params)
  File "../tomatlab.py", line 103, in export
    output['TS'] = export_mealy(mealy_machine, is_continuous)
  File "../tomatlab.py", line 227, in export_mealy
    state_dict['loc'] = state_tuple[1]['loc']
KeyError: 'loc'
```

johnyf, 2017-09-21:
This error and some other issues have been addressed by further work on branch `matlabsim_fixes`.
I also refactored the code to use the [`networkx`](https://networkx.github.io/) interface as much as possible.

The double tank example raises an error within `tulip`, but its maintained version under `examples/developer/fuel_tank` works, so updates based on the latter are expected to resolve that error.

One way to see the graphs is to plot them, by calling the method `tulip.transys.machines.MealyMachine.save` (in the future `tulip.transys` will become simpler, a minimal addition on top of `networkx`, in the direction of [this module](https://github.com/johnyf/omega/blob/d63b1b3ad9d110748292400cae3d2d04788d8be8/omega/automata.py)).

johnyf, 2017-09-23:
The double tank example needed updating (), based on its version under `examples`. The branch `matlabsim_fixes` has been merged into `master`. The OP and any other issues on the Python side have been addressed. All examples run successfully up to the export stage.

I recommend opening a pull request based off branch `master`, with the fixes on Matlab's side mentioned above, so that this issue be closed. There is no time pressure for this, though sooner is better than later, while we remember the details. Further additions can always be done in future pull requests.

johnyf, 2017-09-25:
Relevant to this, and other, interfaces: [Matlab supports calling Python code](https://www.mathworks.com/help/matlab/getting-started-with-python.html) almost as if from within a Python interpreter ([with a prefix `py.`](https://www.mathworks.com/help/matlab/matlab_external/create-object-from-python-class.html)). This means that our existing infrastructure can be used directly from Python. In principle, polytopic operations could be done by GLPK in the background, and other supporting packages used as needed.

Tkabb, 2017-09-25:
Hi, Ok so I will gather all the Matlab fixes in a pull request. Thanks for the python in Matlab advice, I am already using it.

In the python side, there rests only that when using gr1c in python code for synthesis, an error is raised in Matlab code (In particular after executing the following [line](https://github.com/tulip-control/tulip-control/blob/e4258ac28a093e4c8c39099daeeff8238d84a30f/contrib/matlabsim/load_tulip.m#L246) ). As gr1c gives a memory position as names for states, in the mentioned line the state will have naming of an array of doubles instead of one double. So I would suggest to use "omega" in the python side instead, at least for this contribution, as it then works perfectly in Matlab.

In Matlab, I have managed to make the discrete example work in Matlab with few changes. However, for the continues one, which is more interesting for me, I am having an error while simulating that I am not able to solve, I will post it here to see if someone could help or someone in the future bump with it:

```
An error occurred while running the simulation and the simulation was terminated
Caused by:
Error due to multiple causes.
Matrix dimensions must agree.
Error evaluating MATLAB function in 'Robot_Continuous_old/RHC/RHC Input'
```

For the `fuel_tank`, it worked will and I have exported the controller to Matlab. But when simulating I get the following error:

```
An error occurred while running the simulation and the simulation was terminated
Caused by:
Error due to multiple causes.
Subscript indices must either be real positive integers or logicals.
Error evaluating MATLAB function in 'Fuel_Tank/RHC/RHC Input'
```

Those errors were raised after I have made the fixes that I will propose in the pull request.

I will close the issue if no comments are made on the Matlab errors.

Tkabb, 2017-09-28:
So, for the Matlab errors I managed to solve them but I am not sure what was the main source.

For the continuous case, the problem I believe it is the MPT solver. This is because when I implemented the matlab code with the exported .m file in Matlab with MPT that has gurobi solver, the simulation ran correctly. 
I noted this problem with MPT when I check the regions structure that were built from the Tulip object (.m file) and some regions had been marked redundant V-representation and had empty vertex array. This happens in [load_continuous.m]( tulip-control/contrib/matlabsim/load_continuous.m) in particular when calculating the Polyhedron function.

For the switched case, I solved the last error by setting output initial values for the transition of the initial state, this because at the beginning of the simulation those values are set to zero which is not correct. But even though the following error is raised because the solver some how did not find the control sequence:

```
Error in 'Fuel_Tank/RHC/RHC Input'. Evaluation of expression resulted in an invalid output. Only finite double vector or matrix outputs are supported
Component: Simulink | Category: Block error
```

I will not be able to continue on this, so I will close this issue until someone figures it out.

closed 2017-09-28 by Tkabb