tulip-control/tulip-control #191: something wrong with 'transys.tuple2fts'

zexiangliu, 2018-02-13:
https://github.com/tulip-control/tulip-control/blob/4dbe152bcf44da6cc91747c8318cccc0d61f9ac0/tulip/transys/transys.py#L583

The data type of ts.actions is dict, and when the function 'transys.tuple2fts' is called, an error pointing to the line above says that 'TypeError: unsupported operand type(s) for |=: 'dict' and 'dict''

matteolucchi, 2018-02-14:
I also get the same error when I try to add actions to a FiniteTransitionSystem() with:
`r.actions |= ['think', 'write']`
I get
_TypeError: unsupported operand type(s) for |=: 'dict' and 'list'_

Could this be something related to the fact that I am running it in python 2.7?

johnyf, 2018-02-14:
The attribute `FiniteTransitionSystem.actions` appears to have been a `MathSet` (which behaves similarly to a `set`) but then changed to a `dict` [1], leading to this error. The docstring of the function `tuple2fts` is outdated:

https://github.com/tulip-control/tulip-control/blob/4dbe152bcf44da6cc91747c8318cccc0d61f9ac0/tulip/transys/transys.py#L502

Locally one could change the failing statement to `ts.actions.update(actions)`, assuming that a `dict` is passed as`actions`. However, the code may be outdated at other locations too. Another approach is to not use the function `tuple2fts`.

[1] https://github.com/tulip-control/tulip-control/commit/582183049118e9b0e0c834980877ceb8a1cbc058#diff-087a578768555f9cd3adea5aaade5803L219

slivingston, 2018-06-13:
@johnyf shall we mark `tuple2fts` as deprecated? In this case, the recommended method of constructing FTS objects is as documented in the docstring of `FiniteTransitionSystem` and as in the examples of [examples/developer/transys/transys_basic_examples.py](https://github.com/tulip-control/tulip-control/blob/afa152d8fa044b797f9681c7fd976a3814a59b75/examples/developer/transys/transys_basic_examples.py)