tulip-control/tulip-control #194: `epydoc` does not support Python 3

johnyf, 2018-02-20:
The API documentation uses `epydoc`:

https://github.com/tulip-control/tulip-control/blob/f704eaec18335a9bdbb4ad279c854756a10090e6/doc/Makefile#L12

but `epydoc` is written in Python 2:

```python
>>> from epydoc.cli import cli
Traceback (most recent call last):
...
  File ".../.virtualenvs/dev/lib/python3.6/site-packages/epydoc/cli.py", line 428
    print '\n' + msg.rstrip() + '\n'
             ^
SyntaxError: invalid syntax
```

Alternatives to consider:

- https://github.com/BurntSushi/pdoc
- https://pypi.python.org/pypi/sphinx-epytext/

slivingston, 2018-06-13:
I tried both of the alternatives using Python 3.6.5, and I recommend that we use `sphinx-epytext`. The main advantages of `sphinx-epytext` are the following:

1. We can finally move the API documents into the User Guide, which makes them easier to discover and reference together, and which provides a common Web site design.

2. The conversion seems to require few changes to render the existing markup in Tulip docstrings. I found that parameters that are listed only with `@type` are missed, but `@param` with a corresponding `@type` and only `@param` (no `@type`) are found. I tried to run it with
```
.. automodule:: tulip.synth
   :members:
```

To compare, I tried `pdoc tulip --html`. The main advantage of `pdoc` is that it supports Markdown and has a more concise presentation because it is not embedded in the general context of a reST document (the Sphinx-generated pages).

slivingston, 2018-06-13:
If there are no other recommendations about alternatives to consider, then I will create a pull request that switches us to `sphinx-epytext`.