tulip-control/tulip-control #207: Triangular Convex Region with Polytope

hashim19, 2018-12-27:
Hey Everyone, 
I am very new to TULIP and polytope. For my problem I want to define a triangular region instead of a rectangular region as you guys have explained in the tutorials and examples. What I wanna do is described in this image
[image]
I want green to be given proposition (let's say A) and blue part to B. But as in the tutorial we can only define rectangular regions or as half planes but I am not what I am doing is right or not. This is what I am doing 
# Continuous state space
cont_state_space = box2poly([[0., 5.], [0., 5.]])
#This is how I am defining the lower part of the region i.e. the blue part

cont_props = {}
A = np.array([[1.0, -1.0],
              [-1.0, -0.0],
              [-0.0, -1.0]])

b = np.array([0.0, 0.0, 0.0, 0.0])

p = pc.Polytope(A, b)

# But when I print p, it gives
Single polytope 
  [[ 0.70711 -0.70711] |    [[0.]
   [-1.      -0.     ]         x <=  [0.]
   [-0.      -1.     ]]          |     [0.]]

P.S. This diagonal line line is x1 = x2. So this gives me x1 - x2 = 0. Converting to matrix form gives the above form.

Thanks in advance

slivingston, 2019-01-01:
What is your question?

Are you asking why the first row of the matrix is `0.70711 -0.70711` instead of `1 -1` ?

hashim19, 2019-01-01:
Thank you for your response. 
Actually, I am not sure whether I am doing it right or not. and yes why does it change to 0.70711 -0.70711 instead of 1 -1 in the first row of the matrix?

slivingston, 2019-01-06:
I read your opening post (OP) again, but I still do not understand fully what you want to do. Thus, I cannot answer whether you are doing it right or not. If you want to pursue this, can you provide a minimal but complete example that demonstrates "it" (i.e., what you want to do)? E.g., share it via a Python program in a GitHub Gist <https://gist.github.com/>.

The change in elements of the `A` matrix is from normalization that occurs during instantiation. The corresponding code is in the definition of [`Polytope.__init__` at lines 123 - 133](https://github.com/tulip-control/polytope/blob/82e492b7634bcf0435c265d0c95c4f838abb42b3/polytope/polytope.py#L123-L133).

slivingston, 2019-01-26:
@hashim19: given no response from you for > 14 days, I will close this issue. Feel free to re-open if you want more help, or [join the users mailing list](https://sourceforge.net/projects/tulip-control/lists/tulip-control-users).

closed 2019-01-26 by slivingston