tulip-control/tulip-control #209: continuous_with_simulation.py gives error

hashim19, 2019-01-01:
Hello everyone,
I am trying to run continuous with simulation example and it is giving me the following error. Can anyone tell me what can be the issue?
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-33-f7841238a7a3> in <module>()
----> 1 s0_part = find_controller.find_discrete_state([x[0], y[0]], disc_dynamics.ppp)
      2 ctrl = synth.determinize_machine_init(ctrl, {'loc': s0_part})
      3 (s, dum) = ctrl.reaction('Sinit', {'park': randParkSignal[0]})
      4 print(dum)

/home/hashim/anaconda2/envs/LTL-Verification/lib/python2.7/site-packages/tulip/abstract/find_controller.pyc in find_discrete_state(x0, part)
    526     """
    527     for (i, region) in enumerate(part):
--> 528         if pc.is_inside(region, x0):
    529             return i
    530     return None

/home/hashim/anaconda2/envs/LTL-Verification/lib/python2.7/site-packages/polytope/polytope.pyc in is_inside(polyreg, point, abs_tol)
    965         'calling this function.',
    966         DeprecationWarning)
--> 967     return polyreg.__contains__(point, abs_tol)
    968 
    969 

TypeError: __contains__() takes exactly 2 arguments (3 given)

#Also, I don't know whether the above error is related to the following error but still, I want to mention it here. When I try to do 
import polytope as pc
it give me this error.
No handlers could be found for logger "polytope.solvers"

#Also I searched for **__contains__()** function inside polytope and I think **contains()** should have been used instead of "__contains__" but I am new to python and Tulip and not sure about it. 

Thanks in advance

slivingston, 2019-01-06:
when you write "run continuous with simulation example", are you referring to [examples/continuous_with_simulation.py](https://github.com/tulip-control/tulip-control/blob/afa152d8fa044b797f9681c7fd976a3814a59b75/examples/continuous_with_simulation.py)?

if so, can you tell me the following?
* version of Python (it seems to be 2.7, but I want to confirm)
* version of TuLiP
* version of Polytope

You should be able to get all of these from the command-line via

    python -c 'import sys; print(sys.version)'
    python -c 'import tulip; print(tulip.__version__)'
    python -c 'import polytope; print(polytope.__version__)'

slivingston, 2019-01-06:
I do not understand your comment:
> #Also I searched for contains() function inside polytope and I think contains() should have been used instead of "contains" but I am new to python and Tulip and not sure about it.

Can you send a patch instead of using prose?

hashim19, 2019-01-06:
These are the results for above commands

```
python -c 'import sys; print(sys.version)'
   2.7.15 |Anaconda, Inc.| (default, Dec 14 2018, 19:04:19) 
   [GCC 7.3.0]
python -c 'import tulip; print(tulip.__version__)'
    1.3.0
python -c 'import polytope; print(polytope.__version__)'
    No handlers could be found for logger "polytope.solvers"
    0.2.1

```

hashim19, 2019-01-06:
> I do not understand your comment:
> 
> > #Also I searched for contains() function inside polytope and I think contains() should have been used instead of "contains" but I am new to python and Tulip and not sure about it.
> 
> Can you send a patch instead of using prose?

I am not sure about the suggestion I made. Let me see how can I send you a patch. 

Thanks

slivingston, 2019-01-07:
thanks for telling me the version numbers. can you also tell me which "continuous with simulation example" you are trying?

For example, is it examples/continuous_with_simulation.py downloaded directly from GitHub, or did you clone the repo?

My goal is to know precisely which file you ran, so I can replicate the error.

hashim19, 2019-01-07:
> thanks for telling me the version numbers. can you also tell me which "continuous with simulation example" you are trying?
>
> For example, is it examples/continuous_with_simulation.py downloaded [...]

I am running examples/continuous_with_simulation.py after cloning the repo

slivingston, 2019-01-07:
I have not been able to get the same error. My hypothesis: the error is due to you using TuLiP and Polytope as released on PyPI (e.g., installed via `pip install`), whereas the example in some recent commit of the repo is not compatible. The sourcetree in the repo at the tip of `master` branch is under development and can be broken without notice, unlike official releases.

Assuming that you do not want to wait for the next release, and assuming that version 1.3.0 of TuLiP is sufficient for your current work, then your best option is to get the example file from that release. You can find the release sourcetree at <https://github.com/tulip-control/tulip-control/archive/v1.3.0.zip> or you can browse it on GitHub at https://github.com/tulip-control/tulip-control/tree/v1.3.0

The continuous_with_simulation.py file can be downloaded directly from the command-line:

    curl -L -O https://github.com/tulip-control/tulip-control/raw/v1.3.0/examples/continuous_with_simulation.py

hashim19, 2019-01-08:
@slivingston thank you for digging into the problem. If I try to run the this example file I have on collab, will it generate the same error?
What I understand from your comment above is that, the master branch I cloned is not compatible with the Tulip and Polytope on PyPI. 
1. I will try to run it on collab. 
2. I will try to download his example file you mentioned above and will get back to you soon.

Thanks

hashim19, 2019-01-08:
@slivingston  so now, I copied the example file from here 
[(https://github.com/tulip-control/tulip-control/tree/v1.3.0)]

it was giving me this error 

ValueErrorTraceback (most recent call last)
<ipython-input-17-6db961b29e52> in <module>()
     51     cont_partition, sys_dyn, closed_loop=False,
     52     conservative=True,
---> 53     N=5, min_cell_volume=0.1, plotit=show)
     54 # Visualize transitions in continuous domain (optional)
     55 plot_partition(disc_dynamics.ppp, disc_dynamics.ts,

/usr/local/lib/python2.7/dist-packages/tulip/abstract/discretization.pyc in discretize(part, ssys, N, min_cell_volume, closed_loop, conservative, max_num_poly, use_all_horizon, trans_length, remove_trans, abs_tol, plotit, save_img, cont_props, plot_every)
    641         S0 = solve_feasible(
    642             si, sj, ss, N, closed_loop,
--> 643             use_all_horizon, trans_set, max_num_poly
    644         )
    645 

/usr/local/lib/python2.7/dist-packages/tulip/abstract/feasible.pyc in solve_feasible(P1, P2, ssys, N, closed_loop, use_all_horizon, trans_set, max_num_poly)
    135             P1, P2, ssys, N,
    136             trans_set=trans_set,
--> 137             max_num_poly=max_num_poly
    138         )
    139 

/usr/local/lib/python2.7/dist-packages/tulip/abstract/feasible.pyc in solve_open_loop(P1, P2, ssys, N, trans_set, max_num_poly)
    275     for p1 in start_polys:
    276         for p2 in target_polys:
--> 277             cur_s0 = poly_to_poly(p1, p2, ssys, N, trans_set)
    278             s0 = s0.union(cur_s0, check_convex=True)
    279 

/usr/local/lib/python2.7/dist-packages/tulip/abstract/feasible.pyc in poly_to_poly(p1, p2, ssys, N, trans_set)
    298     dims = range(1, n+1)
    299 
--> 300     s0 = s0.project(dims)
    301 
    302     return pc.reduce(s0)

/usr/local/lib/python2.7/dist-packages/polytope/polytope.pyc in project(self, dim, solver, abs_tol, verbose)
    352         For usage details see function: L{projection}.
    353         """
--> 354         return projection(self, dim, solver, abs_tol, verbose)
    355 
    356     def scale(self, factor):

/usr/local/lib/python2.7/dist-packages/polytope/polytope.pyc in projection(poly1, dim, solver, abs_tol, verbose)
   1644     else:
   1645         logger.debug("projection: using iterative hull.")
-> 1646         return projection_iterhull(poly1, new_dim)
   1647 
   1648 

/usr/local/lib/python2.7/dist-packages/polytope/polytope.pyc in projection_iterhull(poly1, new_dim, max_iter, verbose, abs_tol)
   1929             OK = 1
   1930             for i in xrange(np.shape(Vert)[0]):
-> 1931                 if not P1.contains([Vert[i, new_dim]], abs_tol=1e-5):
   1932                     # If all new points are inside
   1933                     # old polytope -> Finished

/usr/local/lib/python2.7/dist-packages/polytope/polytope.pyc in contains(self, points, abs_tol)
    207         @rtype: bool, 1d array
    208         """
--> 209         test = self.A.dot(points) - self.b[:, np.newaxis] < abs_tol
    210         return np.all(test, axis=0)
    211 

ValueError: shapes (3,2) and (1,2) not aligned: 2 (dim 1) != 1 (dim 0)

Then, I changed closed_loop = true in the following line
`disc_dynamics = discretize(
    cont_partition, sys_dyn, closed_loop=False,
    conservative=True,
    N=5, min_cell_volume=0.1, plotit=show)`

Then it started giving the following error as mentioned in the original [question](https://github.com/tulip-control/tulip-control/issues/209#issue-395098365) 

TypeErrorTraceback (most recent call last)
<ipython-input-5-e92fe88e0792> in <module>()
     98 N = disc_dynamics.disc_params['N']
     99 s0_part = find_controller.find_discrete_state(
--> 100     [x[0], y[0]], disc_dynamics.ppp)
    101 ctrl = synth.determinize_machine_init(ctrl, {'loc': s0_part})
    102 (s, dum) = ctrl.reaction('Sinit', {'park': randParkSignal[0]})

/usr/local/lib/python2.7/dist-packages/tulip/abstract/find_controller.pyc in find_discrete_state(x0, part)
    526     """
    527     for (i, region) in enumerate(part):
--> 528         if pc.is_inside(region, x0):
    529             return i
    530     return None

/usr/local/lib/python2.7/dist-packages/polytope/polytope.pyc in is_inside(polyreg, point, abs_tol)
    965         'calling this function.',
    966         DeprecationWarning)
--> 967     return polyreg.__contains__(point, abs_tol)
    968 
    969 

TypeError: __contains__() takes exactly 2 arguments (3 given)

This error is in the simulation part. So, I consider closed_loop = False was a mistake. 
I am also sharing colab link here. 
[colab-link](https://colab.research.google.com/drive/1h9wE-IH2ZexVjyCQDuvxhY7ctVIWrlJq#scrollTo=bdq6XJR5uEOW)

slivingston, 2019-01-12:
can you share the Jupyter notebook via gist.github.com ?
Running in Colab requires me to sign-in with a Google account, which I do not want to do.

slivingston, 2019-01-31:
@hashim19 given no response > 14 days and because I have not able to reproduce the error that you described, I will assume that you found a solution for your deployment. Feel free to re-open this issue if the error remains, or we can debug your code [via the users mailing list](https://sourceforge.net/projects/tulip-control/lists/tulip-control-users).

In any case, we will work toward a new release as soon as possible, which might help you [as described above](https://github.com/tulip-control/tulip-control/issues/209#issuecomment-452051870).

closed 2019-01-31 by slivingston