tulip-control/tulip-control #210: migrate to travis-ci.com

slivingston, 2019-04-04:
As presented [on their website](https://docs.travis-ci.com/user/migrate/open-source-on-travis-ci-com/), open source projects will eventually be moved to travis-ci.com. The task is to complete this migration as described at https://docs.travis-ci.com/user/migrate/open-source-repository-migration

checklist for doing this in tulip-control related repositories:
- [ ] Tulip (here)
- [ ] dd (https://github.com/tulip-control/dd)
- [ ] Polytope (https://github.com/tulip-control/polytope)

slivingston, 2020-10-01:
actually, this is still in "beta" as described at https://docs.travis-ci.com/user/migrate/open-source-repository-migration
so I will close this issue and re-open when the time is right.

slivingston, 2021-05-31:
The future is upon us. I just migrated this repository to travis-ci.com and pushed 2 small documentation-only commits that successfully triggered CI jobs: https://travis-ci.com/github/tulip-control/tulip-control/builds/227465703

johnyf, 2021-06-03:
Thank you for migrating the CI of the repository to travis-ci.com. The transition of the CI for the repositories `gr1c`, `tulip-control`, `polytope`, `omega`, and `dd` to travis-ci.com has now been completed. The repositories `cudd` and `data` also appear on `travis-ci.com`, I do not know why (when I migrated the CI entries of other repositories, I remember that I only selected the specific repositories I was about to migrate). In any case, regarding the last item in the checklist above (https://github.com/tulip-control/tulip-control/issues/210#issue-429080777), I think that there is no issue with having these additional repositories present in travis-ci.com.

slivingston, 2021-06-04:
I accidentally added the repositories `data` and `cudd`. I removed them now.

closed 2021-06-04 by slivingston

johnyf, 2021-06-04:
Thank you for removing the two repositories from travis-ci.com.