tulip-control/tulip-control #211: syntax for initializing variables with a range of discrete values

abadithela, 2019-08-18:
Hi, 

I am trying to implement an example similar to the Discrete State Robot Motion Planning example given [here](https://tulip-control.sourceforge.io/doc/tutorial.html). My example has a bigger gridworld with an additional moving obstacle that the system is trying to avoid while it continually reaches its goal states. Right now, I'm initializing each grid cell as a state for both the system and the obstacle. I know that there is a better way to do this -- variables don't have to just be Boolean but they can hold a range of discrete values. However, I'm not sure how the syntax works for this. For example, do I instantiate env variables with discrete values like this: env_vars = {'obs' : {'1', '2', '3'}}? 

Thanks!

murrayrm, 2019-08-18:
@johnyf @slivingston I thought I remembered there was a way to do this, but can't find  anything in the documentation (?).  Perhaps this was something that `omega` supported?

necozay, 2019-08-18:
See line 63 in:

https://github.com/tulip-control/tulip-control/blob/master/examples/gr1_set.py

abadithela, 2019-08-19:
Thanks everyone!

slivingston, 2019-10-29:
@abadithela is your question therefore answered?

slivingston, 2019-10-29:
while we wait for confirmation from @abadithela , I recommend that the task of this ticket is to revisit documentation and decide how easily someone with this question in the future can find an answer.

abadithela, 2020-03-19:
Hi Scott, my question has been answered. Sorry for the delay.

closed 2020-03-19 by abadithela