tulip-control/tulip-control #219: LTL2BA parser has problem parsing `skip`

mhtb32, 2020-01-21:
Hi!

The module ltl2ba.py cannot parse `skip`, which is equivalent to true. Did you not include the parsing rule for `skip` on purpose?
I can make a pull request if you mind.

slivingston, 2020-02-26:
I do not understand. @mhtb32 can you give an example to illustrate?

Also, can you tell me which version of Tulip that you are using?  You can find it via

    python -c 'import tulip; print(tulip.__version__)'

mhtb32, 2020-02-26:
I was using `ltl2ba.py` file separately like:
```
python ltl2ba.py
```
and I replaced the default formula
```
f = '[] !s && <> (a || c) && <> (b U c)'
```
with
```
p1 <> p2
```
Then I received this error: 
```
ERROR:__main__:Syntax error at skip
```
I have a [fork](https://github.com/mhtb32/foggy-planner/blob/master/foggy/ltl2ba.py) of your code using [sly](https://github.com/dabeaz/sly) package(a successor to ply) which I have a parsing rule for skip to create a self transition like this:
```
    @_('state ":" SKIP')
    def clause(self, p):
        u = p.state
        self.automaton.add_transition({(u, u, '1')})
        return u, p.SKIP
```