tulip-control/tulip-control #221: upgrade NetworkX dependency

slivingston, 2020-02-08:
Upgrading to a recent version of NetworkX is generally motivated to use recent features, bug fixes, etc. While there is some risk that users of Tulip are pinned to Networkx 1 and cannot work with Tulip if dependency switches to Network >= 2.4, I think it is best to allow them to open issues to motivate creating a compatibility wrapper. Otherwise, the most recent release of the 1.x series is [1.11 from 4 years ago](https://pypi.org/project/networkx/1.11/).

related work: https://github.com/tulip-control/omega/pull/10

slivingston, 2020-02-08:
also, there is work in progress by @tichakorn832 at https://github.com/tulip-control/tulip-control/pull/220

murrayrm, 2020-02-21:
Resolved in #220.

closed 2020-02-21 by murrayrm