tulip-control/tulip-control #222: Extra states created in omega generate warnings when synthesizing

murrayrm, 2020-02-17:
When running `examples/discrete.py`,  there are a bunch of warnings generated:
```
The following edge attributes:
{'_goal': 1}
are not allowed.
Currently the allowed attributes are:park, X0reach, home, lot, loc
Allowed because you passed: check = True
```
This message is generated about 30 times and appears to occur in the `synth._trim_strategy` function.  I did a bit of looking around and it seems like `omega` is creating an output `_goal` and then  `_trim_strategy` complains because this is not a variable it recognizes.

I think the strategy that is generated is still OK, but it would be nice to turn off the (spurious?) warning message.

@johnyf: any thoughts on this?

johnyf, 2020-02-17:
The warnings are generated from the function `tulip.transys.labeled_graphs.LabeledDiGraph._check_for_untyped_keys` which is called from the function `tulip.synth.strategy2mealy`, which is called from the function `tulip.synth._trim_strategy`.

https://github.com/tulip-control/tulip-control/blob/f0001a1cf26776ab7c0db587f7f5f79d2e6e808f/tulip/transys/labeled_graphs.py#L851-L857

https://github.com/tulip-control/tulip-control/blob/f0001a1cf26776ab7c0db587f7f5f79d2e6e808f/tulip/synth.py#L1132

https://github.com/tulip-control/tulip-control/blob/f0001a1cf26776ab7c0db587f7f5f79d2e6e808f/tulip/synth.py#L1239

The two statements that raise this warning are

https://github.com/tulip-control/tulip-control/blob/eee6daf15ab7c2c40d64a8642dc8336050a3cdb8/tulip/synth.py#L1274

and

https://github.com/tulip-control/tulip-control/blob/eee6daf15ab7c2c40d64a8642dc8336050a3cdb8/tulip/synth.py#L1304

One possibility would be to filter the dictionary of transition attributes before passing it to the method `tulip.transys.labeled_graphs.Transitions.add`:

```diff
diff --git a/tulip/synth.py b/tulip/synth.py
index deaafaa..7d5a6d0 100644
--- a/tulip/synth.py
+++ b/tulip/synth.py
@@ -1266,10 +1266,13 @@ def strategy2mealy(A, spec):
         k: v for k, v in sys_vars.items()
         if isinstance(v, list)})
     mach.states.add_from(A)
+    all_vars = dict(env_vars)
+    all_vars.update(sys_vars)
     # transitions labeled with I/O
     for u in A:
         for v in A.successors(u):
             d = A.nodes[v]['state']
+            d = {k: v for k, v in d.items() if k in all_vars}
             d = _int2str(d, str_vars)
             mach.transitions.add(u, v, attr_dict=None, check=False, **d)

@@ -1300,6 +1303,7 @@ def strategy2mealy(A, spec):
         # add edge: Sinit -> u ?
         tmp.update(var_values)
         if eval(isinit, tmp):
+            var_values = {k: v for k, v in var_values.items() if k in all_vars}
             label = _int2str(var_values, str_vars)
             mach.transitions.add(initial_state, u, attr_dict=None, check=False, **label)
             # remember variable values to avoid
```

I pushed this change (and a typo correction mentioned below) to the branch `filter_transition_attributes`, which extends the branch `fix/py3-compatibility` of https://github.com/tulip-control/tulip-control/pull/220.

Another possibility would be to change the level of the logging message to `logging.DEBUG`:

```diff
diff --git a/tulip/transys/labeled_graphs.py b/tulip/transys/labeled_graphs.py
index 97e877f..7686c47 100644
--- a/tulip/transys/labeled_graphs.py
+++ b/tulip/transys/labeled_graphs.py
@@ -859,8 +859,8 @@ class LabeledDiGraph(nx.MultiDiGraph):
                         'in the existing types, pass: check = False')
                 raise AttributeError(msg)
             else:
-                msg += '\nAllowed because you passed: check = True'
-                logger.warning(msg)
+                msg += '\nAllowed because you passed: check = False'
+                logger.debug(msg)
         else:
             logger.debug('no untyped keys.')
```

(Also, `check = True` in the warning message would need to be replaced with `check = False`.)

murrayrm, 2020-02-17:
Thanks for jumping on this @johnyf.  The fix looks good to me.

@tichakorn832 You may want to incorporate this change into your current PR.

murrayrm, 2020-02-21:
Resolved in #223

closed 2020-02-21 by murrayrm