tulip-control/tulip-control #231: populate `__init__.py`

johnyf, 2020-03-02:
Add imports to `tulip/__init__.py` of modules and ensure that direct calls to functions work, for example `tulip.discretize` and `tulip.synthesize`.

murrayrm, 2020-03-02:
Interestingly, there is a line that imports the main modules in `__init__.py`, but it is commented out.  Does anyone remember why?

johnyf, 2020-03-02:
It appears to have been commented out to reduce runtime in the commit https://github.com/tulip-control/tulip-control/commit/7e05aeac30d3d24db2cbc00154c5b80ad2bc2743#diff-4505ab709ad2fcabf4c351b5c4084ea5R47, which changed some imports of `matplotlib` and `ipython` to be inline. `matplotlib` is now (ce54897c242689f45ad33650f157bf1805b35ed6) imported inline, except in the module `tulip.graphics`

https://github.com/tulip-control/tulip-control/blob/ce54897c242689f45ad33650f157bf1805b35ed6/tulip/graphics.py#L53-L58

slivingston, 2020-03-05:
I am reviewing PR #234, but I think that this issue is the best place for my question: is this change sufficiently motivated to justify the increase in time required to `import`?

As I understand it, populating the root `__init__.py` is motivated to make `import` statements shorter in practice. E.g., `from tulip import GRSpec` is shorter than `from tulip.spec import GRSpec`. Being shorter is better because it requires less time to write code and because it is easier to remember.

However, populating `__init__.py` has the disadvantage of increasing the time to import subpackages of TuLiP because all or most subpackages will always be imported, independently of whether they are used.

As a concrete example, with the changes of PR #234, the time required to execute `examples/gridworlds/genrand.py` is approximately 1.6 s. If I modify that PR to perform no imports in the root `__init__.py`, then `examples/gridworlds/genrand.py` executes within approximately 450 ms.

slivingston, 2020-03-05:
One counter-argument in favor of allowing the increased time to `import` is that for problems of nontrivial size, the difference of 1 or 2 seconds is not a significant proportion of total execution time.

However, small examples like `genrand.py` ([mentioned above](https://github.com/tulip-control/tulip-control/issues/231#issuecomment-594952997).) are nonetheless interesting for education and getting started with TuLiP.

slivingston, 2020-03-05:
To conclude, there are at least 3 options:
1. remove all internal imports from the root `__init__.py`,
2. measure the cost introduced by each `import`, observe the distribution to find `import` statements that involve much more time than others, and if found, decide whether to remove the offenders,
3. keep all of the `import` statements, close this issue, and in future work explore whether subpackages can be refactored to reduce the time to `import tulip`.

slivingston, 2022-06-19:
done in https://github.com/tulip-control/tulip-control/pull/234

closed 2022-06-19 by slivingston