tulip-control/tulip-control #233: Bug in omega interface: action_to_steps() got multiple values for argument 'qinit'

murrayrm, 2020-03-03:
Something got changed in `tulip` or `omega` and the following code now generates an error:
```
from tulip import transys, spec, synth

# Create a finite transition system
sys = transys.FTS()

# Define the states of the system
sys.states.add_from(['C0', 'C1', 'C2', 'C3', 'C4', 'C5'])
sys.states.initial.add('C0')    # start in state C0

# Define the allowable transitions
sys.transitions.add_comb({'C0'}, {'C1', 'C3'})
sys.transitions.add_comb({'C1'}, {'C0', 'C4', 'C2'})
sys.transitions.add_comb({'C2'}, {'C1', 'C5'})
sys.transitions.add_comb({'C3'}, {'C0', 'C4'})
sys.transitions.add_comb({'C4'}, {'C3', 'C1', 'C5'})
sys.transitions.add_comb({'C5'}, {'C4', 'C2'})

# Add atomic propositions to the states
sys.atomic_propositions.add_from({'home', 'lot'})
sys.states.add('C0', ap={'home'})
sys.states.add('C5', ap={'lot'})

#232  Create specifications
env_vars = {'park'}
env_init = []                   # empty set
env_prog = '!park'
env_safe = set()                # empty set (another way)
sys_vars = {'X0reach'}          # infer the rest from TS
sys_init = {'X0reach'}
sys_prog = {'home'}             # []<>home
sys_safe = {'(X (X0reach) <-> lot) || (X0reach && !park)'}
sys_prog |= {'X0reach'}
specs = spec.GRSpec(env_vars, sys_vars, env_init, sys_init,
                    env_safe, sys_safe, env_prog, sys_prog)

specs.moore = True
# synthesizer should find initial system values that satisfy
# `env_init /\ sys_init` and work, for every environment variable
# initial values that satisfy `env_init`.
specs.qinit = '\E \A'
ctrl = synth.synthesize(specs, sys=sys)
```
Error is
```
     60     gr1.make_streett_transducer(z, yij, xijk, aut)
     61     t2 = time.time()
---> 62     g = enum.action_to_steps(aut, 'env', 'impl', qinit=aut.qinit)
     63     h = _strategy_to_state_annotated(g, aut)
     64     del z, yij, xijk

TypeError: action_to_steps() got multiple values for argument 'qinit'
```
@johnyf Clues?

johnyf, 2020-03-03:
This error could be due to `omega == 0.2.1` installed. The function `action_to_steps` has changed in `omega == 0.3.1` to take additional arguments. This code runs without errors with `omega == 0.3.1` installed.

murrayrm, 2020-03-03:
Confirming that my local copy is `omega == 0.2.0` for some reason (thought I had installed the latest version...).  Will work on this and close when fixed.  Thanks for the quick response.

murrayrm, 2020-03-03:
Found the problem: when I installed an older version of TuLiP using `setup.py`, it looks like it pip installed an old version of `omega` and this was getting picked up instead of the latest version.  Now fixed.

closed 2020-03-03 by murrayrm