tulip-control/tulip-control #240: about the slugs interface

he-z-r, 2020-06-05:
Hi,
When I was trying to use the slugs interface for GR(1) synthesis, I got the following error which I could not understand. I am wondering what kind of issues is this. Is this an error of not running in python2 environment? I referred to the slugs documentation and it said that its python scripts are in python2. But the latest version of TuLiP was in python3. I am not sure whether I got this correct. I'd appreciate it if someone can help me with this.

Error message:
subprocess.CalledProcessError: Command '['python2', '/home/xxxxxx/.local/bin/slugs/tools/StructuredSlugsParser/compiler.py', '/tmp/tmp7mju9z7u']' returned non-zero exit status 1.

murrayrm, 2020-06-05:
We haven't been actively maintaining the slugs interface, so I am not sure that it works with the latest master.

A couple of things to try:

* Make sure there is a command 'python2' installed on your system, since this is what it looks like Slugs is trying to call.

* Try checking out an earlier version of `tulip` to see if you can get back to something that works.

It would help to let us know what you try, what works, and what doesn't.  Also, if you can tell us the version of Python that you are using and the version of TuLiP that you are using, that would help.

he-z-r, 2020-06-09:
> We haven't been actively maintaining the slugs interface, so I am not sure that it works with the latest master.
>
> A couple of things to try:
>
> * Make sure there is a command 'python2' installed on [...]

I am trying to include a "next" operator in the system liveness properties sys_prog. The default solver does not support this, which made me turn to slugs. 
The python version I am using is 3.6.9. And I am using the latest version of TuLiP.

murrayrm, 2020-06-21:
Can you post some code showing a simple example of the failure and then I can test on my local machine and see if I can figure out what is happening.