tulip-control/tulip-control #241: confusing assertion error

necozay, 2020-06-12:
It looks like when a spec is trivially realizable by rendering the environment initial condition or environment safety false, the ``synthesize`` function, with default options, returns an assertion error: e.g., ```AssertionError: removed all NUMBER nodes!```. It looks like in this case the solver generates an automaton where all paths lead to a dead end and trimming this automaton returns an empty controller, hence the error. However this error message is confusing for new users. I  suggest updating the error to something like ```AssertionError: removed all NUMBER nodes! Please check env_init and env_safety to avoid trivial realizability.```  If you have any thoughts on this (and on how to make the error message more informative), please let me know. Otherwise, I will send in a pull request in a week.

Minimal example:
```
env_vars = {'e'}
env_init = set()                
env_init |= {'e'}
env_prog = set()
env_safe = {'e -> X(! e)', '!e -> s'}
sys_vars = {'s'}
sys_init = set() 
sys_prog = set() 
sys_safe = set() 
specs = spec.GRSpec(env_vars, sys_vars, env_init, sys_init,
                    env_safe, sys_safe, env_prog, sys_prog)
specs.qinit = '\A \E'  # Moore initial condition synthesized too
specs.moore = False
specs.plus_one = False
ctrl = synth.synthesize(specs, solver='gr1c', rm_deadends = 1)
```

murrayrm, 2020-06-12:
Definitely a good idea to add some more information to the error message. @tichakorn832 and I noticed that there are some pretty opaque messages that get returned.

slivingston, 2020-10-01:
done in https://github.com/tulip-control/tulip-control/pull/243

closed 2020-10-01 by slivingston