tulip-control/tulip-control #246: updating the binaries used in CI tests for the dependency `storm`

johnyf, 2021-06-03:
On the branch `ci_update_download` I have updated the CI configuration and dependencies (this was motivated by `bintray.com` being discontinued, as explained in a section below). The only remaining error:

https://travis-ci.com/github/tulip-control/tulip-control/jobs/510638673#L15866

reads:

```
File "/home/travis/build/tulip-control/tulip-control/tests/stormpy_interface_test.py", line 25, in <module>
    from tulip.interfaces import stormpy as stormpy_int
  File "/home/travis/virtualenv/python3.9.1/lib/python3.9/site-packages/tulip/interfaces/stormpy.py", line 39, in <module>
    import stormpy
  File "/home/travis/build/tulip-control/tulip-control/stormpy-1.6.2/lib/stormpy/__init__.py", line 6, in <module>
    from . import core
ImportError: libboost_filesystem.so.1.65.1: cannot open shared object file: No such file or directory
```

This error appears to be from [`storm`](https://github.com/moves-rwth/storm), to which [`stormpy`](https://github.com/moves-rwth/stormpy) interfaces. `stormpy` is built on the CI run:

https://github.com/tulip-control/tulip-control/blob/ece84cdccb4deb4cb14b144df7e5b7f5697eb869/extern/get-stormpy.sh#L79

so I would expect that it links (if it links at all) to the version of `libboost_filesystem.so` that is present on `focal` (and thus not raise the above error. `stormpy` is built using [`pybind11`](https://github.com/pybind/pybind11), and the [`core`](https://github.com/moves-rwth/stormpy/tree/1.6.2/src/core) mentioned in the traceback above is a C++ extension that is [built with](https://github.com/moves-rwth/stormpy/blob/1.6.2/src/mod_core.cpp#L14) `pybind11`). That the error is from `storm` is confirmed by attempting to run

[`storm --version`](https://www.stormchecker.org/documentation/usage/running-storm.html#first-steps)

in the CI run, which [raises](https://travis-ci.com/github/tulip-control/tulip-control/jobs/510660367#L15483):

```
./storm/build/bin/storm: error while loading shared libraries: libboost_system.so.1.65.1: cannot open shared object file: No such file or directory
```

(`libboost_filesystem.so.1.65.1` was the dependence in the traceback from above, but that was from `stormpy`, which is linked to `storm` when built, so which libraries are loaded from `stormpy`, and in which order, can differ from which libraries are loaded, and in which order, when calling `storm --version`.)

Indeed, the dependence on `libboost_filesystem` from above is confirmed as follows after unpacking the archive downloaded from https://sourceforge.net/projects/cudd-mirror/files/storm.tar.xz/download and `cd` inside it:

```
> pwd
.../storm/build/lib
> strings * | ag libboost_filesystem
libboost_filesystem.so.1.65.1
> ag --search-binary libboost_filesystem
Binary file libstorm.so matches.
```

(Using [`string`](https://en.wikipedia.org/wiki/Strings_(Unix)) and [`ag`](https://github.com/ggreer/the_silver_searcher).)

(Boost is a [requirement of `storm`](https://www.stormchecker.org/documentation/obtain-storm/dependencies.html#boost).)

Also, the error does not seem to be from the dependency of `storm`, because [`ldd carl/libcarl.so.14.20` prints in the CI run](https://travis-ci.com/github/tulip-control/tulip-control/jobs/510660367#L2586-L2597):

```
linux-vdso.so.1 (0x00007ffe28fa5000)
libgmpxx.so.4 => /home/travis/build/tulip-control/tulip-control/carl/resources/lib/libgmpxx.so.4 (0x00007f6708d80000)
libgmp.so.10 => /home/travis/build/tulip-control/tulip-control/carl/resources/lib/libgmp.so.10 (0x00007f6708b0a000)
libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f6708af5000)
libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f6708ad2000)
libcln.so.6 => /home/travis/build/tulip-control/tulip-control/carl/resources/lib/libcln.so.6 (0x00007f670875b000)
libginac.so.6 => /home/travis/build/tulip-control/tulip-control/carl/resources/lib/libginac.so.6 (0x00007f67082ba000)
libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f67080d9000)
libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f6707f8a000)
libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f6707f6f000)
libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f6707d7d000)
/lib64/ld-linux-x86-64.so.2 (0x00007f670924a000)
```

whereas [`ldd storm/build/lib/libstorm.so` prints in the CI run](https://travis-ci.com/github/tulip-control/tulip-control/jobs/510660367#L15461-L15482):

```
linux-vdso.so.1 (0x00007ffd11d93000)
libcarl.so.14.20 => /home/travis/build/tulip-control/tulip-control/carl/libcarl.so.14.20 (0x00007f5bb1a68000)
libginac.so.6 => /home/travis/build/tulip-control/tulip-control/carl/resources/lib/libginac.so.6 (0x00007f5bb15c9000)
[... 56 lines cut ...]
Using this change, the [CI raised an error](https://travis-ci.org/github/tulip-control/tulip-control/jobs/772877421#L2833) that seems to relate to the version of [GLIBC](https://en.wikipedia.org/wiki/GNU_C_Library):

> ValueError: invalid literal for int() with base 10: "gr1c: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by gr1c)"

The requirement of GLIBC 2.29 is confirmed using `readelf -s gr1c-0.13.0-Linux_x86-64/gr1c`, which starts with:

```
Symbol table '.dynsym' contains 63 entries:
   Num:    Value          Size Type    Bind   Vis      Ndx Name
     0: 0000000000000000     0 NOTYPE  LOCAL  DEFAULT  UND
     1: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND free@GLIBC_2.2.5 (2)
     2: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND log2@GLIBC_2.29 (3)
```

(`greadelf` on macOS is [Readelf](https://en.wikipedia.org/wiki/Readelf), which can be installed on macOS by installing the [MacPorts](https://www.macports.org) package [`binutils`](https://ports.macports.org/port/binutils/summary), on Linux `ldd gr1c-0.13.0-Linux_x86-64/gr1c` is possible too).

So I changed the OS for the CI run to `focal` (Ubuntu 20.04), from `bionic` (Ubuntu 18.04). This [raised another error](https://travis-ci.org/github/tulip-control/tulip-control/jobs/772881862#L266):

> The command "sudo -E apt-get -yq --no-install-suggests --no-install-recommends $(travis_apt_get_options) install gfortran libatlas-base-dev liblapack-dev libgmp-dev libmpfr-dev graphviz libglpk-dev libboost-dev libboost-filesystem-dev libboost-program-options-dev libboost-regex-dev libboost-test-dev libeigen3-dev z3 libz3-dev python-z3 libhwloc-dev" failed and exited with 100 during .

This error is because the `apt` package [`python-z3`](https://packages.ubuntu.com/bionic/python-z3), which is available on `bionic`, is not available on `focal` (from the default package repository). Instead, the corresponding package on `focal` is [`python3-z3`](https://packages.ubuntu.com/focal/python3-z3). So I changed the CI configuration to install `python3-z3`.

(`tulip` now supports only Python 3, so installing `python3-z3` suffices. Even if Python 2 was still supported, and thus tested in CI, `python3-z3` is required only for testing the interface to `stormpy`. The package `stormpy` requires Python 3, so the tests for that interface were run only on Python 2. Thus, it would have sufficed to conditionally intall `python3-z3` only if the CI run was for Python 3. In any case, this conditional installation is not needed now.)

After these changes, and an update to fetch an archive of a more recent commit of `slugs` (which contains Python scripts updated to Python 3), the error from the `storm` binary remains, which is discussed above.

slivingston, 2021-06-08:
@johnyf thanks for this detailed treatment. Regarding updating and uploading built files for `storm` and `carl`, another idea is to configure https://github.com/tulip-control/data for [Git LFS](https://git-lfs.github.com/), which causes a hash of specified files to be tracked in the Git repository while the file itself is uploaded to a Git LFS conforming server. GitHub provides one at no cost.

> It will be some time before I will have access to a machine where I can build newer versions of these binaries suitable for use on Travis CI

I can work on this, but likely not until later this month.

johnyf, 2021-06-08:
Using Git LFS I think is a better approach, @slivingston thank you for suggesting it. Thank you for being available to work on updating the binaries.

johnyf, 2021-08-21:
Addressed in commit b38b525aca9d918b4737c10a1e02625cba4a49fd, by uploading new binaries for `storm` and `carl` as release assets of a GitHub release associated to a tag of the repository https://github.com/tulip-control/data.

I noted in the README of the `data` repository that tags are by default considered temporary, and in the release page what that release is for, and that the release may be replaced in the future (https://github.com/tulip-control/data/releases/tag/stormpy_dependencies).

These changes are tested successfully using GitHub Actions in commit 7ececa4bcae51c85759e7678f706c160e0fc4fa5, and have been merged to the mainline branch. So now all tests pass.

## About LFS on GitHub

I think that Git LFS is a good solution for this issue. However, GitHub's LFS storage appears to have limitations that could limit CI testing, as described below. In summary:

- downloads of Git LFS files from GitHub are limited to 1 GB in total per month
  (for free accounts). Any Git LFS activity on forks also counts towards the
  root repository owner's quotas.

- deleting old files from GitHub's LFS storage requires deleting the entire
  repository (`https://github.com/tulip-control/data`), and recreating it.

In comparison, I am not aware of limits on file size of release assets, and as far as I know, deleting a release from a GitHub repository's releases suffices to delete the release assets associated with the deleted release.

(Sidenote: The closest thing in the `git` ecosystem appears to be `git annex dropunused --force`, using [`git-annex`](https://en.wikipedia.org/wiki/Git-annex).)

## GitHub LFS storage and bandwidth limitations

There are limitations on storage bandwidth of GitHub LFS per repository owner:
https://docs.github.com/en/github/managing-large-files/versioning-large-files/about-storage-and-bandwidth-usage
It seems to me that this limitation could imply that testing on GitHub Actions may be limited by bandwidth usage.

### GitHub LFS storage limitation

The `stormpy`-related binaries currently in the `data` repository release assets have total size 76 MB. The binaries that were previously downloaded from SourceForge and used in CI had total size 72 MB.

With a 1 GB limit on GitHub LFS storage, about 13 versions of these binaries can be stored. Although these files need to change only when either the CI environment changes (e.g., newer versions of `apt` packages), or the dependencies being built are updated, when newer versions are being created, a number of iterations may be needed until the CI tests pass. These iterations would probably exhaust the 1 GB space.

### GitHub LFS bandwidth limitation

#### GitHub LFS and `git` bandwidth

The bandwidth limitation could be restrictive on CI testing, because each push triggers a CI build, which will download the files (one option might be to cache the files on GitHub Actions, and these particular files appear to be the use case that matches the reason that caching exists on GitHub Actions, though cache is cleared every 45 days), will be counted towards the bandwidth limit:

> If you choose to include Git LFS objects in archives of your repository, every download of those archives will count towards bandwidth usage for your account. Each account receives 1 GB per month of bandwidth for free, and you can pay for additional usage.

https://docs.github.com/en/github/administering-a-repository/managing-repository-settings/managing-git-lfs-objects-in-archives-of-your-repository

There are months when more than 13 pushes occur in the repository `tulip-control`, so having such a limit would make the builds fail after the limit is reached:

> If you use more than 1 GB of bandwidth per month without purchasing a data pack, Git LFS support is disabled on your account until the next month.

https://docs.github.com/en/github/managing-large-files/versioning-large-files/about-storage-and-bandwidth-usage#bandwidth-quota

#### GitHub LFS and bandwidth of repository downloads

There is an option in GitHub repository settings (in this case for the settings of the repository `data`) to include Git LFS files in repository archives (ZIP files and tarballs) that GitHub makes available for download:
https://docs.github.com/en/github/administering-a-repository/managing-repository-settings/managing-git-lfs-objects-in-archives-of-your-repository
However, including LFS files in repository archives makes the downloads of
archives count towards the repository organization's limit:

> If you choose to include Git LFS objects in archives of your repository, every download of those archives will count towards bandwidth usage for your account. Each account receives 1 GB per month of bandwidth for free, and you can pay for additional usage.

(I do not know what, if any, limit on download bandwidth GitHub has on repository downloads, e.g., release assets or the repository as a ZIP file or tarball. I have never encountered such a limit, so it does not seem to be of concern for testing on GitHub Actions (given that we download tarballs during the CI build from GitHub repositories, e.g., from the repository of `slugs`).)

(Also, my understanding is that in GitHub's documentation, the word "bandwidth" refers to cummulative amount of information transferred, and not to the rate of information transfer.)

(Sidenote: Git LFS uploads to GitHub's LFS storage do not count towards the bandwidth limit:
[... cut ...]
## `git` LFS supports only HTTPS

Recently, GitHub [discontinued password-based authentication](https://github.blog/2021-08-16-securing-your-github-account-two-factor-authentication/#no-more-password-based-authentication-for-git-operations). [`git-lfs` supports HTTPS/HTTP](https://github.com/git-lfs/git-lfs/wiki/Tutorial#credential-cachingstorage), so I think that using an SSH key is not possible. Some configuration that does not use a user password for HTTPS would be needed. I mention this detail for completeness.

closed 2021-09-05 by johnyf