tulip-control/tulip-control #247: move CI to GitHub Actions, because Travis CI is not free anymore

johnyf, 2021-06-11:
I switched `tulip-control` to using [GitHub Actions](https://docs.github.com/en/actions) for [continuous integration](https://en.wikipedia.org/wiki/Continuous_integration) [testing](https://en.wikipedia.org/wiki/Continuous_testing) (this change is not yet merged to the mainline branch).

Previously, Travis CI was used. The reasons for this transition are described below. Since Travis CI is not used anymore, I deactivated builds from the "More options" -> "Settings" page:
https://travis-ci.com/github/tulip-control/tulip-control/settings

Except for the `stormpy`-related tests, all other tests pass:

```
=========================== short test summary info ============================
ERROR stormpy_interface_test.py
======= 193 passed, 19 xfailed, 2 warnings, 1 error in 67.78s (0:01:07) ========
Error: Process completed with exit code 1.
```
https://github.com/tulip-control/tulip-control/runs/2803920342?check_suite_focus=true#step:21:272

Also, I am considering using caching on GitHub Actions to speed up the builds, by caching compiled files (perhaps downloaded dependencies too). Doing so could reduce the duration of each CI run from about 15 minutes to about 3 minutes. This will save resources and human time.

I plan to transition also CI testing `polytope`, `dd`, and `omega` to GitHub Actions.

## Travis CI is not free

Travis CI changed to nonfree, summarizing from:
https://blog.travis-ci.com/2020-11-02-travis-ci-new-billing

- the `travis-ci.com` "free" plan means a an allotment of 10,000 ["credits"](https://docs.travis-ci.com/user/billing-overview/#usage---credits)

- after this allotment has been used, either switch to a paid plan,
  or apply to Travis CI support to be considered for the "OSS allotment"

- Each Linux instance minute [counts as](https://docs.travis-ci.com/user/billing-overview/#usage---credits)10 credits

I noticed this change when I pushed and there was no Travis CI build.
To find out what happened, I had to click on "More options" -> "Requests" at the top right of the Travis CI webpage, to go to the webpage:
https://travis-ci.com/github/tulip-control/tulip-control/requests
where it says:
> Could not authorize build request for tulip-control/tulip-control.

I summed the minutes that have been used for all builds of all repositories of the GitHub organization `tulip-control`, and the sum corresponds to about 76\% of the "credits" available.

In any case, Travis CI can no longer be used for CI testing, so a transition is needed to another provider. My suggestion is to use GitHub Actions. We could also consider other providers, for example Circle CI appears to have quite high an upper bound on testinng time per month.

Both GitHub Actions and Circle CI appear to require that the YAML configuration file for the CI builds be placed in a specific path (e.g., `.github/workflows/*`).

## GitHub Actions

GitHub Actions appears to allow unlimited minutes for CI testing of public repositories, quoting:

https://docs.github.com/en/github/getting-started-with-github/learning-about-github/faq-about-changes-to-githubs-plans#is-anything-changing-for-teams-on-the-github-team-for-open-source-plan
> ... 2,000 Actions minutes for private repositories, and GitHub Community Support.
> Actions minutes and Packages storage are still free for your public repositories.

and:

https://docs.github.com/en/billing/managing-billing-for-github-actions/about-billing-for-github-actions#about-billing-for-github-actions
> GitHub Actions usage is free for both public repositories and self-hosted runners.

From what I can see for other existing GitHub Actions usage, it does not count
towards an organization's monthly minutes of GitHub Actions.

I also changed the access of GitHub Actions to the repository to read-only for
now, while experimenting with GitHub Actions, to guard against unintended
things happening accidentally.

## About Circle CI

There is also Circle CI (and other continuous integration providers).
Circle CI offers:

> Organizations on our free plan get 400,000 credits per month for open source builds.
https://circleci.com/open-source/

> ... when using a single machine at the default rate of 10 credits per minute.
https://circleci.com/docs/2.0/faq/#what-are-credits

This translates to about 295 builds per repository, assuming 3 active
repositories, with each build comprised of 3 test instances, and each
test instance lasting 15 minutes.

murrayrm, 2021-06-11:
I think this is a good move overall.  Other projects I am involved with have switched to GitHub actions.

johnyf, 2021-06-11:
I intend to review my changes and then merge.

johnyf, 2021-08-21:
The transition has been completed in commit 7ececa4bcae51c85759e7678f706c160e0fc4fa5. The other repositories mentioned above (https://github.com/tulip-control/tulip-control/issues/247#issue-918979847) have been moved to using GitHub Actions on the development branches. I set up a `cron` task in GitHub Actions to run the tests once per month.

closed 2021-08-21 by johnyf

johnyf, 2021-09-05:
As noted above (https://github.com/tulip-control/tulip-control/issues/247#issuecomment-903095842), I added a `cron` task in GitHub Actions:

https://github.com/tulip-control/tulip-control/blob/7ececa4bcae51c85759e7678f706c160e0fc4fa5/.github/workflows/main.yml#L11

The scheduled task run as expected today, confirming that this configuration had the intended effect.

slivingston, 2022-06-17:
@johnyf excellent work on this!