tulip-control/tulip-control #249: Failed to build PLY table.  Please run setup.py

khassibi, 2023-06-21:
I was given the advice by Apurva Badithela to install TuLiP by using these commands
`$ git clone https://github.com/tulip-control/tulip-control.git`
`$ cd tulip-control`
`$ git checkout eeci2020`
`$ pip install wheel`
`$ pip install cvxopt`
`$ pip install -r requirements.txt`
`$ python setup.py install`
(These directions were found towards the bottom of [https://murray.cds.caltech.edu/EECI-IGSC_2020](https://murray.cds.caltech.edu/EECI-IGSC_2020).)

I used a virtual environment (venv) when doing these commands. Everything worked well, except when running `python setup.py install`, I got the message
`!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!    Failed to build PLY table.  Please run setup.py again.!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!`

This issue was brought up in https://github.com/tulip-control/tulip-control/issues/132#issuecomment-152637362, where the advice was to disable pip's caching behavior with `--no-cache-dir`. I have tried adding this tag to the pip install's (`pip install wheel`, `pip install cvxopt`, `pip install -r requirements.txt`), but I still run into the same issue. 

Does anyone know how to resolve this issue? Thank you!

slivingston, 2023-06-22:
Do you observe this on the `main` branch, too?

```
git checkout main
```

The code about minimum violation planning was merged into `main` after that workshop (https://github.com/tulip-control/tulip-control/pull/236).

tichakornw, 2023-06-22:
Scott is right that you should have all the eeci code in main.

I tried setting up using the main branch and was able to reproduce the
issue. In fact, I ran into more issues. Please find below the issues and a
way I fixed it. Mostly, it's about getting the right versions of the
dependencies.

1. When running

 $ pip install -r requirements.txt

I got an error when it tries to install gr1py==0.2.0

Collecting gr1py==0.2.0 (from -r requirements.txt (line 10))
  Downloading gr1py-0.2.0.tar.gz (22 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [4 lines of output]
      !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
      Please run again to cache the parsing table.
      !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
      error in gr1py setup command: use_2to3 is invalid.
      [end of output]

I fixed this by simply installing gr1py without specifying the version,
i.e.,

  $ pip install gr1py

2. When running

   $ python setup.py install

I got a long error that starts with

Failed to build PLY tables: module 'numpy' has no attribute 'int'.
`np.int` was a deprecated alias for the builtin `int`. To avoid this error
in existing code, use `int` by itself. Doing this will not modify any
behavior and is safe. When replacing `np.int`, you may wish to use e.g.
`np.int64` or `np.int32` to specify the precision. If you wish to review
your current use, check the release note link for additional information.

and ends with

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!    Failed
to build PLY table.  Please run setup.py
again.!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

This gave some hints that there is some incompatibility between the version
of numpy I installed and what the installation wants. So I went back a few
version. What seems to work is to do

   $ pip install numpy==1.19.1 scipy==1.5.2

Then, rerun

   $ python setup.py install

There are warning about setup.py install is deprecated but at least there
is no error.

Hope this helps.

Best,
Nok

khassibi, 2023-06-22:
Thank you for the help!
I started over and tried following https://github.com/tulip-control/tulip-control/issues/249#issuecomment-1601962166 exactly. I did the following in a virtual environment:
`git clone https://github.com/tulip-control/tulip-control.git`
`cd tulip-control`
`pip install wheel`
`pip install cvxopt`
`pip install gr1py`
`pip install -r requirements.txt`
`python3.9 setup.py install`
`pip install numpy==1.19.1 scipy==1.5.2`
`python3.9 setup.py install`

Everything went well, except in the very end, `python3.9 setup.py install` failed to build PLY table again. Below, I have included everything that was outputted from the `python3.9 setup.py install` command.

WARNING:polytope.solvers:`polytope` failed to import `cvxopt.glpk`.
WARNING:polytope.solvers:will use `scipy.optimize.linprog`
WARNING:omega.symbolic.symbolic:`omega.symbolic.symbolic` failed to import `dd.cudd`.
Will use `dd.autoref`.
WARNING: Token 'TRUE' defined, but not used
WARNING: Token 'FALSE' defined, but not used
WARNING: Token 'XOR' defined, but not used
WARNING: Token 'BIMP' defined, but not used
WARNING: Token 'EQUALS' defined, but not used
WARNING: Token 'NEQUALS' defined, but not used
WARNING: Token 'LT' defined, but not used
WARNING: Token 'LE' defined, but not used
WARNING: Token 'GT' defined, but not used
WARNING: Token 'GE' defined, but not used
WARNING: Token 'PLUS' defined, but not used
WARNING: Token 'MINUS' defined, but not used
WARNING: Token 'TIMES' defined, but not used
WARNING: Token 'DIV' defined, but not used
WARNING: Token 'SKIP' defined, but not used
WARNING: /Users/kimiahassibi/Desktop/Caltech/SURF2023/tulip-control/tulip/interfaces/ltl2ba.py:242: Rule 'empty' defined, but not used
WARNING: There are 15 unused tokens
WARNING: There is 1 unused rule
WARNING: Symbol 'empty' is unreachable
WARNING:tulip.abstract.find_controller:`tulip` failed to import `cvxopt`.
No quadratic cost for controller computation.
Generating LALR tables
Failed to build PLY tables: 'Parser' object has no attribute 'build'
No git info: Assume release.
INFO:root:running install
/usr/local/lib/python3.9/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/local/lib/python3.9/site-packages/setuptools/command/easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
INFO:root:running bdist_egg
INFO:root:running egg_info
INFO:root:creating tulip.egg-info
INFO:root:writing tulip.egg-info/PKG-INFO
INFO:root:writing dependency_links to tulip.egg-info/dependency_links.txt
INFO:root:writing requirements to tulip.egg-info/requires.txt
INFO:root:writing top-level names to tulip.egg-info/top_level.txt
INFO:root:writing manifest file 'tulip.egg-info/SOURCES.txt'
INFO:root:reading manifest file 'tulip.egg-info/SOURCES.txt'
INFO:root:reading manifest template 'MANIFEST.in'
INFO:root:adding license file 'LICENSE'
INFO:root:adding license file 'AUTHORS'
INFO:root:writing manifest file 'tulip.egg-info/SOURCES.txt'
INFO:root:installing library code to build/bdist.macosx-12-x86_64/egg
INFO:root:running install_lib
INFO:root:running build_py
INFO:root:creating build
INFO:root:creating build/lib
INFO:root:creating build/lib/tulip
INFO:root:copying tulip/graphics.py -> build/lib/tulip
INFO:root:copying tulip/synth.py -> build/lib/tulip
INFO:root:copying tulip/_version.py -> build/lib/tulip
INFO:root:copying tulip/hybrid.py -> build/lib/tulip
[... 248 lines cut ...]
Using /usr/local/lib/python3.9/site-packages/dd-0.5.7-py3.9.egg
INFO:root:Searching for astutils==0.0.4
INFO:root:Best match: astutils 0.0.4
INFO:root:Processing astutils-0.0.4-py3.9.egg
INFO:root:astutils 0.0.4 is already the active version in easy-install.pth
INFO:root:
Using /usr/local/lib/python3.9/site-packages/astutils-0.0.4-py3.9.egg
INFO:root:Searching for pyparsing==3.1.0
INFO:root:Best match: pyparsing 3.1.0
INFO:root:Processing pyparsing-3.1.0-py3.9.egg
INFO:root:pyparsing 3.1.0 is already the active version in easy-install.pth
INFO:root:
Using /usr/local/lib/python3.9/site-packages/pyparsing-3.1.0-py3.9.egg
INFO:root:Searching for setuptools==65.6.3
INFO:root:Best match: setuptools 65.6.3
INFO:root:Adding setuptools 65.6.3 to easy-install.pth file
INFO:root:
Using /usr/local/lib/python3.9/site-packages
INFO:root:Searching for psutil==5.9.4
INFO:root:Best match: psutil 5.9.4
INFO:root:Adding psutil 5.9.4 to easy-install.pth file
INFO:root:
Using /Users/kimiahassibi/Library/Python/3.9/lib/python/site-packages
INFO:root:Finished processing dependencies for tulip==1.4.0
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!    Failed to build PLY table.  Please run setup.py again.!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

slivingston, 2023-06-23:
@khassibi Are you certain that `python3.9` is running in the virtual environment? Usually it is named `python`. You can confirm which program is running by comparing
```
which python3.9
```
and
```
which python
```

tichakornw, 2023-06-23:
I tried with python 3.9 and I couldn't even do

$ pip install scipy==1.5.2

So perhaps you're not using python 3.9? The fix I talked about earlier uses
python 3.8.

Best,
Nok

khassibi, 2023-06-23:
So my virtual environment uses python 3.8. When I do `python setup.py install`, I get the following output.

WARNING: no p_error() function is defined
WARNING: Token 'COMMENT' defined, but not used
WARNING: Token 'EVENTUALLY_OP' defined, but not used
WARNING: Token 'newline' defined, but not used
WARNING: There are 3 unused tokens
Generating LALR tables
WARNING: 16 shift/reduce conflicts
WARNING:omega.symbolic.symbolic:`omega.symbolic.symbolic` failed to import `dd.cudd`.
Will use `dd.autoref`.
WARNING: Token 'BIMP' defined, but not used
WARNING: Token 'DIV' defined, but not used
WARNING: Token 'EQUALS' defined, but not used
WARNING: Token 'FALSE' defined, but not used
WARNING: Token 'GE' defined, but not used
WARNING: Token 'GT' defined, but not used
WARNING: Token 'LE' defined, but not used
WARNING: Token 'LT' defined, but not used
WARNING: Token 'MINUS' defined, but not used
WARNING: Token 'NEQUALS' defined, but not used
WARNING: Token 'PLUS' defined, but not used
WARNING: Token 'SKIP' defined, but not used
WARNING: Token 'TIMES' defined, but not used
WARNING: Token 'TRUE' defined, but not used
WARNING: Token 'XOR' defined, but not used
WARNING: /Users/kimiahassibi/Desktop/Caltech/SURF2023/tulip-control/tulip/interfaces/ltl2ba.py:242: Rule 'empty' defined, but not used
WARNING: There are 15 unused tokens
WARNING: There is 1 unused rule
WARNING: Symbol 'empty' is unreachable
Generating LALR tables
Failed to build PLY tables: 'Parser' object has no attribute 'build'
No git info: Assume release.
running install
running bdist_egg
running egg_info
writing tulip.egg-info/PKG-INFO
writing dependency_links to tulip.egg-info/dependency_links.txt
writing requirements to tulip.egg-info/requires.txt
writing top-level names to tulip.egg-info/top_level.txt
adding license file 'LICENSE' (matched pattern 'LICEN[CS]E*')
adding license file 'AUTHORS' (matched pattern 'AUTHORS*')
reading manifest file 'tulip.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'tulip.egg-info/SOURCES.txt'
installing library code to build/bdist.macosx-10.9-x86_64/egg
running install_lib
running build_py
copying tulip/_version.py -> build/lib/tulip
copying tulip/spec/ltl_parsetab.py -> build/lib/tulip/spec
copying tulip/interfaces/ltl2ba_parsetab.py -> build/lib/tulip/interfaces
creating build/bdist.macosx-10.9-x86_64/egg
creating build/bdist.macosx-10.9-x86_64/egg/tulip
creating build/bdist.macosx-10.9-x86_64/egg/tulip/abstract
copying build/lib/tulip/abstract/plot.py -> build/bdist.macosx-10.9-x86_64/egg/tulip/abstract
copying build/lib/tulip/abstract/discretization.py -> build/bdist.macosx-10.9-x86_64/egg/tulip/abstract
copying build/lib/tulip/abstract/__init__.py -> build/bdist.macosx-10.9-x86_64/egg/tulip/abstract
copying build/lib/tulip/abstract/find_controller.py -> build/bdist.macosx-10.9-x86_64/egg/tulip/abstract
copying build/lib/tulip/abstract/prop2partition.py -> build/bdist.macosx-10.9-x86_64/egg/tulip/abstract
copying build/lib/tulip/abstract/feasible.py -> build/bdist.macosx-10.9-x86_64/egg/tulip/abstract
copying build/lib/tulip/graphics.py -> build/bdist.macosx-10.9-x86_64/egg/tulip
copying build/lib/tulip/synth.py -> build/bdist.macosx-10.9-x86_64/egg/tulip
copying build/lib/tulip/_version.py -> build/bdist.macosx-10.9-x86_64/egg/tulip
copying build/lib/tulip/hybrid.py -> build/bdist.macosx-10.9-x86_64/egg/tulip
copying build/lib/tulip/gridworld.py -> build/bdist.macosx-10.9-x86_64/egg/tulip
creating build/bdist.macosx-10.9-x86_64/egg/tulip/spec
copying build/lib/tulip/spec/ltl_parsetab.py -> build/bdist.macosx-10.9-x86_64/egg/tulip/spec
copying build/lib/tulip/spec/transformation.py -> build/bdist.macosx-10.9-x86_64/egg/tulip/spec
copying build/lib/tulip/spec/gr1_fragment.py -> build/bdist.macosx-10.9-x86_64/egg/tulip/spec
copying build/lib/tulip/spec/__init__.py -> build/bdist.macosx-10.9-x86_64/egg/tulip/spec
[... 174 lines cut ...]
Searching for astutils==0.0.4
Best match: astutils 0.0.4
Processing astutils-0.0.4-py3.8.egg
astutils 0.0.4 is already the active version in easy-install.pth

Using /Users/kimiahassibi/Desktop/Caltech/SURF2023/env/lib/python3.8/site-packages/astutils-0.0.4-py3.8.egg
Searching for pyparsing==3.1.0
Best match: pyparsing 3.1.0
Processing pyparsing-3.1.0-py3.8.egg
pyparsing 3.1.0 is already the active version in easy-install.pth

Using /Users/kimiahassibi/Desktop/Caltech/SURF2023/env/lib/python3.8/site-packages/pyparsing-3.1.0-py3.8.egg
Searching for setuptools==56.0.0
Best match: setuptools 56.0.0
Adding setuptools 56.0.0 to easy-install.pth file

Using /Users/kimiahassibi/Desktop/Caltech/SURF2023/env/lib/python3.8/site-packages
Searching for psutil==5.9.5
Best match: psutil 5.9.5
Processing psutil-5.9.5-py3.8-macosx-10.9-x86_64.egg
psutil 5.9.5 is already the active version in easy-install.pth

Using /Users/kimiahassibi/Desktop/Caltech/SURF2023/env/lib/python3.8/site-packages/psutil-5.9.5-py3.8-macosx-10.9-x86_64.egg
Finished processing dependencies for tulip==1.4.0
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!    Failed to build PLY table.  Please run setup.py again.!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

khassibi, 2023-06-26:
Hi, so I figured out that if I do the following commands

`pip install wheel`
`pip install cvxopt`   
`pip install gr1py`
`pip install -r requirements.txt`
`python setup.py install`  
`pip install numpy==1.19.1 scipy==1.5.2` 
`python setup.py install`

slivingston, 2023-06-26:
@khassibi It is outdated, but possibly not in a way that is significant for your work.

I am going to explore this issue more today. I will try a new Tulip installation on a MacBook with macOS 13.4.1 and arm64 (Apple Silicon).

abadithela, 2023-09-05:
Hi @slivingston, has this issue been resolved? I'm happy to help fix it if it has not.

slivingston, 2023-09-29:
@abadithela Sorry, I missed your message. It is not resolved. If you have some time, your investigating it is welcome!