tulip-control/tulip-control #250: tulip.transys.machines.interactive_run() is not working as intended

khassibi, 2023-07-21:
Hello,

I have been trying to use the [interactive_run](https://github.com/tulip-control/tulip-control/blob/main/tulip/transys/machines.py#L694) function found in tulip-control/tulip/transys/machines.py. 
I have tried using the code found in [gr1.py of the eeci2020 branch](https://github.com/tulip-control/tulip-control/blob/eeci2020/examples/gr1.py). The code is able to run `machines.random_run(strategy, from_state='Sinit', N=10)` successfully. However, when I change this line to `machines.interactive_run(strategy, from_state='Sinit')`, I get stuck in a loop. First, I see the following text printed:

`Found more than 1 outgoing transitions:

        0 : Sinit ---> 0
         inputs:{'park': False}  outputs:{'X0': True, 'X0reach': True, 'X5': False, 'X4': False, 'X3': False, 'X1': False, 'X2': False}

        1 : Sinit ---> 1
         inputs:{'park': True}   outputs:{'X0': True, 'X0reach': True, 'X5': False, 'X4': False, 'X3': False, 'X1': False, 'X2': False}

Select from the available transitions above
by giving its integer,
Press "Enter" to stop the simulation:
         int = `

I am asked to pick a transition for the environment to take, but any option I choose once again prints the text above. Moreover, if I try pressing "Enter", the simulation does not stop, and the text above is printed once more. 

I have also tried running this code using the line `strategy.run('Sinit')` ([function found here](https://github.com/tulip-control/tulip-control/blob/main/tulip/transys/machines.py#L582)). I get the same results.

Please let me know if I am doing anything wrong or if there is an issue with this code. Thank you!

slivingston, 2023-08-26:
@khassibi Did you find a solution, or is this question open?

khassibi, 2023-08-30:
I did not find a solution, but I found other functions that allow me to select an environment action and see the system's response.