tulip-control/tulip-control #262: Unable to install tulip

ranai-srivastav, 2024-04-16:
I'm trying to install TuLiP and I am unable to install the library successfully. I am on the main branch and I tried installing using the command `pip3 install .` and it is failing with the error below. I have tried with and without `--no-cache-dir`

I'm using Python 3.8.10

Commands run:
git clone https://github.com/tulip-control/tulip-control.git
cd tulip-control
pip install -r requirements.txt
pip install .

Error message:

```python
root@06ba197e9410:~/software/tulip-control# pip3 install . --no-cache-dir
Processing /root/software/tulip-control
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [22 lines of output]
      Traceback (most recent call last):
        File "/root/software/tulip-control/setup.py", line 77, in run_setup
          import tulip.spec.lexyacc
        File "/root/software/tulip-control/tulip/__init__.py", line 53, in <module>
          import tulip.abstract
        File "/root/software/tulip-control/tulip/abstract/__init__.py", line 34, in <module>
          from . import discretization
        File "/root/software/tulip-control/tulip/abstract/discretization.py", line 644
          match simu_type:
                ^
      SyntaxError: invalid syntax
      
      During handling of the above exception, another exception occurred:
      
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/root/software/tulip-control/setup.py", line 182, in <module>
          run_setup()
        File "/root/software/tulip-control/setup.py", line 100, in run_setup
          _tb.format_exception(e))
      TypeError: format_exception() missing 2 required positional arguments: 'value' and 'tb'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

ranai-srivastav, 2024-04-16:
Solution:

Update python version to 3.10
Update pip3 version to match python3.10

I followed the steps here https://gist.github.com/rutcreate/c0041e842f858ceb455b748809763ddb but this is not the best way.

closed 2024-04-16 by ranai-srivastav

slivingston, 2024-04-16:
@ranai-srivastav I'm happy you noticed the minimum Python version. We are updating the documentation to make first-time installing of TuLiP easier.

Can you tell me: where would you have noticed a warning about the minimum Python version? At the top of the [README](https://github.com/tulip-control/tulip-control/blob/eb4d0868b53ac74c51c40f467951f74bdf55c9e5/README.rst) ? (I just noticed that it does not indicate the minimum Python.)

ranai-srivastav, 2024-04-17:
I did not find it in the documentation. Dr. Nok sent me a Dockerfile and that didn't work either. 

I was going through the closed issues to see if someone else had had the same issue. I actually was on the main branch and saw the CI was failing and i was wondering if that had the same issue as well. There I noticed that you CI only had tests for Python 3.10 and 3.11 and so I tried and it worked. 

I commented it here in case people need this as well.